ptrkrysik/gr-gsm #256: Problem grgsm_livemon.grc

Anonymous002, 2017-02-12:
# hey # ptrkrysik how can i solve this problem

Loading: "grgsm_livemon.grc"
>>> Error: Block key "gsm_bcch_ccch_demapper" not found
>>> Error: Block key "gsm_clock_offset_control" not found
>>> Error: Block key "gsm_control_channels_decoder" not found
>>> Error: Block key "gsm_control_channels_decoder" not found
>>> Error: Block key "gsm_decryption" not found
>>> Error: Block key "gsm_input" not found
>>> Error: Block key "gsm_message_printer" not found
>>> Error: Block key "gsm_receiver" not found
>>> Error: Block key "gsm_sdcch8_demapper" not found
>>> Error: Block key "rtlsdr_source" not found
>>> Done

Anonymous002, 2017-02-12:
problem fixed thanks 
can i use grgsm_livemon withou hackrf ??

Alster, 2017-02-12:
Of course. What errors in output you see?

Anonymous002, 2017-02-15:
i don't have rtl sdr or hackrf - it's just the gnuradio when i start it the radio still white there's not blue lines , please solution Alster , or tell if i missing something in installation

Alster, 2017-02-16:
For gotting "blue lines" GNU Radio must have some input like hackrf, rtlsdr, airspy, usrp, bladerf or other stuff like this. Or you can use already recorded sample located in "test_data" folder. See [wiki ](https://github.com/ptrkrysik/gr-gsm/wiki/Usage:-Decoding-How-To) for details.
I think issue can be closed =)

Anonymous002, 2017-02-16:
Thanks :+1:

ptrkrysik, 2017-02-17:
You can use any device supported by GNU Radio. Even if the device is not supported by gr-osmosdr block (that supports plenty of SDR devices) you can change RTL-SDR block with any other gnuradio source block. For example if you have USRP and find UHD block more convenient - you can use it as the source of the signal for gr-gsm.

p.s. for such questions/discussions the mailing list is preferred place. Here after closing the github issue not many will see it in the future.

closed 2017-02-17 by ptrkrysik

belahssen, 2017-02-17:
hey ptrkrysik please show me how to install pybombs and gr-gsm 
show me what i should do or type in terminal 
i use ubuntu 16.04