ptrkrysik/gr-gsm #259: GSM 1800MHZ

Mitchell1314, 2017-02-22:
Hi sir,
    I have installed gr-gsm on my ubuntu14.03. And I can run "grgsm_scanner" with hackrf, then i get the result:
[image]
And I want to get the same results of GSM 1800MHZ. How can I do that? By running "grgsm_scanner 1800e6"?

romankh, 2017-02-26:
```grgsm_scanner -h``` will give you all available options of the command, what you need is to specify the DCS 1800 band with option ```-b```

s6thsense, 2017-02-26:
Hi Piotr,

i've think you are confused about my topic.
my topic was not " grgsm_scanner on DCS1800
<https://groups.google.com/d/msg/gr-gsm/Ds6JbCn8eL8/J2BE1yOkCwAJ>"
But it was "grgsm_scanner" via googlegroups. (Willem)
<https://groups.google.com/d/msg/gr-gsm/Ds6JbCn8eL8/J2BE1yOkCwAJ>

This is the link:
ttps://groups.google.com/forum/?_escaped_fragment_=topic/gr-gsm/WPZFnLwua0M#!topic/gr-gsm/WPZFnLwua0M

Kind regards,
Sense
<https://groups.google.com/d/msg/gr-gsm/Ds6JbCn8eL8/J2BE1yOkCwAJ>

<https://groups.google.com/d/msg/gr-gsm/Ds6JbCn8eL8/J2BE1yOkCwAJ>

Mitchell1314, 2017-02-28:
yeah i can find dcs1800 option which resolve my problem, and now I want to call "grgsm_scanner" by "os.popen" in python but I have got nothing.By the way, I  can call "grgsm_scanner -h" by "os.popen" successfully:
[image]

Can you give me some help? Thank you.

romankh, 2017-03-01:
grgsm_scanner itself does tamper the file descriptors for stdout and stdin to avoid unwanted output, this may be the cause for having no output in your case. 

If you want to change the behaviour of the script, I would rather suggest to make a copy of grgsm_scanner, rename it and implement the changes in the copy.

Btw, this topic is something that you should rather ask on the mailing list.

Mitchell1314, 2017-03-03:
Thanks for your suggestion. I followed your suggestion and  my problem have been solved. Thanks a lot!

closed 2017-03-03 by romankh