ptrkrysik/gr-gsm #260: how Can i fix this (gnuradio)

belahssen, 2017-02-22:
please solution !
<<< Welcome to GNU Radio Companion 3.7.10 >>>

Block paths:
	/usr/share/gnuradio/grc/blocks

Loading: "/root/Desktop/gsm/gr-gsm/apps/grgsm_livemon"
Error: /root/Desktop/gsm/gr-gsm/apps/grgsm_livemon:1:1:FATAL:PARSER:ERR_DOCUMENT_EMPTY: Start tag expected, '<' not found
>>> Failure

XML parser: Found 1 erroneous XML file while loading the block tree (see "Help/Parser errors" for details)

ptrkrysik, 2017-02-23:
Can you tell me what are you trying to achieve. It looks that you are
trying to open Python file in gnuradio-companion, which expects xml file.

belahssen, 2017-02-24:
i'm sorry this problem fixed and grgsm_livemon.grc opened in gnuradio but now this problem come to me and i don't know how can i fix it -- i'm just want to sniffing GSM SMS

Traceback (most recent call last):
  File "/root/Desktop/gr-gsm/apps/grgsm_livemon.py", line 31, in <module>
    import grgsm
  File "/usr/local/lib/python2.7/dist-packages/grgsm/__init__.py", line 45, in <module>
    from grgsm_swig import *
  File "/usr/local/lib/python2.7/dist-packages/grgsm/grgsm_swig.py", line 21, in <module>
    _grgsm_swig = swig_import_helper()
  File "/usr/local/lib/python2.7/dist-packages/grgsm/grgsm_swig.py", line 20, in swig_import_helper
    return importlib.import_module('_grgsm_swig')
  File "/usr/lib/python2.7/importlib/__init__.py", line 37, in import_module
    __import__(name)
ImportError: No module named _grgsm_swig

Varbaek, 2017-03-31:
Do you have an RTL-SDR or another compatible SDR? Consider reinstalling on a fresh Ubuntu VM.

belahssen, 2017-03-31:
no i don't have / this don't work without this device ?

ptrkrysik, 2017-04-02:
W dniu 01.04.2017 o 00:01, virusssss pisze:
> no i don't have / this don't work without this device ?
> 

To run grgsm_livemon you will need a SDR device.

grgsm_decode and some of the examples work for capture files stored to
disk. There are some in the test_data repository:
https://github.com/ptrkrysik/test_data

With your problem, try running:
sudo ldconfig

closed 2017-06-11 by ptrkrysik