ptrkrysik/gr-gsm #261: block dummy_burst_filter filter error

nikosvir, 2017-02-27:
Hello,

I'm pretty sure I'm missing something here, but sometimes when I try to decode an already captured burst file (grgsm_decode -a 8 -s 3M -b captured.burst) I get the following error and grgsm_decode just hangs:

thread[thread-per-block[4]: <block dummy_burst_filter (2)>]: pmt_cdr: wrong_type : #t

I'm I missing something or it is a bug? If needed, I can upload the burst file somewhere.

Cheers

s6thsense, 2017-02-28:
> thread [2].
> Links:

Hi,

I think there is a mix up about topics.
My question was asked via googlegroups. (user Willem)
https://groups.google.com/forum/?_escaped_fragment_=topic/gr-gsm/WPZFnLwua0M#!topic/gr-gsm/WPZFnLwua0M

Kind regards,
Willem

romankh, 2017-03-01:
@nikosvir: could be a bug, please provide the burst file for debugging.
Thank you.

ptrkrysik, 2017-06-11:
Please provide a recipe how to reproduce this bug. I will reopen the issue then.

closed 2017-06-11 by ptrkrysik