ptrkrysik/gr-gsm #262: Feature Request: grgsm_capture.py: Split burst or cfile to multiple files

nikosvir, 2017-02-28:
Hi,

A nice feature would be the ability for grgsm_capture.py to sniff continuously and every X seconds to save a new burst or cfile:

eg. capture.burst.01, capture.burst.02 

The idea is to have grgsm_capture.py running continuously, and have multiple instances of grgsm_decode (probably on different systems as they cannot run in parallel (Bind error blocks.socket_pdu("UDP_SERVER", "127.0.0.1", "4729", 10000))), to analyze the captured files sequentially.

Regarding the bind error, is it possible to have grgsm_decode printing the bursts at the command line but not binding to 'lo' interface so multiple instances can be run on the same machine? Or just bind on different ports?

Cheers,
Nick

ptrkrysik, 2017-06-11:
This would be a nice feature in GNU Radio generally to be able to split capture files. If you are willing to implement such feature I can provide some help. I don't have time to do this myself and since February nobody else showed up so I'm closing the issue for now.

closed 2017-06-11 by ptrkrysik

Alster, 2017-06-12:
Try to just comment out this line
https://github.com/ptrkrysik/gr-gsm/blob/master/apps/grgsm_decode#L112

On account of file splitting: you can try to play with bash, something like: 
make fifo file (mkfifo fname)
and then try to split it (split -l 200000 fname)
If it works - tell me)