ptrkrysik/gr-gsm #264: ubuntu14.04  on vmware

Mitchell1314, 2017-03-06:
Hi sir,
       I have installed gr-gsm on ubuntu14.04 on vmware, but I have got nothing:
[image]
what can I do to fix this? 
      What's more, I want to install gr-gsm on RaspberryPi 3 with ubuntu mate 16. Can I do this?

Robert-steve, 2017-03-09:
are you using an rtl-sdr or a usurp?
I would suggest to try grgsm_livemon first to see if you are capturing any signal.

Mitchell1314, 2017-03-09:
Hi sir,
    I have ran "grgsm_scanner" with hackrf. When I ran "kal -s GSM900 -g 40"，I can get the result:
[image]
But when I ran "grgsm_scanner" ,I have got nothing:
[image]

By the way, the result of running "grgsm_livemon" is as follow:
[image]

What can I do to fix this? Thank you. Waiting for your reply.

Robert-steve, 2017-03-09:
when you set the frequency in grgsm_livemon to a frequency where there is a BTS broadcasting channel, do you get any data ? If you do not get any data then maybe you should set the samp_rate to a value other than the default.

Mitchell1314, 2017-03-09:
I set the frequency to 95.4e6, and i get the result as follows:
[image]

Robert-steve, 2017-03-09:
it should be 954e6 instead of 95.4e6

ptrkrysik, 2017-03-14:
Generally it isn't recommended to use gr-gsm inside virtual machine.

However you... did you try to change frequency correction (ppm) in the
grgsm_livemon? The offsets computed by kal don't look very reliable but
some of them are quite large.

P.S. Can you use the mailing list to continue this topic:
https://github.com/ptrkrysik/gr-gsm#mailing-list ? :)

closed 2017-06-11 by ptrkrysik