ptrkrysik/gr-gsm #266: grgsm on Raspberry Pi 2 Model B

Mitchell1314, 2017-03-06:
Hi sir,
   I tried to install grgsm on mu raspberry pi 2, and firstly I ran "kal -s GSM900 -g 40",i have got the result:
[image]

Secondly, I ran "grgsm_scanner -p 36",i have got nothing:
[image]

Finally,here are some infomation about my raspberry pi:
[image]

Can you give me some help to fix it?

ars0, 2017-03-29:
It seems that the problem is no RPi related. 
I have exactly the same problem with the default installation on Ubuntu 16.04. Installed gr-gsm (and all dependencies) with pybombs.

rojavatechops, 2017-04-10:
Same problem here.

tomturner123, 2017-04-12:
I also have this problem with the raspberry pi 3 (raspbian jessie) - I followed the wiki installation, fresh install. All that happens is:
"linux; GNU C++ version 4.9.1; Boost_105500; UHD_003.007.003-0-unknown"

ptrkrysik, 2017-04-12:
W dniu 12.04.2017 o 17:00, tomturner123 pisze:
> I also have this problem with the raspberry pi 3 (raspbian jessie) - I
> followed the wiki installation, fresh install. All that happens is:
> "linux; GNU C++ version 4.9.1; Boost_105500; UHD_003.007.003-0-unknown"
> 
Which gr-gsm git revision do you have?

Can you try to comment out lines (347,348):
            os.dup2(null_fds[0], 1)
            os.dup2(null_fds[1], 2)

from grgsm_scanner and post the result?

rojavatechops, 2017-04-12:
After changing the script, and running grgsm_scanner rom the CLI, I get the same error. 

 "linux; GNU C++ version 4.9.1; Boost_105500; UHD_003.007.003-0-unknown"

tomturner123, 2017-04-12:
I get:
linux; GNU C++ version 4.9.1; Boost_105500; UHD_003.007.003-0-unknown

Args= 
gr-osmosdr 0.1.3 (0.1.3) gnuradio 3.7.5
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy 
Using device #0 Realtek RTL2838UHIDIR
Found Elonics E4000 tuner
Traceback (most recent call last):
  File "/usr/local/bin/grgsm_scanner", line 355, in <module>
    ppm=options.ppm, gain=options.gain, args=options.args)
  File "/usr/local/bin/grgsm_scanner", line 213, in __init__
    self.rtlsdr_source.set_min_output_buffer(int(sample_rate*rec_len))
AttributeError: 'source_sptr' object has no attribute 'set_min_output_buffer'

How do I get the git revision? git clone was done yesterday.

Also, in case its relevant, airprobe_rtlsdr.py outputs:
Traceback (most recent call last):
  File "/usr/local/bin/airprobe_rtlsdr.py", line 29, in <module>
    from gnuradio.qtgui import Range, RangeWidget
ImportError: cannot import name Range

tomturner123, 2017-04-20:
I managed to get it to work. I am pretty confident that the raspberry pi installation method on the wiki is wrong, as it seems the gnuradio version installed with apt-get is incompatible. Unfortunately installing the latest version is not trivial. After following the instructions here: http://lukeberndt.com/2016/compiling-gnuradio-on-a-raspberry-pi/ I was able to use the pybombs method. Note that this required removing many of the packages I originally installed when following the wiki

petterreinholdtsen, 2017-08-12:
I just tested on my Raspberry Pi 2 model B, logged in using ssh -Y (no screen attached), and got this:

```
pi@radiotestpi:~ $ grgsm_livemon 
Warning: failed to XInitThreads()
linux; GNU C++ version 6.2.0 20161010; Boost_106100; UHD_003.009.005-0-unknown

gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.10
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy soapy redpitaya 
*** Error in `/usr/bin/python': malloc(): smallbin double linked list corrupted: 0x018753c8 ***
Aborted
pi@radiotestpi:~ $
```

I'm using the gr-gsm source for the Debian package I am working on, and the gnuradio package from Debian.  This is with gnuradion on Debian Stretch.  Running 'valgrind grgsm_livemon' show several warnings and start like this (cut the list as it was very long):

```
pi@radiotestpi:~ $ valgrind grgsm_livemon   
==16261== Memcheck, a memory error detector
==16261== Copyright (C) 2002-2011, and GNU GPL'd, by Julian Seward et al.
==16261== Using Valgrind-3.7.0 and LibVEX; rerun with -h for copyright info
==16261== Command: /usr/bin/grgsm_livemon
==16261== 
--16261-- WARNING: Serious error when reading debug info
--16261-- When reading debug info from /lib/arm-linux-gnueabihf/ld-2.24.so:
--16261-- Ignoring non-Dwarf2/3/4 block in .debug_info
--16261-- WARNING: Serious error when reading debug info
--16261-- When reading debug info from /lib/arm-linux-gnueabihf/ld-2.24.so:
--16261-- Last block truncated in .debug_info; ignoring
==16261== Conditional jump or move depends on uninitialised value(s)
==16261==    at 0x401B230: index (in /lib/arm-linux-gnueabihf/ld-2.24.so)
==16261== 
==16261== Conditional jump or move depends on uninitialised value(s)
==16261==    at 0x401B234: index (in /lib/arm-linux-gnueabihf/ld-2.24.so)
==16261== 
==16261== Conditional jump or move depends on uninitialised value(s)
==16261==    at 0x4008B8C: expand_dynamic_string_token (in /lib/arm-linux-gnueabihf/ld-2.24.so)
==16261== 
==16261== Conditional jump or move depends on uninitialised value(s)
==16261==    at 0x401B6E0: strlen (in /lib/arm-linux-gnueabihf/ld-2.24.so)
==16261== 
==16261== Conditional jump or move depends on uninitialised value(s)
==16261==    at 0x401B6E4: strlen (in /lib/arm-linux-gnueabihf/ld-2.24.so)
==16261== 
==16261== Conditional jump or move depends on uninitialised value(s)
==16261==    at 0x40194AC: __libc_memalign (in /lib/arm-linux-gnueabihf/ld-2.24.so)
==16261== 
==16261== Conditional jump or move depends on uninitialised value(s)
==16261==    at 0x40194B8: __libc_memalign (in /lib/arm-linux-gnueabihf/ld-2.24.so)
==16261== 
==16261== Conditional jump or move depends on uninitialised value(s)
==16261==    at 0x401C0F8: memcpy (in /lib/arm-linux-gnueabihf/ld-2.24.so)
==16261== 
==16261== Conditional jump or move depends on uninitialised value(s)
==16261==    at 0x401C118: memcpy (in /lib/arm-linux-gnueabihf/ld-2.24.so)
==16261== 
==16261== Conditional jump or move depends on uninitialised value(s)
==16261==    at 0x401C128: memcpy (in /lib/arm-linux-gnueabihf/ld-2.24.so)
==16261== 
```

I'm not sure the results are accurate, given that the problem is claimed to be in ld.so...

petterreinholdtsen, 2017-08-24:
Note, I was using Raspbian (not Debian) above.  When I tested with binaries from Debian Stretch, grgsm_livemon worked.  So I guess the problem is in Raspbian, not grgsm.

tomturner123, 2017-08-24:
Im pretty sure its a compatibility thing with gnuradio. It worked once i installed it from source. Great to know that the stretch binaries work

imranrajjad, 2018-03-20:
can I get the link to the Debian Stretch? I am having same problems

ptrkrysik, 2026-11-09:
This was caused by an outdated GNU Radio 3.7 package on Raspbian and has been solved by installing a newer GNU Radio; the current release targets GNU Radio 3.10, so please open a new issue if you see problems with it on a Raspberry Pi.

I am closing this as part of a clean-up for the new release (v0.43.0), which ports gr-gsm to GNU Radio 3.10 (plain CMake build, no SWIG, installation steps in the README). If the problem still occurs with it, please open a new issue with your operating system, the output of `gnuradio-config-info --version`, how gr-gsm and its dependencies were installed, the exact command and the complete output.

closed 2026-11-09 by ptrkrysik