ptrkrysik/gr-gsm #268: grgsm_decode non-deterministic output

nikosvir, 2017-03-13:
Hello,

grgsm_decode does not always produce the same output for the same burst file. e.g.

grgsm_decode -c testburst -a 33 -m SDCCH8 -t 1 -v -s 1M > 1

grgsm_decode -c testburst -a 33 -m SDCCH8 -t 1 -v -s 1M > 2

diff 1 2
30a31,32
> 2383642 3680398:  03 84 0d 06 35 01 2b 51 f8 3c 42 91 f1 ac 44 7d f3 61 85 e8 63 0e a9
> 2383646 3680530:  3f 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b
41a44
> 2384360 3682564:  3f 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b
44c47
< 2384423 3682959:  03 20 0d 05 18 01 2b e3 11 89 cc 66 d5 ea df 0f 18 04 0a 83 47 21 f2
---
> 2384513 3682561:  3f 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b
52,53d54
< 2384921 3682579:  3f 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b
< 2384972 3682578:  3f 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b

This is the case for multiple burst files. 

Cheers

romankh, 2017-03-14:
Using option `-c` indicates that you are using cfile format, not burst format file. Cfile is kind of raw signal data, where burst file are pre-processed bursts. 

Is the output of the diff command you provided complete ?
The order of the output is not guaranteed.

nikosvir, 2017-03-14:
Hi,

Yeap, my bad. Most of the time I work with burst files but in the example above I was using a cfile.

Yes, the diff output is complete.

I noticed it because sometimes grgsm decodes 2 Cipher Mode Commands and other times only 1, for the same cfile, which is weird.

Do you want me to upload it somewhere so you can take a look?

ptrkrysik, 2017-03-14:
The cause of non-deterministic output is that asynchronous message
passing is used in frequency offset correction loop.
If you require deterministic output in grgsm_decode you can disable
automatic frequency offset correction by not giving carrier frequency
(--fc parameter shouldn't be used). If necessary you can manually supply
frequency offset correction with --ppm option.

nikosvir, 2017-03-17:
Thanks for the clarification!

closed 2017-03-17 by nikosvir