ptrkrysik/gr-gsm #269: grgsm_livemon -f : hackrf_set_freq has failed  HACKRF_ERROR_LIBUSB

duchengyao, 2017-03-16:
I have finished install gr-gsm and run grgsm_livemon well
But when I using `-f` to set frequency, crashed:`RuntimeError: hackrf_set_freq(-399053) has failed (-1000) HACKRF_ERROR_LIBUSB`

The full log is:
```
root@kali:/usr/local/bin# grgsm_livemon -f 947.4
linux; GNU C++ version 6.2.0 20161103; Boost_106200; UHD_003.009.005-0-unknown

gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.10
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy soapy redpitaya 
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for 4294967295, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for 4294967295, skipping unlock
Number of USB devices: 8
USB device 1d50:6089: 0000000000000000909864c8283593cf match
Using HackRF One with firmware 2015.07.2 
Traceback (most recent call last):
  File "/usr/local/bin/grgsm_livemon", line 321, in <module>
    main()
  File "/usr/local/bin/grgsm_livemon", line 309, in main
    tb = top_block_cls(args=options.args, fc=options.fc, gain=options.gain, ppm=options.ppm, samp_rate=options.samp_rate, shiftoff=options.shiftoff, osr=options.osr)
  File "/usr/local/bin/grgsm_livemon", line 96, in __init__
    self.rtlsdr_source_0.set_center_freq(fc_slider-shiftoff, 0)
  File "/usr/lib/python2.7/dist-packages/osmosdr/osmosdr_swig.py", line 1688, in set_center_freq
    return _osmosdr_swig.source_sptr_set_center_freq(self, freq, chan)
RuntimeError: hackrf_set_freq(-399053) has failed (-1000) HACKRF_ERROR_LIBUSB
```

duchengyao, 2017-03-17:
I have found the reason.

`grgsm_livemon -f 947.4` crashed
`grgsm_livemon -f 947400000`  and hackrf worked!

closed 2017-03-17 by duchengyao

ptrkrysik, 2017-03-21:
W dniu 17.03.2017 o 03:37, S1NH pisze:
> I have found the reason.
> 
> |grgsm_livemon -f 947.4| crashed
> |grgsm_livemon -f 947400000| and hackrf worked!
> 

947.4 is 947.4 Hz. You probably meant 947MHz (so you should give 947e6
or 947M).