ptrkrysik/gr-gsm #270: wrong uplink frame number in receiver_impl.cc

bdyaods, 2017-03-20:
Dear all:
I'm working with a sample data with uplink channels, which the data is in timeslot 7. when I try to decode the uplink channels, I can't got the complete data frames. I found it seems that some uplink frame numbers are wrong, so some data was lost. and I try to modify the line 846 of receiver_impl.cc , I can got more data. This problem is with time slot 5, 6 and 7, the others have no problem.
for example:
frame_nr=1323419 timeslot=7
frame_nr=1323420 timeslot=7
frame_nr=1323421 timeslot=7
frame_nr=1323422 timeslot=7
frame_nr=1323423 timeslot=7
frame_nr=1323424 timeslot=7
frame_nr=1323629 timeslot=7  // frame_nr should be 1323425
frame_nr=1323426 timeslot=7
frame_nr=1323427 timeslot=7
frame_nr=1323428 timeslot=7
frame_nr=1323440 timeslot=7

My modified code of line 846 is:
if ( tap_header->timeslot >= 5 )
            tap_header->frame_number = htobe32(d_burst_nr.get_frame_nr()-1);
        else
            tap_header->frame_number = htobe32(d_burst_nr.subtract_timeslots(3).get_frame_nr());
But I think it is not the appreciate way.  how can I resolve this problem?
I have the data file, but it is too big to be uploaded.

E3V3A, 2017-06-21:
@bdyaods 
What's your setup and how's the "data" collected?

E3V3A, 2017-07-05:
@bdyaods We need to know how you collected the above data in order to reproduce the error.

@ptrkrysik Has this perhaps been fixed since? If so please close.

bdyaods, 2017-07-13:
we have fixed this problem. It is because of the advanced 3 timeslots of downlink and uplink channel. result in the frame number of the last three timeslots was wrong.
we modified a function in  /lib/receiver/receiver_config.cc:
burst_counter burst_counter::subtract_timeslots(unsigned int number_of_timeslots)
{
  int timeslot_nr = (int)d_timeslot_nr - (int)number_of_timeslots;
  int t1,t2,t3;
  if (timeslot_nr < 0) {
    timeslot_nr = timeslot_nr + 8;

    t2 = (d_t2+26 - 1) % 26;
    t3 = (d_t3+51 - 1) % 51;

    if ((d_t2 == 0) && (d_t3 == 0)) {
      t1 = (d_t1 - 1) % (1 << 11);
    } else
    {
      t1 = d_t1;
    }
  }
  else
  {
    t1 = d_t1;
    t2 = d_t2;
    t3 = d_t3;
  }
  
  return burst_counter(d_OSR, t1, t2, t3, timeslot_nr);
}

bdyaods, 2017-07-17:
@ptrkrysik

ptrkrysik, 2017-07-23:
@bdyaods , thanks for the information. I will look at this.

ptrkrysik, 2017-07-23:
@bdyaods, you changed only these two lines:
```
t2 = (d_t2+26 - 1) % 26;
t3 = (d_t3+51 - 1) % 51;
```
by adding 26 and 51. I think it doesn't change too much and this part isn't dependent on timeslot number. If there are incorrect frame numbers depending on timeslot number, this is not the solution.

ptrkrysik, 2017-07-23:
@bdyaods, this whole 'if (timeslot_nr < 0)' is for the situation when timeslot number after subtracting some number is lower than 0. So for subtracted value 3 it timeslots 0, 1 and 2 that are these three for which you should see difference.

I'm a bit confused by this.

ptrkrysik, 2017-07-23:
Ok, I understand now that value 7 of timeslot was equal to 2 before subtracting 3. I still don't know why the change you made solved the problem for you.

ptrkrysik, 2017-07-23:
Might have something to do with d_t2 and d_t3 being unsigned ;).

ptrkrysik, 2017-07-23:
@bdyaods, many thanks for reporting and solving this problem. Your solution is correct. The problem was that I assumed that C++ modulo operator works for negative values in the same way as for modulo operator in Python and mod function in Octave/Matlab. But this is not the case.

closed 2017-07-23 by ptrkrysik