ptrkrysik/gr-gsm #273: grgsm_scanner does not show any output

ars0, 2017-03-29:
Hi, 

I've installed gr-gsm using the pybombs on Ubuntu 16.04, and grgsm_livemon works, kalibrate-sdr finds GSM base stations. The only thing that does not work is grgsm_scanner: it shows only one line, then it hangs for a minute or so (probably scanning), and then it exits without any output.

It looks like the similar issue: https://github.com/ptrkrysik/gr-gsm/issues/266

Not sure how to debug it: "-v"  command line switch does not add anything to the output: 
`$ grgsm_scanner -p -24 -v -g 30
[INFO] [UHD] linux; GNU C++ version 5.4.0 20160609; Boost_105800; UHD_3.11.0.git-94-g5964adcd
$`

Not sure if it's related, it seems to ignore unknown options that do not start with a "-" (dash):
`$ grgsm_scanner -v -g 20 -p -24 OPTION_IS_IGNORED
[INFO] [UHD] linux; GNU C++ version 5.4.0 20160609; Boost_105800; UHD_3.11.0.git-94-g5964adcd
 `

rojavatechops, 2017-04-10:
Same problem here.

ptrkrysik, 2017-04-10:
W dniu 10.04.2017 o 20:35, rojavatechops pisze:
> Same problem here.
> 
Did you download and compile the newest version of gr-gsm?

SirCosty, 2017-07-05:
Same grgsm_scanner crash after few seconds 
Have installed the last version of gr-gsm folowing the instruction from here
https://github.com/ptrkrysik/gr-gsm/wiki/Installation
The error that I get is the next

*** Error in `/usr/bin/python2': double free or corruption (!prev): 0x000000000350e210 ***
======= Backtrace: =========
/lib/x86_64-linux-gnu/libc.so.6(+0x777e5)[0x7f16ec4237e5]
/lib/x86_64-linux-gnu/libc.so.6(+0x8037a)[0x7f16ec42c37a]
/lib/x86_64-linux-gnu/libc.so.6(cfree+0x4c)[0x7f16ec43053c]
/usr/local/lib/libgnuradio-runtime-3.7.12git.so.0.0.0(+0x76f1a)[0x7f16ea246f1a]
/lib/x86_64-linux-gnu/libc.so.6(+0x39ff8)[0x7f16ec3e5ff8]
/lib/x86_64-linux-gnu/libc.so.6(+0x3a045)[0x7f16ec3e6045]
/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf7)[0x7f16ec3cc837]
/usr/bin/python2(_start+0x29)[0x49d9d9]
======= Memory map: ========
00400000-006ea000 r-xp 00000000 08:02 10487002                           /usr/bin/python2.7
008e9000-008eb000 r--p 002e9000 08:02 10487002                           /usr/bin/python2.7
008eb000-00962000 rw-p 002eb000 08:02 10487002                           /usr/bin/python2.7
00962000-00985000 rw-p 00000000 00:00 0 
026ab000-0506b000 rw-p 00000000 00:00 0                                  [heap]

E3V3A, 2017-08-02:
@SirCosty Seem you have a problem with Python2, are you using the latest? Using PIP etc? What does your versions say? 

@ars0 Have you resolved the issue? (Please update or close.) Please try manual installation if PyBombs doesn't work... I've got all this to work on Kali, Mint and Ubuntu etc. perhaps you're missing some dependency. Please note that different distro's often use slightly different names for their dependencies.

Y2Kot, 2017-08-04:
any solutions?

velichkov, 2017-08-04:
Hi @Y2Kot,

Comment out lines (347,348) in grgsm_scanner

346             # put /dev/null fds on 1 and 2
347             os.dup2(null_fds[0], 1)
348             os.dup2(null_fds[1], 2)

and post the output

SirCosty, 2017-08-10:
**# @E3V3A  To make me very clear about the versions and other stuffs. 
      Today 10 of August of 2017**

1. Have formatted  the laptop HDD and clean installed again the Ubuntu 16.04 LTS

2. Have updated and upgraded the packages 
```
        sudo apt update
        sudo apt upgrade
```

3. Have  installed the last version of _gr-gsm_ following the official instruction from your own page
https://github.com/ptrkrysik/gr-gsm/wiki/Installation
```
        sudo apt-get purge --auto-remove gqrx
        sudo apt-get purge --auto-remove gqrx-sdr
        sudo apt-get purge --auto-remove libgnuradio*
        
        sudo add-apt-repository -y ppa:bladerf/bladerf
        sudo add-apt-repository -y ppa:ettusresearch/uhd
        sudo add-apt-repository -y ppa:myriadrf/drivers
        sudo add-apt-repository -y ppa:myriadrf/gnuradio
        sudo add-apt-repository -y ppa:gqrx/gqrx-sdr
        sudo apt-get update
        
        sudo apt-get -y install gqrx-sdr
        
        sudo apt-get install libvolk1-bin
        sudo volk_profile
```

4. Finally have tested the _grgsm_scanner first without sudo but just exits with no feedback error or some
Then have used the sudo but crash after 20 seconds and here is the result
```
ubuntu@notebook:~$ grgsm_scanner

ubuntu@notebook:~$ sudo grgsm_scanner

*** Error in `/usr/bin/python2': double free or corruption (!prev): 0x000000000320a650 ***
======= Backtrace: =========
/lib/x86_64-linux-gnu/libc.so.6(+0x777e5)[0x7fafed4027e5]
/lib/x86_64-linux-gnu/libc.so.6(+0x8037a)[0x7fafed40b37a]
/lib/x86_64-linux-gnu/libc.so.6(cfree+0x4c)[0x7fafed40f53c]
/usr/local/lib/libgnuradio-runtime-3.7.12git.so.0.0.0(+0x76f1a)[0x7fafeb223f1a]
/usr/local/lib/libgnuradio-runtime-3.7.12git.so.0.0.0(+0x76eec)[0x7fafeb223eec]
/usr/local/lib/libgnuradio-runtime-3.7.12git.so.0.0.0(+0x76eec)[0x7fafeb223eec]
/lib/x86_64-linux-gnu/libc.so.6(+0x39ff8)[0x7fafed3c4ff8]
/lib/x86_64-linux-gnu/libc.so.6(+0x3a045)[0x7fafed3c5045]
/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf7)[0x7fafed3ab837]
/usr/bin/python2(_start+0x29)[0x49d9d9]
======= Memory map: ========
00400000-006ea000 r-xp 00000000 08:03 4850906                            /usr/bin/python2.7
008e9000-008eb000 r--p 002e9000 08:03 4850906                            /usr/bin/python2.7
008eb000-00962000 rw-p 002eb000 08:03 4850906                            /usr/bin/python2.7
00962000-00985000 rw-p 00000000 00:00 0 
023e4000-04da0000 rw-p 00000000 00:00 0                                  [heap]
```

5. Now here will execute more commands for you to see the version of all is here
```
ubuntu@notebook:~$ cat /etc/*release
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=16.04
DISTRIB_CODENAME=xenial
DISTRIB_DESCRIPTION="Ubuntu 16.04.3 LTS"
NAME="Ubuntu"
VERSION="16.04.3 LTS (Xenial Xerus)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 16.04.3 LTS"
VERSION_ID="16.04"
HOME_URL="http://www.ubuntu.com/"
SUPPORT_URL="http://help.ubuntu.com/"
BUG_REPORT_URL="http://bugs.launchpad.net/ubuntu/"
VERSION_CODENAME=xenial
UBUNTU_CODENAME=xenial

ubuntu@notebook:~$ python --version
Python 2.7.12

ubuntu@notebook:~$ pip -V
pip 8.1.1 from /usr/lib/python2.7/dist-packages (python 2.7)

ubuntu@notebook:~$ pybombs --version
2.3.1

ubuntu@notebook:~$ lsusb
Bus 002 Device 003: ID 1d50:6066 OpenMoko, Inc. 

ubuntu@notebook:~$ bladeRF-cli -i

bladeRF> version

  bladeRF-cli version:        1.5.0-git-6116d09
  libbladeRF version:         1.8.0-git-6116d09

  Firmware version:           1.9.1
  FPGA version:                0.6.0

bladeRF> info

  Serial #:                              246af9d79ca5b88349dc25fa8978e93d
  VCTCXO DAC calibration:   0x8bc9
  FPGA size:                          40 KLE
  FPGA loaded:                     yes
  USB bus:                            2
  USB address:                      3
  USB speed:                         SuperSpeed
  Backend:                             libusb
  Instance:                             0

ubuntu@ubuntu-security:~$  gnuradio-config-info --version
3.7.12git-218-g811bee8c

ubuntu@ubuntu-security:~$ gnuradio-companion --version
GNU Radio Companion 3.7.12git-218-g811bee8c
```

6. At the end I think anyone following the same steps should have the same problem
Have made it all clear and clean but still doesn't work

E3V3A, 2017-08-16:
@SirCosty I never use PyBOMBS as it never worked very well while leaving you clueless as to what it does and how/where things were installed. Clearly from your output, something works and perhaps it just a matter of some permission problems or outdated libraries. Do a manual install, and save a log.

fipso, 2018-03-25:
any updates on this ? same problem

velichkov, 2018-03-25:
Hi @fipso,

Try to comment out the following in two lines in `grgsm_scanner`, run it and post the full output
```python
            # put /dev/null fds on 1 and 2
            os.dup2(null_fds[0], 1)
            os.dup2(null_fds[1], 2)
```

fipso, 2018-03-25:
i get some type of restart loop.

```
set_min_output_buffer on block 399 to 4000000
Args= 
gr-osmosdr v0.1.4-98-gc653754d (0.1.5git) gnuradio 3.7.12git-377-g273392bd
built-in source types: file osmosdr fcd rtl rtl_tcp uhd hackrf bladerf rfspace airspy soapy redpitaya 
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Exact sample rate is: 2000000.052982 Hz
[R82XX] PLL not locked!
set_min_output_buffer on block 530 to 4000000
set_min_output_buffer on block 532 to 4000000
set_min_output_buffer on block 531 to 4000000
Args= 
gr-osmosdr v0.1.4-98-gc653754d (0.1.5git) gnuradio 3.7.12git-377-g273392bd
built-in source types: file osmosdr fcd rtl rtl_tcp uhd hackrf bladerf rfspace airspy soapy redpitaya 
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Exact sample rate is: 2000000.052982 Hz
[R82XX] PLL not locked!
set_min_output_buffer on block 662 to 4000000
set_min_output_buffer on block 664 to 4000000
set_min_output_buffer on block 663 to 4000000
Args= 
gr-osmosdr v0.1.4-98-gc653754d (0.1.5git) gnuradio 3.7.12git-377-g273392bd
built-in source types: file osmosdr fcd rtl rtl_tcp uhd hackrf bladerf rfspace airspy soapy redpitaya 
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Exact sample rate is: 2000000.052982 Hz
[R82XX] PLL not locked!
set_min_output_buffer on block 794 to 4000000
set_min_output_buffer on block 796 to 4000000
set_min_output_buffer on block 795 to 4000000
Args= 
gr-osmosdr v0.1.4-98-gc653754d (0.1.5git) gnuradio 3.7.12git-377-g273392bd
built-in source types: file osmosdr fcd rtl rtl_tcp uhd hackrf bladerf rfspace airspy soapy redpitaya 
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Exact sample rate is: 2000000.052982 Hz
[R82XX] PLL not locked!
set_min_output_buffer on block 926 to 4000000
set_min_output_buffer on block 928 to 4000000
set_min_output_buffer on block 927 to 4000000
Args= 
gr-osmosdr v0.1.4-98-gc653754d (0.1.5git) gnuradio 3.7.12git-377-g273392bd
built-in source types: file osmosdr fcd rtl rtl_tcp uhd hackrf bladerf rfspace airspy soapy redpitaya 
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Exact sample rate is: 2000000.052982 Hz
[R82XX] PLL not locked!
set_min_output_buffer on block 1058 to 4000000
set_min_output_buffer on block 1060 to 4000000
set_min_output_buffer on block 1059 to 4000000
Args= 
gr-osmosdr v0.1.4-98-gc653754d (0.1.5git) gnuradio 3.7.12git-377-g273392bd
built-in source types: file osmosdr fcd rtl rtl_tcp uhd hackrf bladerf rfspace airspy soapy redpitaya 
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Exact sample rate is: 2000000.052982 Hz
[R82XX] PLL not locked!
set_min_output_buffer on block 1190 to 4000000
set_min_output_buffer on block 1192 to 4000000
set_min_output_buffer on block 1191 to 4000000
Args= 
gr-osmosdr v0.1.4-98-gc653754d (0.1.5git) gnuradio 3.7.12git-377-g273392bd
built-in source types: file osmosdr fcd rtl rtl_tcp uhd hackrf bladerf rfspace airspy soapy redpitaya 
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
```

velichkov, 2018-03-25:
Hi @fipso,

> i get some type of restart loop.

That's normal as it loops over all frequencies in the given band. 

> Args= 
gr-osmosdr v0.1.4-98-gc653754d (0.1.5git) gnuradio 3.7.12git-377-g273392bd
built-in source types: file osmosdr fcd rtl rtl_tcp uhd hackrf bladerf rfspace airspy soapy redpitaya 
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Exact sample rate is: 2000000.052982 Hz
[R82XX] PLL not locked!
set_min_output_buffer on block 530 to 4000000
set_min_output_buffer on block 532 to 4000000
set_min_output_buffer on block 531 to 4000000

The output looks normal. Try to specify a different band (-b DCS1800) and experiment with different gain values (-g 40)
```
grgsm_scanner -b DCS1800 -g 40
```
See also #368

fipso, 2018-03-25:
i want to find gsm stations. what args do i have to use ?

velichkov, 2018-03-25:
> i want to find gsm stations. what args do i have to use ?

For me it works without any args but it depends on where you live and what SDR you have

fipso, 2018-03-25:
germany. i am using an rtl2838 dvbt usb

velichkov, 2018-03-25:
> germany. 

For Germany the correct bands are GSM900 (the default one) and DCS1800 (you need to specify it on the command line as show in my previous post)

> i am using an rtl2838 dvbt usb

Do you have an external antenna attached to the device?
You could try also  [kalibrate-rtl](http://github.com/steve-m/kalibrate-rtl)

KR0SIV, 2018-05-01:
I have the same corruption issue causing the application to crash before it can complete a scan. Although I'm trying to scan PCS1900.

Originally used pybombs, and also tried the pre-compiled binary.. both fail the same way. Compiling by hand also produces the same result and I'm on a fresh debian installation to boot.

velichkov, 2018-05-01:
Hi @KR0SIV,

> fresh debian installation to boot.

Which version? And what SDR do you use?
If you are using HackRF see https://github.com/ptrkrysik/gr-gsm/issues/336#issuecomment-332957714
Also provide the full output attached as a text file.

jsa91, 2018-06-23:
Having the same problem. It occurred after I was forced to comment out line 64 in grgsm_livemon to make it run. I have instead tried with "kalibrate-rtl" and two different SDR/antennas and it works like a charm. But as soon as I run grgsm_scanner the output is blank.

velichkov, 2018-06-23:
Hi @jsa91,

Which Linux distribution and version do you use and what SDR?

> It occurred after I was forced to comment out line 64 in grgsm_livemon to make it run

What problem did you solved by commenting this line?

> But as soon as I run grgsm_scanner the output is blank.

Comment out lines (347,348) in grgsm_scanner
```
346 # put /dev/null fds on 1 and 2
347 os.dup2(null_fds[0], 1)
348 os.dup2(null_fds[1], 2)
```
build, install, test and post the output.

jsa91, 2018-06-23:
@velichkov 

Im using 

> Linux kali 4.16.0-kali2-amd64 #1 SMP Debian 4.16.12-1kali1 (2018-05-28) x86_64 GNU/Linux

and the NooElec NESDR SMArt.

The reason i had to comment out line 64 in grgsm_livemon was because of the following problem:

AttributeError: 'module' object has no attribute 'QWidget'

When i comment out lines (347,348) in grgsm_scanner the following occur:

```
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Detached kernel driver
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Exact sample rate is: 2000000.052982 Hz
[R82XX] PLL not locked!
Allocating 15 zero-copy buffers
OOOOOOOOOOOOOOOOOOARFCN:   82, Freq:  951.4M, CID: 41152, LAC:  2013, MCC: 240, MNC:   1, Pwr: -27
ARFCN:   84, Freq:  951.8M, CID: 41972, LAC:  2013, MCC: 240, MNC:   1, Pwr: -31
ARFCN:   85, Freq:  952.0M, CID:     0, LAC:  2013, MCC: 240, MNC:   1, Pwr: -33
ARFCN:   87, Freq:  952.4M, CID:     0, LAC:     0, MCC:   0, MNC:   0, Pwr: -50
ARFCN:   88, Freq:  952.6M, CID: 41920, LAC:  2013, MCC: 240, MNC:   1, Pwr: -43
Reattached kernel driver
Args= 
gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.13.2
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Detached kernel driver
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Exact sample rate is: 2000000.052982 Hz
[R82XX] PLL not locked!
Allocating 15 zero-copy buffers
OOOOOOOOOOOOOOOARFCN:   91, Freq:  953.2M, CID:     0, LAC:  2013, MCC: 240, MNC:   1, Pwr: -28
ARFCN:   92, Freq:  953.4M, CID: 46462, LAC:  2013, MCC: 240, MNC:   1, Pwr: -27
Reattached kernel driver
Args= 
gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.13.2
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Detached kernel driver
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Exact sample rate is: 2000000.052982 Hz
[R82XX] PLL not locked!
Allocating 15 zero-copy buffers
Reattached kernel driver
Args= 
gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.13.2
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Detached kernel driver
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Exact sample rate is: 2000000.052982 Hz
[R82XX] PLL not locked!
Allocating 15 zero-copy buffers
Reattached kernel driver
Args= 
gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.13.2
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Detached kernel driver
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Exact sample rate is: 2000000.052982 Hz
[R82XX] PLL not locked!
Allocating 15 zero-copy buffers
Reattached kernel driver
```

As you can see I suddenly get some GSM BS output, but the software repeats it self over and over again which is pretty annoying.

velichkov, 2018-06-24:
Hi @jsa91,

> Allocating 15 zero-copy buffers
OOOOOOOOOOOOOOOARFCN:   91, Freq:  953.2M, CID:     0, LAC:  2013, MCC: 240, MNC:   1, Pwr: -28

The several Os here indicates buffer overflows which means that grgsm_scanner does not read fast enough from the SDR buffer. Recently a change was applied (ea44d98558eee1751e77f02b7bb46f3f1ae269b4) that prevents a crash when used with HackRF, but this change could cause performance problems.  Try to revert it or if you are not familiar with git just uncomment line 212 in grgsm_scanner.

> As you can see I suddenly get some GSM BS output, but the software repeats it self over and over again which is pretty annoying.

It's normal to repeats itself as it scans all frequencies in a loop and the output was redirected to /dev/null because it is annoying but we need it to debug the issue. Once we solve it you could uncomment lines 347 and 348.

jsa91, 2018-06-24:
Thank you very much for your help!

cornacchia, 2018-07-13:
I had this problem today and it looks like these lines do not work properly:

```          
# restore file descriptors so we can print the results
os.dup2(save[0], 1)
os.dup2(save[1], 2)
# close the temporary fds
os.close(null_fds[0])
os.close(null_fds[1])

for info in sorted(found_list):
  print info
  if options.verbose:
    print info.get_verbose_info()
```
It looks like the `print` function doesn't get the correct file descriptor after the `os.dup2` operations.

I tried also with `sys.stdout.write` with no result.

Finally I came out with this solution which effectively prints output to stdout:

```
for info in sorted(found_list):
  # Explicitly write to stdout
  os.write(1, str(info) + '/n')
  if options.verbose:
    os.write(1, str(info.get_verbose_info()) + '/n')
```

mishaturnbull, 2018-12-26:
I am having this issue as well, I think.

Kali Linux rolling (Debian 4.18.20), RTL-SDR v3 with an RTL2832U DBV-T:

```bash
$ grgsm_scanner

[prints one newline, then stops forever] ^C^Z
[1]+ Stopped grgsm_scanner
```
```bash
$ sudo grgsm_scanner
[sudo] password for misha:

[prints one newline, then stops forever] ^C^Z
[1]+ Stopped sudo grgsm_scanner
```
```bash
$ grgsm_scanner --help
Usage: grgsm_scanner: [options]

Options:
  -h, --help            show this help message and exit
  -b BAND, --band=BAND  Specify the GSM band for the frequency. Available
                        bands are: GSM900, DCS1800, GSM850, PCS1900, GSM450,
                        GSM480, GSM-R
  -s SAMP_RATE, --samp-rate=SAMP_RATE
                        Set sample rate [default=2000000.0] - allowed values
                        even_number*0.2e6
  -p PPM, --ppm=PPM     Set frequency correction in ppm [default=0]
  -g GAIN, --gain=GAIN  Set gain [default=24.0]
  --args=ARGS           Set device arguments [default=]
  --speed=SPEED         Scan speed [default=4]. Value range 0-5.
  -v, --verbose         If set, verbose information output is printed: ccch
                        configuration, cell ARFCN's, neighbour ARFCN's
```

`grgsm_scanner` (and accompanying `grgsm_livemon, grgsm_capture, grgsm_channelize, grgsm_decode, grgsm_livemon_headless, grgsm_trx`) were installed with Aptitude, not PyBOMBS or source.  Both PyBOMBS and source failed to compile.  I have not been able to reproduce any of the memory map-outputting errors listed here.

It may also be interesting to note that I am not able to kill the process with SIGINT (i.e. `CTRL+C`).  SIGTSTP (`CTRL+Z`) works, but is not *really* a kill signal.

velichkov, 2018-12-27:
Hi @mishaturnbull,

> Both PyBOMBS and source failed to compile.

Could you post the exact error message when compiling manually from source (not with pybombs)

Try to comment out the following two lines in `/usr/bin/grgsm_scanner` and post the full output
```
320             os.dup2(null_fds[0], 1)
321             os.dup2(null_fds[1], 2)
```
Unfortunately the version with `--debug` flag is still not available as an apt package.

mishaturnbull, 2018-12-28:
Hi @velichkov,

Commenting out those two lines worked perfectly.  I'd be happy to open a PR if there's a good enough reason to make that change permanent and for everyone.

When compiling from source, I followed the instructions [at osmocom.org](https://osmocom.org/projects/gr-gsm/wiki/Installation).

Interestingly, I got a different error this time than I did before the successful `apt` install.  Previously, I was getting something like:
```bash
root@slippery-weasel:~/gr-gsm/build# make
[lots of output]
[ 89%] Message goes here, I don't remember what it was
Segmentation fault
root@slippery-weasel:~/gr-gsm/build#
```
This is the (relevant, [full pastebin here](https://pastebin.com/7BqntzM7), part of the) output of attempting to compile from source just barely, following the same instructions:
```bash
root@slippery-weasel:~/gr-gsm/build# make
[lots of output]
[ 56%] Building CXX object lib/CMakeFiles/grgsm.dir/receiver/time_sample_ref.cc.o
[ 57%] Building CXX object lib/CMakeFiles/grgsm.dir/transmitter/txtime_setter_impl.cc.o
/root/gr-gsm/lib/transmitter/txtime_setter_impl.cc: In member function ‘void gr::gsm::txtime_setter_impl::process_txtime_of_burst(pmt::pmt_t)’:
/root/gr-gsm/lib/transmitter/txtime_setter_impl.cc:129:22: error: ‘format’ was not declared in this scope
                   << format("burst (fn=%u, tn=%u)", frame_nr, ts_num)
                      ^~~~~~
/root/gr-gsm/lib/transmitter/txtime_setter_impl.cc:129:22: note: suggested alternative:
In file included from /usr/include/boost/format.hpp:34,
                 from /usr/include/gnuradio/logger.h:104,
                 from /usr/include/gnuradio/block.h:29,
                 from /root/gr-gsm/include/grgsm/transmitter/txtime_setter.h:28,
                 from /root/gr-gsm/lib/transmitter/txtime_setter_impl.h:26,
                 from /root/gr-gsm/lib/transmitter/txtime_setter_impl.cc:32:
/usr/include/boost/format/format_fwd.hpp:27:37: note:   ‘boost::format’
     typedef basic_format<char >     format;
                                     ^~~~~~
make[2]: *** [lib/CMakeFiles/grgsm.dir/build.make:713: lib/CMakeFiles/grgsm.dir/transmitter/txtime_setter_impl.cc.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:353: lib/CMakeFiles/grgsm.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
root@slippery-weasel:~/gr-gsm/build# 
```

velichkov, 2018-12-28:
Hi @mishaturnbull,

> Commenting out those two lines worked perfectly. I'd be happy to open a PR if there's a good enough reason to make that change permanent and for everyone.

Commenting out those two lines is not meant to be a solution. I'm asking to do it in order to get some more information to be able to debug the problem. In fb1f65c572e210e222edef564c6e739bac3f7ec0 I've added a `--debug` flag. 

If you uncomment them does it continue to work?
Could you test the @cornacchia's modifications from https://github.com/ptrkrysik/gr-gsm/issues/273#issuecomment-404842126

> /root/gr-gsm/lib/transmitter/txtime_setter_impl.cc:129:22: error: ‘format’ was not declared in this scope
                   << format("burst (fn=%u, tn=%u)", frame_nr, ts_num)

This is a newly introduced error in dc342dc88ad378ab5dbe21d87659c46a66f3abeb, the [Travis build](https://travis-ci.org/ptrkrysik/gr-gsm/jobs/473083755#L4188-L4203) fails with the same error as well.

You could try building from the previous commit 
```
git checkout b73b4a86378929dc53153804dc7a8b2c0988cea4
make
```

Which Linux distribution and version do you use?

mishaturnbull, 2019-01-01:
Hi @velichkov,

I'm using Kali Linux Rolling, which is in turn built on Debian Testing:
```bash
root@slippery-weasel:~# uname -a
Linux slippery-weasel 4.18.0-kali3-amd64 #1 SMP Debian 4.18.20-2kali2 (2018-11-30) x86_64 GNU/Linux
```

If I uncomment those lines, the program again does not work (same symptoms as before -- no output).
Unfortunately, @cornacchia's modifications did not work.  There is still no output.

Compiling from the previous commit, b73b4a86378929dc53153804dc7a8b2c0988cea4, has a mix result.  Compilation [succeeds, but with a bunch of warnings](https://pastebin.com/jzBkdHXg); however there is still no program output.

velichkov, 2019-01-02:
Hi @mishaturnbull,

> If I uncomment those lines, the program again does not work (same symptoms as before -- no output).

Ok, the version you have installed from source should support the `--debug` flag so run `grgsm_scanner` with it, wait few minutes, copy the whole output to pastbin and provide a link.

> Compiling from the previous commit, b73b4a8, has a mix result. 

If you haven't removed the apt package please do so (`apt remove gr-gsm`) as having two versions installed at the same time could lead to various problems.

> Compilation succeeds, but with a bunch of warnings

>> Warning: Block with key "gsm_tmsi_dumper" already exists.
    Ignoring: /usr/local/share/gnuradio/grc/blocks/gsm_tmsi_dumper.xml

These warnings are somehow expected so please ignore them for now.

EnjoyDev, 2020-04-14:
Hi @velichkov,
I am currently trying to use grgsm with a hackrf on a Kali distrib running in a vmware machine unsuccessfully. I installed grgsm manually (without Pybombs) from maint-3.8 branch as my gnuradio is a 3.8. 
grgsm_scanner showed nothing so I tried with -debug option and here is what I got : 
<pre><font color="#EF2929"><b>root@Host-001</b></font>:<font color="#729FCF"><b>~</b></font># grgsm_scanner -d

Args= 
Traceback (most recent call last):
  File &quot;/usr/local/bin/grgsm_scanner&quot;, line 438, in &lt;module&gt;
    main()
  File &quot;/usr/local/bin/grgsm_scanner&quot;, line 434, in main
    do_scan(options.samp_rate, options.band, options.speed,
  File &quot;/usr/local/bin/grgsm_scanner&quot;, line 330, in do_scan
    scanner = wideband_scanner(rec_len=6 - speed,
  File &quot;/usr/local/bin/grgsm_scanner&quot;, line 212, in __init__
    str(grgsm.device.get_default_args(args)))
  File &quot;/usr/local/lib/python3/dist-packages/grgsm/device.py&quot;, line 55, in get_default_args
    raise RuntimeError(&quot;Unable to find any supported SDR devices&quot;)
RuntimeError: Unable to find any supported SDR devices
</pre>
hackrf_info:
<pre><font color="#EF2929"><b>root@Host-001</b></font>:<font color="#729FCF"><b>~</b></font># hackrf_info
hackrf_info version: git-43e6f99
libhackrf version: git-43e6f99 (0.5)
Found HackRF
Index: 0
Serial number: 000000000000000087c867dc2d60055f
Board ID Number: 2 (HackRF One)
Firmware Version: 2017.02.1 (API:1.02)
Part ID Number: 0xa000cb3c 0x00624f4c
</pre>
Kalibrate works fine 
<pre><font color="#EF2929"><b>root@Host-001</b></font>:<font color="#729FCF"><b>~</b></font># kal -s GSM900
kal: Scanning for GSM-900 base stations.
GSM-900:
	chan:  111 (957.2MHz + 23.621kHz)	power:  614832.97
	chan:  112 (957.4MHz + 7.588kHz)	power:  435761.93
	chan:  113 (957.6MHz - 17.806kHz)	power:  429908.77
</pre>
I also tried to upgrade my hackrf firmware with no more success

velichkov, 2020-04-15:
Hi @EnjoyDev,

>   File "/usr/local/lib/python3/dist-packages/grgsm/device.py", line 55, in get_default_args
>     raise RuntimeError("Unable to find any supported SDR devices")
> RuntimeError: Unable to find any supported SDR devices

Are you sure that gr-osmosdr is compiled with HackRF support? Does `osmocom_fft` works?

When you start grgsm_scanner with --debug (and osmocom_fft) in the terminal you should get some lines like
```
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.1.0
built-in source types: file rtl rtl_tcp uhd hackrf bladerf rfspace airspy soapy redpitaya 
```
Do you get `hackrf` in the `build-in source types`?

> Kalibrate works fine

Which kalibrate version and from which repository as there are quite a lot of forks out there? 

Kalibrate does not use gr-osmosdr internally but the `UHD: USRP Source` block or the `librtlsdr` library directly so I'm wondering how it works with HackRF.

Please provide the output of:
```
uhd_find_devices
SoapySDRUtil --find
```

EnjoyDev, 2020-04-15:
Thank you for your reactivity @velichkov,

>Are you sure that gr-osmosdr is compiled with HackRF support? Does osmocom_fft works?

my gr-osmosdr install was not supporting hackrf but it is fixed now (I did nothing more than reinstall it ; as I tried many manipulation of removing/reinstalling packages I cannot explain exactly why hackrf was not in built in list before).   

> Please provide the output of:
> uhd_find_devices
> SoapySDRUtil --find

<pre><font color="#EF2929"><b>root@Host-001</b></font>:<font color="#729FCF"><b>~</b></font># uhd_find_devices
<font color="#4E9A06">[INFO] [UHD] </font>linux; GNU C++ version 9.2.1 20200104; Boost_106700; UHD_3.15.0.0-2
No UHD Devices Found
</pre>
SoapySDR was not installed even if I followed instructions on https://github.com/pothosware/SoapySDR/wiki/BuildGuide (git clone then cmake/make/install). I had to use apt-get install soapysdr-tools to make it work.

Remark : if not specified --arg=hackrf this error appears : 
<pre><font color="#EF2929"><b>[ERROR] avahi_service_browser_new() failed: Bad state</b></font>
</pre>

> Which kalibrate version and from which repository as there are quite a lot of forks out there?
kalibrate v0.4.1-hackrf
https://github.com/scateu/kalibrate-hackrf.git

It seems to be alright now but I can't terminate scan with ctrl+c may be there is a keyboard combination to finish operation ?

velichkov, 2020-04-15:
> Remark : if not specified --arg=hackrf this error appears :
> 
> [ERROR] avahi_service_browser_new() failed: Bad state

That's normal, when you do not specify your device it tries to detect it and this results in various error and warning messages.
 
> > Which kalibrate version and from which repository as there are quite a lot of forks out there?
> > kalibrate v0.4.1-hackrf
> > https://github.com/scateu/kalibrate-hackrf.git

This fork of kalibrate uses the libhackrf library directly and this explains why it was working.

> It seems to be alright now but I can't terminate scan with ctrl+c may be there is a keyboard combination to finish operation ?

It's a known problem but I don't have solution for it. I usually press Ctrl+C multiple times until it finally stops.

Yacinetel, 2020-05-19:
Hi @velichkov

i build gr-gsm on ubuntu 16.04, but grgsm_scanner give me this erroe

 grgsm_scanner -b GSM900 -g 40
Segmentation fault (core dumped)

i have uninstall gnuradio and gr-osmocom  and re-build a gain but the same error
could you help

nbalkanas, 2020-10-11:
Commenting on the no output (not the crash). These are my 3 points:
1) os.dup2 in python, doesn't work as it should (and does in C). This can be fixed by using file streams instead of fds:
import sys
save = sys.stdout, sys.stderr
nullfs = open(os.nulldev, "a")
sys.stdout = nullfs
...
sys.stdout = save[0]
nullfs.close()
This will take care of correct disabling and reenabling stdout and stderr. However, there is another issue:
2) rtlsdr_source is instanced at gr-osmosdr. Osmo seems to override and reset stdout, so even if you do (1) you still get these annoying output, but you miss on all other prints (Args, etc) until you reset stdout:(
3) I don't do much python, I am a C programmer. In C I would configure only once rtlsdr, then just change the carrier frequency for every loop. That would save much time reinitializing the device and printing all that output in each loop:) 
You can still do the bulk initialization once from gr-osmosdr, and then instance just the frequency change for each device:)

mggyi, 2021-03-21:
> os.dup2(null_fds[0], 1)

Args= 
[INFO] [UHD] linux; GNU C++ version 10.2.1 20201207; Boost_107400; UHD_3.15.0.0-4+b1
Found Rafael Micro R820T tuner
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.2.0
built-in source types: file fcd rtl rtl_tcp uhd hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
Using device #0 Generic RTL2832U SN: 77771111153705700
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Exact sample rate is: 2000000.052982 Hz
[R82XX] PLL not locked!
Allocating 15 zero-copy buffers
Args= 
Hash collision!!! Fatal error!!
pure virtual method called
terminate called without an active exception
Aborted

ptrkrysik, 2026-11-09:
This thread collects many different causes of "grgsm_scanner shows nothing" (no SDR found, an old GNU Radio / Python 2 setup, antenna and gain, overflows). Two defects in the scanner itself are fixed in [v0.43.0](https://github.com/ptrkrysik/gr-gsm/releases/tag/v0.43.0): `--args` was ignored, so a specific device could not be selected, and the IF and baseband gains were hard-coded (`--if-gain` and `--bb-gain` were added). `grgsm_scanner -d` prints the debug output that is otherwise hidden.

If the scanner still finds nothing with the new release, please open a new issue with your operating system, the output of `gnuradio-config-info --version`, the SDR in use, the exact command and the output of `grgsm_scanner -d`.

closed 2026-11-09 by ptrkrysik