ptrkrysik/gr-gsm #276: Segmentation fault (core dumped)

wdx7266, 2017-04-13:
### Steps to reproduce
1.Insert hackrf one to USB2.0
2.Input: grgsm_scanner -v

### Version information
**Operating system**:Ubuntu 14.04 LTS

**hackrf_info output:**
hackrf_info version: git-9bbbbbf
libhackrf version: git-9bbbbbf (0.5)
Found HackRF
Index: 0
Serial number: 0000000000000000443464c021651617
Board ID Number: 2 (HackRF One)
Firmware Version: 2017.02.1 (API:1.02)
Part ID Number: 0xa000cb3c 0x005e4f58

### Output
```
[INFO] [UHD] linux; GNU C++ version 4.8.4; Boost_105400; UHD_3.11.0.git-128-g379f922d

Segmentation fault (core dumped)

```

thejunaid, 2017-04-21:
i encountered this segmentation fault every time i tried loading the rtl sdr sink block into flowgraph. still searching for a bug.....

ptrkrysik, 2017-04-26:
@thejunaid, I will post here what I posted in another (closed #278) thread:

Segmentation faults are ofter result of a problem with installation of
GNU Radio/gr-osmosdr. I.e. mixing of source installation with packages
installed from distribution's repository can cause that. Usually
removing GNU Radio and its libraries and reinstalling it from the
beginning helps to solve such problems.

thejunaid, 2017-04-26:
Hi Piotr, I was able to resolve segmentation fault errors last week. U are
right. They are due to installation errors as well as might occur when
updating or upgrading repositories for an existing project. Thanks for your
kind perusal. Also I have taken up tinkering the flow graphs u made for gsm
hop reception. I m still far away from any actual reception though as I
don't know how to obtain kc key or hop seq no. Even from my own phone and
sim.

Junaid Ali

wdx7266, 2017-04-26:
I used `pybombs` to install `GNU Radio`, `gr-gsm `and  `gr-osmosdr`, I will try to reinstall them again, thank you very much!

Alster, 2017-05-23:
I have segfault with grgsm_scanner too. 
But its emitted not immediately: starting scanner with options` -v -b DCS1800` and after 1-2 minutes it fails. With `-v -b E-GSM` it's not fails, because process finished earlier than 2 minutes.
It's issue, because i have experience with installing grgsm, gnuradio and other on several machines. gnuradio usually install from ubuntu repository via apt-get. Now i install all stack via pybombs, and got same segfault too.
Current configuration: `[INFO] [UHD] linux; GNU C++ version 5.4.0 20160609; Boost_105800; UHD_3.11.0.git-193-g82f0d5a6`

thejunaid, 2017-05-23:
U mean to say that u are able to execute grgsm_livemon.grc for upto 2 mins
then segfault occurs?

Junaid Ali

Alster, 2017-05-23:
@thejunaid grgsm_livemon and other does not fails, scanner only.

As addition: i have used usrp, rtl-sdr dongles and hackrf, same issue with approximately same "timeout" before fail.
Also, same situation on Ubuntu, Kail, Kubuntu

thejunaid, 2017-05-23:
Thanks for the valuable information. I will check this and get back to u.

Junaid Ali

Alster, 2017-05-23:
@thejunaid thank you for fast answer)
Tell me if i can give additional info

ptrkrysik, 2017-06-11:
I'm using Ubuntu 16.04 and I'm unable to reproduce the problem. @alster and @wdx7266 I believe there might be some problem. However can you provide steps how starting with clean system (I prefer Ubuntu) get to the point where I can reproduce the problem?

Alster, 2017-06-15:
Sorry, now there is no time to make a clean installation.
If you can not reproduce the bug, then the issue can be closed, I think.
I will try to provide more information later.
But, look at my typical list / script as I usually tune clean system (not accurate, is taken from the memory).
`sudo apt-get install cmake swig doxygen
sudo apt-get install python3-pyopencl

sudo apt-get install rtl-sdr librtlsdr-dev
sudo apt-get install osmo-sdr libosmosdr-dev
sudo apt-get install libosmocore libosmocore-dev

sudo apt-get install python-pip python-dev build-essential
sudo pip install --upgrade pip
sudo pip install --upgrade virtualenv
sudo pip install PyBOMBS
sudo pybombs prefix init /usr/local -a default_prx
sudo pybombs config default_prefix default_prx
sudo pybombs recipes add gr-recipes git+https://github.com/gnuradio/gr-recipes.git
sudo pybombs recipes add gr-etcetera git+https://github.com/gnuradio/gr-etcetera.git
sudo pybombs install gnuradio

sudo apt-get install libfreetype6-dev ocl-icd-opencl-dev libqt4-opengl-dev

sudo pybombs install gr-fosphor
sudo pybombs install gr-osmosdr

# sudo apt-get install gnuradio gnuradio-dev

# sudo apt-get install gr-fosphor

cd ..
git clone https://github.com/ptrkrysik/gr-gsm.git
mkdir build
cd build
cmake ..
make
sudo make install -j

cd /usr/local/lib/uhd/utils
sudo cp uhd-usrp.rules /etc/udev/rules.d/
sudo udevadm control --reload-rules
sudo udevadm trigger

sudo apt-get install python-mako
sudo apt-get install libboost-all-dev `
I hope you will see some obvious mistake.

thejunaid, 2017-06-15:
> gnuradio/gr-recipes.git
> gnuradio/gr-etcetera.git

Hello Piotr, you can close the issue. The steps you have given are aptly
clear and will help anyone with seg faults to come out of it. Moreover I
simple way to remove seg faults that I encountered was to remove multiple
instances of same program like multiple installed versions of gnu radio.
Regards

Junaid Ali

ptrkrysik, 2017-07-23:
Thanks @thejunaid . I'm closing the issue. For anyone reading this, here is everything you need to install in order to compile and install gr-gsm manually on Ubuntu 16.04:

apt-get update && apt-get install -y \
    cmake \
    autoconf \
    libtool \
    build-essential \
    libcppunit-dev \
    swig \
    doxygen \
    liblog4cpp5-dev \
    python-scipy \
    gnuradio-dev \
    gr-osmosdr \
    libosmocore-dev

git clone https://github.com/ptrkrysik/gr-gsm.git
cd gr-gsm
mkdir build
cd build
cmake ..
make
sudo make install #sudo needed when installing in /usr/local
sudo ldconfig

closed 2017-07-23 by ptrkrysik