ptrkrysik/gr-gsm #277: Critical timing

Estacado7706, 2017-04-14:
Is there any way to get the real arrival time of an incoming transmission? Sending out the messages produced by the receiver through udp from a system that is maxed out seems to mix up the data.

Therefore an ordered handling of those packets can be a real pain...

ptrkrysik, 2017-04-14:
W dniu 14.04.2017 o 19:41, Estacado7706 pisze:
> Is there any way to get the real arrival time of an incoming
> transmission? Sending out the messages produced by the receiver through
> udp from a system that is maxed out seems to mix up the data.
> 
> Therefore an ordered handling of those packets can be a real pain...
> 
There are frame numbers in the header of each burst/message that are
unique in time frame of ~3 hours. Currently there is no way to
synchronize messages going through different message based links. I was
thinking about some block that would do sorting of messages. However I
don't know how to do this correctly yet, in order to not increase delay
too much. I don't know what would be good a trigger when to send the
messages to the output.

Estacado7706, 2017-04-15:
Yeah, that is exactly the problem. I'm scanning 5 frequencys at a time and walking through the whole band to scan for active hyperframes. Encountering a possible continuous frame number on another frequency requires me to calculate the timing distance of the two and compare this to the framenumbers.

But for this I'd need the correct time of arrival, not the time of final processing. In some cases the predicted framenumber will be up to 50 frames ahead / behind the one I am receiving even though it should be a match, simply due to latency and reordering while proocessing.

petterreinholdtsen, 2017-09-11:
Are you using grgsm_livemon for this?  It just received a commit to make sure the arfcn field in the gsmtap header is filled in properly.  Perhaps you can do this to map packages to frequencies?  I assume here the frame number is unique per arfcn value.

ptrkrysik, 2018-06-03:
Hi, @Estacado7706, I don't intend to work on this issue in the near future. In case you need this I'll be happy to receive Pull Request with this functionality.

closed 2018-06-03 by ptrkrysik