ptrkrysik/gr-gsm #281: a bug in extract_system_info_impl.cc

k1myok, 2017-05-31:
find a bug in extract_system_info_impl.cc 
gsm48_decode_freq_list(freq, &msg_elements[3], 16, 0x8e, 0x01);  //mask is 0x8e instead of 0x8c

relative function is in libosmocore/src/gsm/gsm_48.c line 1090
/* decode "Cell Channel Description" (10.5.2.1b) and other frequency lists */
int gsm48_decode_freq_list(struct gsm_sysinfo_freq *f, uint8_t *cd,
			   uint8_t len, uint8_t mask, uint8_t frqt)

if ((cd[0] & 0xce & mask) == 0x8e) {}    //if mask is 0x8c, this branch cannot be reached

E3V3A, 2017-06-03:
@k1myok Thanks for reporting, but if you don't wanna clone and make a PR. At least make a diff and clearly  show what was before and after, and for all gods sake, please use GitHubs built-in markup! In the worst case you can also "select" the code snippets in gh, and just post the link. Pull up the code, and add `#Lm-Ln` where `m,n` are the line numbers.

k1myok, 2017-06-05:
https://github.com/ptrkrysik/gr-gsm/blob/master/lib/misc_utils/extract_system_info_impl.cc

 - -gsm48_decode_freq_list(freq, &msg_elements[3], 16, 0x8c, 0x01);
 +gsm48_decode_freq_list(freq, &msg_elements[3], 16, 0x8e, 0x01);

ptrkrysik, 2017-06-11:
Thank you for pointing out this issue. 0x8c is correct for one type of frequency list and 0x8e for another, so if i replace one with another it will work in other instances (correct me if I'm wrong 0x8e will be universally good for this job). The proper solution is most probably to remove hard-coded value with detection of frequency list type.

ptrkrysik, 2017-06-11:
@k1myok  ok, actually what you gave actually allows libosmocom's function to do that. Thank you for your help :)

commit b31689d2 (2017-06-11): Changed mask for System Information type 2ter. Closes #281.
closed 2017-07-20 by ptrkrysik