ptrkrysik/gr-gsm #282: grgsm_livemon fails with empty windows and server error

E3V3A, 2017-06-02:
I just fired up a fresh and just updated Kali and followed your instructions for manual compile. 
Everything seem to have gone fine, but the `grgsm_livemon` has some weird bugs. Apart from below, the window is completely empty. Have tried various way of running as root and not. 

I'm using a *NooElec* `nestdr SMArt`. I'm also running Wireshart on loopback IF that does yield seemingly correct info. 

**Any ideas what could be going wrong?**

```bash
root@kali7:/home/xxxx/mygsm/gr-gsm# grgsm_livemon -g 25 -p 1 

linux; GNU C++ version 6.3.0 20170221; Boost_106200; UHD_003.009.005-0-unknown

gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.10
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy soapy redpitaya 
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for 4294967295, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for 4294967295, skipping unlock
Found Rafael Micro R820T tuner
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Exact sample rate is: 2000000.052982 Hz
[R82XX] PLL not locked!
X Error: BadAccess (attempt to access private resource denied) 10
  Extension:    130 (MIT-SHM)
  Minor opcode: 1 (X_ShmAttach)
  Resource id:  0x1e00004
X Error: BadShmSeg (invalid shared segment parameter) 128
  Extension:    130 (MIT-SHM)
  Minor opcode: 5 (X_ShmCreatePixmap)
  Resource id:  0x1e00011
X Error: BadDrawable (invalid Pixmap or Window parameter) 9
  Major opcode: 62 (X_CopyArea)
  Resource id:  0x1e00012
...
<same>
...
X Error: BadDrawable (invalid Pixmap or Window parameter) 9
  Major opcode: 62 (X_CopyArea)
  Resource id:  0x1e00012
 XX XX XX XX XX XX XX XX XX XX XX XX XX 41 2b 2b 2b 2b 2b 2b
X Error: BadDrawable (invalid Pixmap or Window parameter) 9
  Major opcode: 62 (X_CopyArea)
  Resource id:  0x1e00012

 XX XX XX XX XX XX XX XX XX c0 17 05 f4 f2 d3 3b 6b 2b 2b 2b 2b 2b 2b
 XX XX XX XX XX XX XX XX XX 01 c8 04 3c 54 85 00 79 00 00 80 00 00 4b
 XX XX XX XX XX XX XX XX XX 00 01 c8 04 3c 54 85 00 79 00 00 80 00 00 4b
...
```

E3V3A, 2017-06-03:
Ok, I managed to bypass the `jack server` error with:

```bash
# dbus-x11 is already installed
sudo apt-get install jackd2
<yes>
sudo dbus-launch jack_control start
```

Now the output look like this: 

```bash 
root@kali7:/home/xxxx/mygsm/gr-gsm# grgsm_livemon -g 25 -p 1 

linux; GNU C++ version 6.3.0 20170221; Boost_106200; UHD_003.009.005-0-unknown

gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.10
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy soapy redpitaya 
Found Rafael Micro R820T tuner
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Exact sample rate is: 2000000.052982 Hz
[R82XX] PLL not locked!
X Error: BadAccess (attempt to access private resource denied) 10
  Extension:    130 (MIT-SHM)
  Minor opcode: 1 (X_ShmAttach)
  Resource id:  0x1600004
X Error: BadShmSeg (invalid shared segment parameter) 128
  Extension:    130 (MIT-SHM)
  Minor opcode: 5 (X_ShmCreatePixmap)
  Resource id:  0x1600011
X Error: BadDrawable (invalid Pixmap or Window parameter) 9
  Major opcode: 62 (X_CopyArea)
  Resource id:  0x1600012
```

E3V3A, 2017-06-03:
Actually, I just found out that a better way to run is using Qt jack with: `qjackctl -s`. (So perhaps there is no need to isntall `jackd2`?)

E3V3A, 2017-06-03:
Got it! But this is some weird QT/GUI related shit. 
**Don't use a root shell for GUI shit.**

- If you use: 
```bash
gksudo grgsm_livemon
```  
  You will get **only** the hex output. 
- But if you use:
```bash
sudo QT_X11_NO_MITSHM=1 grgsm_livemon
```
Window problem fixed! It's all good and party!

ptrkrysik, 2017-06-11:
Hi @E3V3A. You shouldn't need sudo to run livemon. What happens when you try it?

From what I see generally there is plenty of weird problems when trying to install/use gr-gsm on Kali linux, but these are some of weirdest.

E3V3A, 2017-06-20:
@ptrkrysik  
> You shouldn't need sudo to run livemon. What happens when you try it?

That works fine too. You can close this issue, but may want to tell people about that `QT_X11_NO_MITSHM=1` thing.

E3V3A, 2017-06-21:
I just did a `sudo apt-get upgrade` and now it fails again without `sudo`, even with the `export QT_X11_NO_MITSHM=1` in `.bashrc`. Strange.

jjbaek35, 2017-06-30:
After solving the problem, 

"[R82XX] PLL not locked!" message is also disappeared?

or regradless of that message, it works?

E3V3A, 2017-06-30:
> [R82XX] PLL not locked!

@jjbaek35  No I never was able to get rid of those. But I did get rid of all `jack server` related warnings by adding the `--args=rtl` option. And the window start as it should (without `sudo`) after specifying all kinds of CLI options.

E3V3A, 2017-06-30:
I think I understand the problem with **"[R82XX] PLL not locked!"**.  I believe it is caused by USB interface latency when running in VM. Also, please note that this message has [been changed](https://github.com/librtlsdr/librtlsdr/blob/development/src/tuner_r82xx.c#L550) to: **"[R82XX] PLL not locked for %u Hz!"**.

closed 2017-06-30 by E3V3A