ptrkrysik/gr-gsm #283: error while installing apache-thrift

suhaybcabdulahi, 2017-06-05:
when i tried to install gr-gsm the apache thrift not configuring he says {Configuration failed. Re-trying with higher verbosity}
and {Error installing package apache-thrift. Aborting}
please help me

ptrkrysik, 2017-06-11:
apache-thrift is one of gr-gsm dependencies. We don't provide support for it but I can provide some limited help if you show what exactly went wrong with configure command when installing it. Run pybombs with -v option and show what you get.

flyoffthehandle, 2017-06-20:
I got the same error.
(ran pybombs install gr-gsm)

installation gets to apache-thrift:

checking for BN_init in -lcrypto... no
"Error: libcrypto required."

(aborts after this error)

I can get a full log to paste later today. That is where the installation stopped for me.

superfrancois, 2017-07-17:
Hello Ptrkrysik.
Same problem for me under kali2.0 AMD64

root@kali:~# pybombs -v install gr-gsm
PyBOMBS.ConfigManager - DEBUG - Reading config info from file: /etc/pybombs/config.yml
PyBOMBS.ConfigManager - DEBUG - New config items: {}
PyBOMBS.ConfigManager - DEBUG - Reading config info from file: /root/.pybombs/config.yml
PyBOMBS.ConfigManager - DEBUG - New config items: ordereddict([('default_prefix', 'default_prx'), ('git-cache', '/root/.pybombs/gitcache'), ('packagers', 'pip,apt,pymod,pkgconfig,cmd'), ('makewidth', 8), ('elevate_pre_args', ['sudo', '-H'])])
PyBOMBS.ConfigManager - DEBUG - Template directory: /usr/local/lib/python2.7/dist-packages/pybombs/templates
PyBOMBS.ConfigManager.PrefixInfo - DEBUG - Inspecting config file: /etc/pybombs/config.yml
PyBOMBS.ConfigManager.PrefixInfo - DEBUG - Inspecting config file: /root/.pybombs/config.yml
PyBOMBS.ConfigManager.PrefixInfo - DEBUG - Resolving prefix alias `default_prx'.
PyBOMBS.ConfigManager.PrefixInfo - DEBUG - Using default_prefix as prefix (/usr/local)
PyBOMBS.ConfigManager.PrefixInfo - DEBUG - Choosing default prefix config dir: /usr/local/.pybombs
PyBOMBS.ConfigManager.PrefixInfo - DEBUG - Inspecting config file: /usr/local/.pybombs/config.yml
PyBOMBS.ConfigManager.PrefixInfo - DEBUG - Prefix source dir is: /usr/local/src
PyBOMBS.Inventory - DEBUG - Trying to load inventory file /usr/local/.pybombs/inventory.yml...
PyBOMBS.ConfigManager.PrefixInfo - DEBUG - Prefix-local recipe dir is: /usr/local/.pybombs/recipes
PyBOMBS.ConfigManager - DEBUG - Reading config info from file: /usr/local/.pybombs/config.yml
PyBOMBS.ConfigManager - DEBUG - New config items: {}
PyBOMBS.ConfigManager - DEBUG - Full list of recipe locations: ['/usr/local/.pybombs/recipes', '/root/.pybombs/recipes/gr-recipes', '/root/.pybombs/recipes/gr-etcetera', '/usr/local/lib/python2.7/dist-packages/pybombs/recipes']
PyBOMBS.ConfigManager - DEBUG - Named recipe locations: {'gr-etcetera': 'git+https://github.com/gnuradio/gr-etcetera.git', 'gr-recipes': 'git+https://github.com/gnuradio/gr-recipes.git'}
PyBOMBS - INFO - PyBOMBS Version 2.3.1
PyBOMBS.RecipeListManager - DEBUG - Adding recipe location: /usr/local/.pybombs/recipes
PyBOMBS.RecipeListManager - DEBUG - Scanning directory '/usr/local/.pybombs/recipes' for recipes...
PyBOMBS.RecipeListManager - DEBUG - Found 0 new recipes.
PyBOMBS.RecipeListManager - DEBUG - Loading templates.
PyBOMBS.RecipeListManager - DEBUG - Adding recipe location: /root/.pybombs/recipes/gr-recipes
PyBOMBS.RecipeListManager - DEBUG - Scanning directory '/root/.pybombs/recipes/gr-recipes' for recipes...
PyBOMBS.RecipeListManager - DEBUG - Found 172 new recipes.
PyBOMBS.RecipeListManager - DEBUG - Loading templates.
PyBOMBS.RecipeListManager - DEBUG - Adding recipe location: /root/.pybombs/recipes/gr-etcetera
PyBOMBS.RecipeListManager - DEBUG - Scanning directory '/root/.pybombs/recipes/gr-etcetera' for recipes...
PyBOMBS.RecipeListManager - DEBUG - Found 89 new recipes.
PyBOMBS.RecipeListManager - DEBUG - Loading templates.
PyBOMBS.RecipeListManager - DEBUG - Adding recipe location: /usr/local/lib/python2.7/dist-packages/pybombs/recipes
PyBOMBS.RecipeListManager - DEBUG - Scanning directory '/usr/local/lib/python2.7/dist-packages/pybombs/recipes' for recipes...
PyBOMBS.RecipeListManager - DEBUG - Found 11 new recipes.
PyBOMBS.RecipeListManager - DEBUG - Loading templates.
PyBOMBS.install - DEBUG - Initializing command class for command install
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager pip
PyBOMBS.PackageManager - DEBUG - pip is supported!
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager apt
PyBOMBS.Packager.apt - INFO - Install python-apt to speed up apt processing.
PyBOMBS.PackageManager - DEBUG - apt is supported!
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager pymod
PyBOMBS.PackageManager - DEBUG - pymod is supported!
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager pkgconfig
PyBOMBS.PackageManager - DEBUG - pkgconfig is supported!
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager cmd
PyBOMBS.PackageManager - DEBUG - cmd is supported!
PyBOMBS.PackageManager - DEBUG - Using binary packagers: ['pip', 'apt', 'pymod', 'pkgconfig', 'cmd']
PyBOMBS.install_manager - INFO - Phase 1: Creating install tree and installing binary packages:
PyBOMBS.PackageManager - DEBUG - Checking if package gr-gsm is installable...
PyBOMBS.PackageManager - DEBUG - Package gr-gsm is requesting a source build.
PyBOMBS.PackageManager - DEBUG - Checking if package gr-gsm is installed...
PyBOMBS.PackageManager - DEBUG - Package gr-gsm is requesting a source build.
PyBOMBS.PackageManager - DEBUG - Package gr-gsm is not installed.
PyBOMBS.install_manager - DEBUG - Testing binary install for package gr-gsm.
PyBOMBS.PackageManager - DEBUG - install(gr-gsm, install_type=binary, static=False)
PyBOMBS.PackageManager - DEBUG - Package gr-gsm is requesting a source build.
PyBOMBS.PackageManager - DEBUG - Returning no packagers -- package is requesting source build, but binary build is requested.
PyBOMBS.PackageManager - DEBUG - Checking if package gnuradio is installable...
PyBOMBS.PackageManager - DEBUG - Package gnuradio is requesting a source build.
PyBOMBS.PackageManager - DEBUG - Checking if package gnuradio is installed...
PyBOMBS.PackageManager - DEBUG - Package gnuradio is requesting a source build.
[... cut ...]
Makefile:542 : la recette pour la cible « all » a échouée
make: *** [all] Erreur 2
PyBOMBS.monitor_process() - DEBUG - Thread signaled termination or returned
PyBOMBS.monitor_process() - DEBUG - Return value: 2
PyBOMBS.Packager.source - ERROR - Build failed. See output above for error messages.
PyBOMBS.Packager.source - ERROR - Problem occurred while building package apache-thrift:
Build failed.
PyBOMBS.install_manager - ERROR - Error installing package apache-thrift. Aborting.

If you've got an idea!!!

Thx by advance

MarioGeckler, 2017-07-27:
BN_init was removed in OpenSSL 1.1
You need to install OpenSSL packages version < 1.1 to build apache-thrift.

petterreinholdtsen, 2017-09-12:
According to https://issues.apache.org/jira/browse/THRIFT-3207 the latest versions of apache-thrift should build with OpenSSL 1.1.

KristaHyer, 2018-04-18:
Another option other than installing an OpenSSL package < 1.1 is to switch out BN_init for BN_new.
This is the fix in the facebook version of thrift.
fbthrift issue - https://github.com/facebook/fbthrift/issues/222
fbthirft fix - https://github.com/facebook/fbthrift/commit/156aa78e57ee8abd56c2ccdfb8f9743a6e081f30

velichkov, 2018-04-18:
Hi Everyone,

The `apche-thrift` is marked as optional package in the [`gnuradio-default`](https://github.com/gnuradio/gr-recipes/blob/af729f6bfd7e82d55e91be68e953c4713b27990f/gnuradio-default.lwr#L25-L26) receipt and actually it is not needed for gr-gsm to work. When a package is marked as optional and its compilation fails then it gets skipped. 

All you need to do is to use `gnuradio-default` receipt to initialize the prefix and not directly to install gr-gsm
```
sudo pybombs prefix init /usr/local -a default_prx -R gnuradio-default
```

closed 2018-06-03 by ptrkrysik