ptrkrysik/gr-gsm #285: Multiple Instances of airprobe_rtlsdr.py

engrmasood2002, 2017-06-27:
Hi, In the IMSI catcher Project  
https://github.com/Oros42/IMSI-catcher
airprobe_rtlsdr.py is used to grab data. I need to run multiple instances of this script to connect with two or more RTL2832u dongles to receive data from different center frequencies.
When i run multiple instances, first one runs fine, in second and more address in use **"127.0.0.1"** comes as all the instances of script try to open same port and IP for writing but it should not not be the case as for writing only to same port and IP one does not need to bind (Bind call normally used when receiving on IP and port is required). Please correct me if i am wrong.
Secondly, how i could connect with multiple dongles , do i need to pass something here in the script or else.
**self.rtlsdr_source_0 = osmosdr.source( args="numchan=" + str(1) + " " + args )**
I am C/C++ programmer so have little problem figuring out python code.
Thanks.

E3V3A, 2017-06-30:
Just FYI. 
`airprobe_rtlsdr.py` is deprecated. The [new name](https://github.com/ptrkrysik/gr-gsm/wiki/Usage) is `grgsm_livemon`.

engrmasood2002, 2017-07-01:
Thanks for the reply. Well the airprobe_rtlsdr.py is still feasible for me if I can have option to connect  dongle of my choice. I am able to run its multiple instances. Do you know how I can select source of dongle manually out of two or more  connected to computer?. Thanks

E3V3A, 2017-07-07:
@engrmasood2002 
I think you can do: `grgsm_livemon --args=rtl=0 ...`. Where `0` is the number of your RTL dongle. Not sure this will work through, as I don't know where these arguments are passed on to.

ptrkrysik, 2017-07-23:
As for addressing dongles the solution of @E3V3A should be fine (the value is just passed to arguments of Osmosdr block). As for IP addresses - you can open apps/grgsm_livemon.grc in gnuradio-companion and add a Parameter block with IP address passed as a string. Then put the ID of the new parameter block in Host field of both Socket PDU blocks. Then generate python new python executable (which will appear as apps/grgsm_livemon.py) with F5. You will learn a bit of gnuradio-companion and gr-gsm in the process :).

closed 2017-07-23 by ptrkrysik

usimplex, 2020-02-02:
It would be a lot easier to choose same the --serverport and --collectorport for other livemon instances

usimplex, 2020-02-02:
So I came across this convenient hack : 

modify grgsm_livemon.py and rename it to something like livemon2  :
comment out these two lines 

`self.blocks_socket_pdu_0_0 = blocks.socket_pdu("UDP_SERVER", '127.0.0.1', serverport, 10000, False)` 
`self.msg_connect((self.blocks_socket_pdu_0_0, 'pdus'), (self.gsm_message_printer_1, 'msgs'))`

run a grgsm_livemon with first sdr device then run other devices with their respected indexes with livemon2 that you just created.

Hope this helps someone 
BTW thanks @ptrkrysik for all the hard work

ChillVibesMushroom, 2022-05-30:
I have a frequency on 959.2 and when I tune to that frequency I'm not getting packets with grgsm_livemon using the hackrf any ideas