ptrkrysik/gr-gsm #286: Cannot get packet using grgsm_livemon

jjbaek35, 2017-06-28:
Hi~ all.

I just installed successfully grgsm_livemon with RTL-SDR dongle.
When I run the capture command, I can't capture any signals or packets.
I don't know why I can't not capture any packets of GSM band.
Please look over my attached figure.
I put the initial frequency 861e6.
There are two messages "PLL not locked!" which means dongle can't set the requested frequency.
I suspect that point. but I don't know why.

[image]

E3V3A, 2017-06-28:
@jjbaek35  Unfortunately you're not supplying enough info to help sort your problem. So most likely the problem is you, and not gr-gsm, as it seem to work in your screenshot. Things to consider: 

* You need to do a GSM scan first to get an active channel. (ARFCN)
* You need to calibrate to get the correct PPM 
* You need to get the sample rate right (~1800000-2M)
* You need to have a good antenna
* You need to set the gain right (~30 for good RTL SDRs)

jjbaek35, 2017-06-29:
My dongle RTL-SDR has 1PPM TCXO oscillator, it means calibrating isn't needed.
Anyway, I tried  to calibrate with Kalibrate as belows:
Could you let me know how can I get the signal?

**ubuntu@ubuntu:~/kal/kalibrate-rtl$ kal -v -s GSM900 -g 50**
Found 1 device(s):
  0:  Generic RTL2832U OEM

Using device 0: Generic RTL2832U OEM
Detached kernel driver
Found Rafael Micro R820T tuner
Exact sample rate is: 270833.002142 Hz
[R82XX] PLL not locked!
Setting gain: 50.0 dB
kal: Scanning for GSM-900 base stations.
channel detect threshold: 39540.808594
GSM-900:
	chan: 6 (936.2MHz - 5.178kHz)	power: 1593200.24
	chan: 11 (937.2MHz - 30.630kHz)	power: 222674.68
	chan: 13 (937.6MHz + 31.674kHz)	power: 112603.63
	chan: 14 (937.8MHz + 22.514kHz)	power: 1510918.61
	chan: 16 (938.2MHz + 36.322kHz)	power: 290972.69
	chan: 18 (938.6MHz - 17.571kHz)	power: 426886.58
	chan: 21 (939.2MHz - 18.739kHz)	power: 336030.01
	chan: 22 (939.4MHz + 6.370kHz)	power: 2732522.87
	chan: 29 (940.8MHz - 4.048kHz)	power: 47734.90
	chan: 82 (951.4MHz + 32.317kHz)	power: 50797.78
	chan: 88 (952.6MHz + 16.579kHz)	power: 1060392.23

**ubuntu@ubuntu:~/kal/kalibrate-rtl$ kal -v -s GSM850 -g 50**
Found 1 device(s):
  0:  Generic RTL2832U OEM

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
Exact sample rate is: 270833.002142 Hz
[R82XX] PLL not locked!
Setting gain: 50.0 dB
kal: Scanning for GSM-850 base stations.
channel detect threshold: 341526.327240
GSM-850:
	chan: 231 (889.8MHz + 38.297kHz)	power: 353247.72

**ubuntu@ubuntu:~/kal/kalibrate-rtl$ kal -v -s GSM850 -g 50**
Found 1 device(s):
  0:  Generic RTL2832U OEM

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
Exact sample rate is: 270833.002142 Hz
[R82XX] PLL not locked!
Setting gain: 50.0 dB
kal: Scanning for GSM-850 base stations.
channel detect threshold: 341526.327240
GSM-850:
	chan: 231 (889.8MHz + 38.297kHz)	power: 353247.72

E3V3A, 2017-06-30:
1. You are not calibrating it right. The right command is: `kal -g 25 -c 22`. That should give you a PPM.
2. You're running in a VM, so there [seem to be](https://github.com/librtlsdr/librtlsdr/issues/50) USB latency causing dropped packets which prevents proper PLL locking. Just ignore.

jjbaek35, 2017-06-30:
Thank you so much, E3V3A!
I run that command with options, so what is the ppm value? 
and so, I need to install grgram on host OS without VM to remove the usb latency?
I checked the source code "PLL not locked", my version is different.
Let me try to update all source code and compile again.

Thanks so much again!

root@deli:~# kal -g 25 -c 22
Found 1 device(s):
  0:  Generic RTL2832U OEM

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
Exact sample rate is: 270833.002142 Hz
[R82XX] PLL not locked!
Setting gain: 25.0 dB
kal: Calculating clock frequency offset.
Using GSM-900 channel 22 (939.4MHz)
average		[min, max]	(range, stddev)
+ 7.274kHz		[5874, 8671]	(2797, 860.250488)
overruns: 0
not found: 15
average absolute error: -7.743 ppm

jjbaek35, 2017-07-01:
I just updated librtlsdr, but still can't not get the GSM packets.
I suspect the usb latency, how can I solve it?

**root@deli:~/2gsm/librtlsdr-development/build# grgsm_livemon -f 934000000 -p -7**
linux; GNU C++ version 6.3.0 20170221; Boost_106200; UHD_003.009.005-0-unknown

gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.10
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy soapy redpitaya 
Found Rafael Micro R820T tuner
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Found Rafael Micro R820T tuner
[R82XX] PLL not locked for 1700000 Hz!
Exact sample rate is: 2000000.052982 Hz
[R82XX] PLL not locked for 1625000 Hz!

E3V3A, 2017-07-03:
Looking at that picture, there is nothing on that frequency, so you didn't scan for any ARFCNs first.  Once you find an active ARFCN then re-run with that frequency. You need to read the Wiki and look at you tube how to do this.

jjbaek35, 2017-07-04:
Ok Thank you, eventually, I got the GSM packets.
US use PCS band 1900 Mhz, so I scanned 

kal -s  PCS -g 40

I found the beacon signal from T-mobile.
The key is the frequency.

closed 2017-07-04 by jjbaek35

ChillVibesMushroom, 2022-10-01:
So if grgsm scans on a gain of 30

I use kal -s 1900 -g 30 to pull the same info and some with grgsm then I use gqrx to get a more focused freq

Polymorphic86, 2024-10-23:
Ive tried all the above and I'm running dragonOS and all I get is no valid pll values for xxxxxxxxxx Hz! R81_set_freq: failed =-1  all the way down the terminal

dark0069, 2025-04-06:
Im also facing same issue on dragon os and im using nooelec nesdr v5. 
I'm attaching the ss of todays grgsm_livemon. Please help me out. [image]

[image]

[image]