ptrkrysik/gr-gsm #288: What does these grgsm_livemon options do?

E3V3A, 2017-06-30:
What does the following `grgsm_livemon` options do? 

```
-o SHIFTOFF, --shiftoff=SHIFTOFF          Set shiftoff [default=400k]
--osr=OSR                                 Set OSR [default=4]
 ```
Clearly, the `SHIFTOFF` is mislabeled and should be some kind of value...

E3V3A, 2017-07-05:
@ptrkrysik ??

axilirator, 2017-07-06:
Just look at the follow graph. They are both follow graph variables.
First is used to adjust the phase increment of 'Rotator' block.
Second one is for 'GSM Receiver' and 'GSM Clock Control' blocks.

Alster, 2017-07-16:
@axilirator I join the question, namely: what is OSR?

ptrkrysik, 2017-07-23:
OSR stands for OverSampling Ratio - it is number of samples per GSM symbol.

The shiftoff is frequency offset added in order to shift DC offset to another frequency so it is possible to filter it out. The receiver isn't setting its central frequency to the centre of a GSM channel but at a frequency different by SHIFTOFF from it.

closed 2017-07-23 by ptrkrysik