ptrkrysik/gr-gsm #290: gr-gsmlivemon on uplink decoding

engrmasood2002, 2017-07-14:
Hi,
I am new to GNU Radio, saw your great project.  
I have seen your comments in forums that gr-gsm apps cant be used to decode uplink due synchronization unless your multi-RTL solution is adopted. i am interested in getting location update message only from uplink. looking at this line of code for grgsm_livemon.py
**self.gsm_sdcch8_demapper_0 = grgsm.gsm_sdcch8_demapper(timeslot_nr=0,)**
Would this line decode sdcch frame ? (location update is present in this frame)
What i need to change in this script to get location update only.
Thanks

engrmasood2002, 2017-07-14:
Hi Piotr
I want to receive uplink mobile messages but was not able to find your uplink-decoding branch (Issue No.94) you mentioned earlier in this post. can you provide me the link to download it and instructions on using it.
I am using RTL SDR usb dongle and want to receive unsynchronized messages not voice.
Thanks

Alster, 2017-07-16:
Location update messages is unencrypted?
I could not get the decoded messages from the uplink without decryption.
If this is possible, and there are such messages, I will be surprised =)

engrmasood2002, 2017-07-17:
I want to receive messages from any mobile nearby to see what they transmit so I don't need any synchronization. I found uplink receiver but it works with 2 channels.
Instead of modifying this I as a test modified airprobe_rtlsdr.py and changed variable from false to true. I think it's a switch for reciver code to grgsmliveon     ...gsmreciver(...,True). Also put encryption keys from example code in airprobe_rtlsdr.py  but received nothing.
Any help will be highly appreciated.

Alster, 2017-07-22:
@engrmasood2002 see [this answer](https://groups.google.com/d/msg/gr-gsm/WJFbdrMCGAE/lDpq4mwDEAAJ), i hope it would be helpful.

ptrkrysik, 2017-07-23:
@engrmasood2002 , the line you mentioned is not doing what you want to achieve. Currently there is no way to receive just uplink as gr-gsm needs downlink to keep synchronization. If you want to continue the thread please ask on the mailing-list and post the link to this discussion.

commit 6629abc5 (2017-07-23): Closes #290 - the problem with subtracting of timeslot number for timeslots 0,1,2.
closed 2017-07-23 by ptrkrysik