ptrkrysik/gr-gsm #291: Receiver performance

Alster, 2017-07-16:
The Receiver block objectively can not cope with processing a large number of channels. Now it is the bottleneck.
For example, if you take 5 channels downlink and another 5 uplink - all, the rest of the flovgraph can only wait.
What are the possible decisions?

Alster, 2017-07-21:
I assume that you can make the block multithreaded, but I'm afraid of the complexity of the block logic.

ptrkrysik, 2017-07-23:
Hi @Alster. Yes, to use multi-threading is a simplest solution to improve performance in situation where many uplink and downlink channels are used. Currently it is done in a for loop - one channel after another.  The best solution (at least in terms of modularity of the project) would be to have a separate block for each channel, with one master block that process C0. But this will add another problem of synchronizing and managing these blocks, so it isn't a simple solution (and it might make other things more complicated as well).

E3V3A, 2017-07-23:
@Alster  Can you post a screen shot of the block in question? Surely sound interesting to see how we could synchronize these. 

@ptrkrysik  What exactly need to be synchronized?

Alster, 2017-07-23:
@E3V3A ,

[image]

Alster, 2017-07-23:
@ptrkrysik As I see, all the data outside the C0 channel is interpreted as normal_or_noise, and the measurements emanate only from their C0 channels. E.g, It turns out that we have something like this:
1. С0Receiver - processes the main channel and generating measurements for InputAdaptor's.
2. CXReceiver - works with one of CX channels.
3. Channel Combiner - actually, connects all the inputs in one output. By analogy with the Receiver: has one C0 and a lot of CX.
Well, or something like that.

The above synchronization problem comes out: how does Channelel Combiner feel confident in the consistency of the messages that have come to him? How to guarantee their order? After all, CXReceivers can operate at different speeds. Did you mean this?

ptrkrysik, 2017-07-23:
I meant that you need to be synchronized with bursts clock in order to
demodulate bursts. This can be done for example by sending some message
to all blocks that will tell them which sample starts some frame with
given frame number. Then each block can count samples on its own. When
everything is inside of one block then you don't need to send such
communicates and you can easily change the settings.

What you mentioned is the next problem added by separating receiver into
multiple blocks where each processes different channel - that if they
have message outputs then one can easily lose synchronization as message
passing is asynchronous.

Alster, 2017-07-23:
@ptrkrysik, thank you for your reply.
It turns out, the decision with division into blocks is unjustifiably complex for such a problem.

The task requires parallel processing of channels, which is quite trivial with the use of multithreading. Therefore, it is reasonable to dwell on multithreading in one block.

ptrkrysik, 2026-11-09:
The receiver block now processes channels in parallel with OpenMP, so this request has been implemented; please open a new issue with performance numbers if it is still a bottleneck.

I am closing this as part of a clean-up for the new release (v0.43.0), which ports gr-gsm to GNU Radio 3.10 (plain CMake build, no SWIG, installation steps in the README). If the problem still occurs with it, please open a new issue with your operating system, the output of `gnuradio-config-info --version`, how gr-gsm and its dependencies were installed, the exact command and the complete output.