ptrkrysik/gr-gsm #292: grgsm_livemon error running

engrmasood2002, 2017-07-16:
Hi
I am getting error 
GSM input adaptor(9): insufficient connected input ports (1 needed, 0 connected)

when running grgsm_livemon in ubuntu 16.04 LTS with gnuradio 3.7.9 following instructions from 
https://github.com/ptrkrysik/gr-gsm/wiki/Installation

I also tried with  gnuradio 3.7.9 but same error following instructions from
https://github.com/ptrkrysik/gr-gsm/issues/135

Any idea what could be wrong. PyBOMBS install correctly in both cases.

engrmasood2002, 2017-07-17:
Just wanted to update.
Problem solved by fresh installation as per gr-gsm installation wiki.

E3V3A, 2017-07-18:
@engrmasood2002 If the problem is solved, and issue invalidated, please close it.

closed 2017-07-18 by engrmasood2002