ptrkrysik/gr-gsm #295: Soapysdr install error on installation

stevevaius77, 2017-07-25:
I am receiving error on installation gr-gsm (Ubuntu 17)
I installed all dependencies and all steps listed but I received following error
How I can fix this to install properly?
[image]

stevevaius77, 2017-07-25:
Error log on CMakeError.log

Performing C++ SOURCE FILE Test SIZE_T_IS_UNSIGNED_INT failed with the following output:
Change Dir: /usr/local/src/soapysdr/build/CMakeFiles/CMakeTmp

Run Build Command:"/usr/bin/make" "cmTC_0b039/fast"
/usr/bin/make -f CMakeFiles/cmTC_0b039.dir/build.make CMakeFiles/cmTC_0b039.dir/build
make[1]: Entering directory '/usr/local/src/soapysdr/build/CMakeFiles/CMakeTmp'
Building CXX object CMakeFiles/cmTC_0b039.dir/src.cxx.o
/usr/bin/c++     -fvisibility-inlines-hidden -DSIZE_T_IS_UNSIGNED_INT   -o CMakeFiles/cmTC_0b039.dir/src.cxx.o -c /usr/local/src/soapysdr/build/CMakeFiles/CMakeTmp/src.cxx
/usr/local/src/soapysdr/build/CMakeFiles/CMakeTmp/src.cxx: In function ‘int main()’:
/usr/local/src/soapysdr/build/CMakeFiles/CMakeTmp/src.cxx:4:38: error: cannot convert ‘unsigned int*’ to ‘size_t* {aka long unsigned int*}’ in initialization
     size_t *x = (unsigned int *)(NULL);
                                      ^
CMakeFiles/cmTC_0b039.dir/build.make:65: recipe for target 'CMakeFiles/cmTC_0b039.dir/src.cxx.o' failed
make[1]: *** [CMakeFiles/cmTC_0b039.dir/src.cxx.o] Error 1
make[1]: Leaving directory '/usr/local/src/soapysdr/build/CMakeFiles/CMakeTmp'
Makefile:126: recipe for target 'cmTC_0b039/fast' failed
make: *** [cmTC_0b039/fast] Error 2

Source file was:

    #include <cstddef>
    int main() {
    size_t *x = (unsigned int *)(NULL);
    return 0; }

bookdude, 2017-07-27:
I know this does not help but I am seeing the same issue.  But If I install soapysdr independently on the same system it runs without issue.  

Unfortunately re-running the pybomb for gr-gsm still fails with the same message.

bookdude, 2017-07-27:
One more note here.  I was able to revert back one commit version and install with no issue.

stevevaius77, 2017-07-27:
Thanks! I solved it by installing it without use pybombs and master folder

closed 2018-06-02 by ptrkrysik