ptrkrysik/gr-gsm #296: Make compile error unterminated #if

forever-balanced, 2017-08-05:
I couldn't use the pybombs to install so I followed the manual install instructions and came across this 

  [ 61%] Built target grgsm
[ 64%] Built target _grgsm_swig_doc_tag
[ 67%] Built target grgsm_swig_swig_doc
[ 69%] Built target _grgsm_swig_swig_tag
[ 73%] Built target grgsm_swig_swig_2d0df
[ 78%] Built target pygen_swig_73903
[ 79%] Building CXX object swig/CMakeFiles/_grgsm_swig.dir/grgsm_swigPYTHON_wrap.cxx.o
/root/gr-gsm/build/swig/grgsm_swigPYTHON_wrap.cxx:914:0: error: unterminated #if
 #if PY_VERSION_HEX < 0x02050000 && !defined(PY_SSIZE_T_MIN)
 
swig/CMakeFiles/_grgsm_swig.dir/build.make:68: recipe for target 'swig/CMakeFiles/_grgsm_swig.dir/grgsm_swigPYTHON_wrap.cxx.o' failed
make[2]: *** [swig/CMakeFiles/_grgsm_swig.dir/grgsm_swigPYTHON_wrap.cxx.o] Error 1
CMakeFiles/Makefile2:524: recipe for target 'swig/CMakeFiles/_grgsm_swig.dir/all' failed
make[1]: *** [swig/CMakeFiles/_grgsm_swig.dir/all] Error 2
Makefile:160: recipe for target 'all' failed
make: *** [all] Error 2

Any ideas on what to do to fix this, thanks.

petterreinholdtsen, 2017-09-13:
I do not see this when building on Debian.  Could it be a version skew problem related to swig or some other dependency?

ptrkrysik, 2018-06-02:
is this sill a problem currently?

closed 2018-07-06 by ptrkrysik