ptrkrysik/gr-gsm #297: grgsm_capture error

Danikon, 2017-08-05:
Ubuntu 16.04 .  grgsm_capture error

ser@ubuntu:~/gr-gsm/apps$ grgsm_capture.py
Traceback (most recent call last):
  File "/usr/local/bin/grgsm_capture.py", line 33, in <module>
    import osmosdr
  File "/usr/local/lib/python2.7/dist-packages/osmosdr/__init__.py", line 26, in <module>
    from osmosdr_swig import *
  File "/usr/local/lib/python2.7/dist-packages/osmosdr/osmosdr_swig.py", line 28, in <module>
    _osmosdr_swig = swig_import_helper()
  File "/usr/local/lib/python2.7/dist-packages/osmosdr/osmosdr_swig.py", line 24, in swig_import_helper
    _mod = imp.load_module('_osmosdr_swig', fp, pathname, description)
ImportError: /usr/local/lib/libgnuradio-osmosdr.so.0.1.5git: undefined symbol: rtlsdr_set_bias_tee

velichkov, 2017-08-06:
Hi @Danikon,

As we discussed in [the google group](https://groups.google.com/forum/#!msg/gr-gsm/zRU715jaVvM/OJieEEs3BgAJ) I think you have two rtlsdr libraries installed and you are compiling with the new one while executing with the old

Please execute
```
find / -name librtlsdr.so.0
```
and then for every found library execute 
```
objdump -TC <path_to_lib>/librtlsdr.so.0 | grep rtlsdr_set_bias_tee
```
Also check which library is loaded by default
```
ldd /usr/local/lib/libgnuradio-osmosdr.so.0.1.5git
```
and then provide the full output of all of the above commands.

Danikon, 2017-08-07:
Ok working | problem solved

+++++++++++++++++++++++++++++++++++++++++

It's the library.

There are roughly 45 forks of the rtl-sdr software - only one is certain 
to work with gr-osmosdr, namely the library from
  https://git.osmocom.org/rtl-sdr

Uninstall it.

Either you can build it from source or use Ubuntu's.
  apt-get install rtl-sdr librtlsdr-dev

To verify Ubuntu's version is the correct version, namely, version 0.53, 
type
  apt list | grep librtlsdr-dev

+++++++++++++++++++++++++++++++++++++++++

Ok solved

Installed old version rtl-sdr(0.5.3) from ubuntu

closed 2017-08-07 by Danikon