ptrkrysik/gr-gsm #305: Unable to see decoded output of Hopping SDCCH8 channel

ppandit26, 2017-08-11:
I have a cfile captured at 2e6 at arfcn 79. In SI type message list of arfcns are 59,77,79.
I obtained the hopping parameters through Immediate assignment message: HSN:56 Timeslot:1 Training sequence Number:3 MAIO:0

I configured the receiver block for cell allocation list obtained from SI type 1 and Cx hopper block with MA:[55,77] HSN:56 MAIO:0.
On CX output I see GSMTAP messages for arfcns 59 and 77. After SDCCH8(Time slot 1) demapper block, I can observe the LAPDm messages in Wireshark. But I do not see any output after control channel decoder block.

[image]

[image]

[image]

Can anyone kindly provide me with some inputs or suggestions as to how can I proceed?

Thank You!

ppandit26, 2017-08-22:
An update to this issue:
I noticed that the arfcn(79) on which the cfile was captured also has to be included in the MA list.
The MA in Immediate assignment showed three arfcns including the main arfcn 79. I edited the flowgraph to take four inputs. One the wideband and other three for hopping(narrowband files on arfcns: 59, 77 ,79). 
Now I get the same output as shown in third image at the output of control channel decoder. But again I do not see ciphering mode command or other messages.

ppandit26, 2017-08-25:
Hello,
An update to this issue:
I can see the ciphering mode command, CM service request and system information messages at the output of the control channel decoder. (The output though is not consistent as it is taking very long to output the messages and every run gives different output).
I am unable to see any decrypted output. The ciphering command shows A5/3 algorithm. I do have the Kc. But I am unable to see any decrypted SMS on downlink.
[image]

closed 2017-09-01 by ppandit26

justhack21, 2017-10-30:
hello mam,
I am facing the same issue as you mentioned in issue 305. I am able to see LAPDm messages on wireshark demapper but nothing after control channel decoder.
Were you able to solve this. Can you please help me with this.

moemusaa, 2018-01-13:
hi , i have the same issue , can someone help us ?