ptrkrysik/gr-gsm #306: No module named osmosdr

RouchB, 2017-08-13:
Hi,

i'm getting the following error

--------------------------------------------------------------
root@kali:~# grgsm_livemon
Traceback (most recent call last):
  File "/usr/local/bin/grgsm_livemon", line 29, in <module>
    import osmosdr
ImportError: No module named osmosdr
--------------------------------------------------------------

can you assist me please

axilirator, 2017-08-15:
Please read the message: "No module named osmosdr".
This means that you don't have gr-osmosdr installed.
There is no reason to open an issue.
Just type "osmosdr" in Google and follow the installation guide.

RouchB, 2017-08-16:
Thanks sir,
i've googled it and went to the process till i got the following

[ 23%] Building CXX object lib/CMakeFiles/gnuradio-osmosdr.dir/file/file_sink_c.cc.o
[ 25%] Building CXX object lib/CMakeFiles/gnuradio-osmosdr.dir/rtl/rtl_source_c.cc.o
/root/gr-osmosdr/lib/rtl/rtl_source_c.cc: In constructor ‘rtl_source_c::rtl_source_c(const string&)’:
/root/gr-osmosdr/lib/rtl/rtl_source_c.cc:224:43: error: ‘rtlsdr_set_bias_tee’ was not declared in this scope
   ret = rtlsdr_set_bias_tee(_dev, bias_tee);
                                           ^
lib/CMakeFiles/gnuradio-osmosdr.dir/build.make:278: recipe for target 'lib/CMakeFiles/gnuradio-osmosdr.dir/rtl/rtl_source_c.cc.o' failed
make[2]: *** [lib/CMakeFiles/gnuradio-osmosdr.dir/rtl/rtl_source_c.cc.o] Error 1
CMakeFiles/Makefile2:135: recipe for target 'lib/CMakeFiles/gnuradio-osmosdr.dir/all' failed
make[1]: *** [lib/CMakeFiles/gnuradio-osmosdr.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2

i hope someone can assist with this
cheers!

bnt2025, 2017-09-03:
I have had this error when using the packaged version of rtl-sdr.
Compiling it manually has fixed this for me.

```bash
git clone git://git.osmocom.org/rtl-sdr.git
cd rtl-sdr
mkdir build
cd build
cmake ../ -DDETACH_KERNEL_DRIVER=ON -DINSTALL_UDEV_RULES=ON
make 
sudo make install 
sudo ldconfig -i
```

closed 2017-09-11 by ptrkrysik