ptrkrysik/gr-gsm #307: Can grgsm_livemon read from several frequencies at once?

petterreinholdtsen, 2017-08-15:
Hi.  I would like to read from several frequencies at once.  Is there some way to set up grgsm_livemon to do this?  Or is there a way to run several instances reading from the same hackrf unit?

E3V3A, 2017-08-16:
@petterreinholdtsen 
I suggest you read up on gr-gsm before posting more bug issues that are not really issues. You've filed multiple issues, from which it is clear that you have simply not bothered to even google the topics.  I'm sure you mean well, but the lack of respect to the developers and people who maintain this project is a sure way to get yourself banned.

petterreinholdtsen, 2017-08-16:
[E:V:A]
> @petterreinholdtsen
> I suggest you read up on gr-gsm before posting more bug issues that
> are not really issues. You've filed multiple issues, from which it is
> clear that you have simply not bothered [...]

Thank you for the feedback.  I am sorry you feel this way, and a bit
curious if this is representative for the gnuradio developer team as a
whole.

I'll consider removing myself from the Debian packaging as well (and
possibly as a consequence dropping the package from Debian), as my
experience with trying to maintain packages in Debian where the
relationship between the upstream developers and the debian maintainer
go sour is a energy sink and not worth the effort.

PS: For the record, while I am indeed new to using gr-gsm and not very
skilled at it, I have done several web searches when it was relevant to
try to locate the answers to my questions before creating github issues.

-- 
Happy hacking
Petter Reinholdtsen

E3V3A, 2017-08-16:
@petterreinholdtsen  I'm speaking from experience and for myself only, as I am not officially part of this dev team. Just searching previous already closed issues should have answered the question for you. Mind you that this github repo is not an FAQ site. Use the mailing list for that or make a request to update the related FAQ's [here](https://github.com/ptrkrysik/gr-gsm/wiki/Frequently-asked-questions). I agree that the documentation here is rather lacking, as it is in most other github projects, but the info is certainly out and around. Thanks for trying to help. Cheers!

petterreinholdtsen, 2017-08-16:
[E:V:A]
> I'm speaking from experience and for myself only, as I am not
> officially part of this dev team.

Duly noted.

> Just searching previous already closed issues should have answered the
> question for you.

If you really had the ability to predict the past and read minds, as
your repeated claim claim about what I would find if I did what you
propose to do, you would know that I did search already closed issues,
without finding the question answered.  So based on my personal
experience, your statement do not reflect realities.

> I agree that the documentation here is rather lacking, as it is in
> most other github projects, but the info is certainly out and
> around. Thanks for trying to help. Cheers!

Perhaps you can help out improving the documentation situation by
writing manual pages for the gr-gsm tools?  It is wanted by the Debian
project for all binaries, but is sadly lacking in gr-gsm.  It could even
answer how to read several frequencies at once, allowing you to point
people to the manual page in the future.

If you, as me, find the roff format hard to write, I warmly recommend
using rst2man to format the manual pages instead.  Writing rst is so
much easier. :)

-- 
Happy hacking
Petter Reinholdtsen

Keskebeu, 2018-02-21:
Hahaha answer NO

ptrkrysik, 2018-06-02:
@petterreinholdtsen, it is potentially possible to process other channels that is inside of the receiver bandwidth. But I wouldn't do it in grgsm_livemon app as it would complicate it. I would probably add example how to do it in gnuradio-companion instead, maybe with some tutorial on the wiki. Are you interested in such solution (still ;) )?

ghazimabdullah, 2019-01-29:
> @petterreinholdtsen, it is potentially possible to process other channels that is inside of the receiver bandwidth. But I wouldn't do it in grgsm_livemon app as it would complicate it. I would [...]

Hey, I was wondering if you have posted any wiki/solution to process multiple frequencies/channels in grgsm_livemon. Also it would be a great idea if it includes details on increasing bandwidth in grgsm_livemon and grgsm_capture

kushagrad28, 2019-08-07:
I would like to see this too. Maybe if you just explain it I will try to implement it. Just a description on how to implement it in gnuradio. I am thinking polyphase channeliser along with different switchable gsm recievers. But does not sounds efficient.

ptrkrysik, 2019-08-21:
Hi @kushagrad28. The first question is exactly what it would be for - what does it mean to "read from several frequencies at once". If it's for decoding C0 channel, then it shouldn't be hard in GNU Radio, but it'll be limited by amount of computations that will grow quite fast with bandwidth. I would test first something like rotator block followed by low-pass filter as polyphase channelizer is not too flexible (although efficient if you want to channelize everything).

If you mean to do frequency hopping this way it'll be harder and it'll be hard to reconfigure this live in GNU Radio.

ptrkrysik, 2026-11-09:
grgsm_livemon decodes a single C0 carrier; decoding several carriers needs a custom flowgraph (e.g. rotator + low-pass filter or channelizer in front of several receivers) as described by the maintainer in this thread, so we are closing this as a usage question.

I am closing this as part of a clean-up for the new release (v0.43.0), which ports gr-gsm to GNU Radio 3.10 (plain CMake build, no SWIG, installation steps in the README). If the problem still occurs with it, please open a new issue with your operating system, the output of `gnuradio-config-info --version`, how gr-gsm and its dependencies were installed, the exact command and the complete output.

closed 2026-11-09 by ptrkrysik