ptrkrysik/gr-gsm #308: grgsm_livemon - No stdout | grgsm_scanner no results

davezii, 2017-08-17:
Hi, I have installed gr-gsm (using git pull, as explained here https://github.com/ptrkrysik/gr-gsm/issues/276)

When running grgsm_scanner I'm getting no results.
When running grgsm_livemon, if though in the right freq, I don't see any data stream on the stdout
and nothing on the local network (sniffing with tshark).

**Sys Info**
Linux box 4.10.0-32-generic #36~16.04.1-Ubuntu SMP Wed Aug 9 09:19:02 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux

I'm using USRP b200 
gr-osmosdr 0.1.4 (0.1.4) 
gnuradio 3.7.9
GNU C++ version 5.3.1 20151219
Boost_105800
UHD_003.009.002-0-unknown

Thank you !

davezii, 2017-08-17:
This is the output after comminting out the stdout to dev/null

`root@box:/home/user/gr-gsm/apps# ./scan 
linux; GNU C++ version 5.3.1 20151219; Boost_105800; UHD_003.009.002-0-unknown

Args= 
gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.9
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy redpitaya 
-- Detected Device: B200
-- Operating over USB 3.
-- Initialize CODEC control...
-- Initialize Radio control...
-- Performing register loopback test... pass
-- Performing CODEC loopback test... pass
-- Asking for clock rate 16.000000 MHz... 
-- Actually got clock rate 16.000000 MHz.
-- Performing timer loopback test... pass
-- Setting master clock rate selection to 'automatic'.
-- Using subdev spec 'A:A'.
-- Asking for clock rate 32.000000 MHz... 
-- Actually got clock rate 32.000000 MHz.
-- Performing timer loopback test... pass
Using Volk machine: sse4_2_64_orc
set_min_output_buffer on block 5 to 4000000
set_min_output_buffer on block 4 to 4000000
OOOOOOOOOOOOOOOOOOOOOOOOOOOOOOArgs= 
gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.9
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy redpitaya 
-- Detected Device: B200
-- Operating over USB 3.
-- Initialize CODEC control...
-- Initialize Radio control...
-- Performing register loopback test... pass
-- Performing CODEC loopback test... pass
-- Asking for clock rate 16.000000 MHz... 
-- Actually got clock rate 16.000000 MHz.
-- Performing timer loopback test... pass
-- Setting master clock rate selection to 'automatic'.
-- Using subdev spec 'A:A'.
-- Asking for clock rate 32.000000 MHz... 
-- Actually got clock rate 32.000000 MHz.
-- Performing timer loopback test... pass
set_min_output_buffer on block 138 to 4000000
set_min_output_buffer on block 137 to 4000000
OOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOArgs= 
gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.9
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy redpitaya 
-- Detected Device: B200
-- Operating over USB 3.
-- Initialize CODEC control...
-- Initialize Radio control...
-- Performing register loopback test... pass
-- Performing CODEC loopback test... pass
-- Asking for clock rate 16.000000 MHz... 
-- Actually got clock rate 16.000000 MHz.
-- Performing timer loopback test... pass
-- Setting master clock rate selection to 'automatic'.
-- Using subdev spec 'A:A'.
-- Asking for clock rate 32.000000 MHz... 
-- Actually got clock rate 32.000000 MHz.
-- Performing timer loopback test... pass
set_min_output_buffer on block 271 to 4000000
set_min_output_buffer on block 270 to 4000000
OOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOArgs= 
gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.9
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy redpitaya 
-- Detected Device: B200
-- Operating over USB 3.
-- Initialize CODEC control...
-- Initialize Radio control...
[... 160 lines cut ...]
-- Actually got clock rate 32.000000 MHz.
-- Performing timer loopback test... pass
set_min_output_buffer on block 1468 to 4000000
set_min_output_buffer on block 1467 to 4000000
OOOOOOOOOOOOOOOOOOOOOOOOOOOOOOArgs= 
gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.9
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy redpitaya 
-- Detected Device: B200
-- Operating over USB 3.
-- Initialize CODEC control...
-- Initialize Radio control...
-- Performing register loopback test... pass
-- Performing CODEC loopback test... pass
-- Asking for clock rate 16.000000 MHz... 
-- Actually got clock rate 16.000000 MHz.
-- Performing timer loopback test... pass
-- Setting master clock rate selection to 'automatic'.
-- Using subdev spec 'A:A'.
-- Asking for clock rate 32.000000 MHz... 
-- Actually got clock rate 32.000000 MHz.
-- Performing timer loopback test... pass
set_min_output_buffer on block 1601 to 4000000
set_min_output_buffer on block 1600 to 4000000
OOOOOOOOOOOOOOOOOOOOOOOOOOOOOO
 `

petterreinholdtsen, 2017-08-19:
I had a similar situation when I was located in the basement of my house, where the radio signals apparently were too weak to be detected.  Moving up to the ground floor helped for me.

davezii, 2017-08-27:
I don't think this is the issue, I had the same thought in mind, I even tried that from a balcony.
I'm phone definitely had signal and after searching a bit it looks like it's related to issue #266
https://github.com/ptrkrysik/gr-gsm/issues/266

same behaviour - no results on the scanner

E3V3A, 2017-08-27:
It's impossible to help you without more info, such as **the complete command line you're using**. Little point in pasting 1000s of lines of repeated debug output. (Also please use the triple back-ticks ``` ` ``` for code markup. 

Normally it is ran with something like:
```
# grgsm_scanner -v -b P-GSM --args=rtl=0 -g 29.7 -p 1 -s 2000000.0
```
So where is your band specification, gain and sample rates specified?

petterreinholdtsen, 2017-08-27:
[E:V:A]
> Normally it is ran with something like:
> ```
> # grgsm_scanner -v -b P-GSM --args=rtl=0 -g 29.7 -p 1 -s 2000000.0

I suspect the normal case is different.  Normally I run it without any
arguments, and I would guess most users do the same, as the defaults
work just fine. :)

-- 
Happy hacking
Petter Reinholdtsen

davezii, 2017-08-27:
@E3V3A 

**These are the command line I was trying:**
`# grgsm_scanner -v`
` # grgsm_scanner -v -p 35`
` # grgsm_scanner -v  -p 35 -b <all options>`

* <all options> - I was enumerating over the possible options

E3V3A, 2017-08-30:
@petterreinholdtsen 
> Normally I run it without any arguments, and I would guess most users do the same, as the defaults work just fine. 

This is probably not true for users with non-pro SDRs like cheaper RTL ones where you need to carefully specify the gain and PPM in order to see anything useful. 

@davezii 
Check your antenna connections... (Duh! ;) But perhaps more importantly, the USB grounding. USB external enclosure should not be grounded with Signal GND. There are loads of info about grounding problems. In addition the effective sample rate (after usb packet loss) may be affected by a shitty USB hub, or if you're running it through a VM.

velichkov, 2017-09-21:
Hi @davezii,

>OOOOOOOOOOOOOOOOOOOOOOOOOOOOOOArgs= 

The multiple 'O' in the output indicates a [buffer overflow](https://files.ettus.com/manual/page_general.html#general_ounotes) problem so it seems that your host for some reason is not processing the input fast enough. Try to run `grgsm_scanner` with smaller sample rate `-s 0.8e6` or `-s1.2e6`

See also: 
https://kb.ettus.com/Verifying_the_Operation_of_the_USRP_Using_UHD_and_GNU_Radio#Benchmarking_your_system
https://kb.ettus.com/About_USRP_Bandwidths_and_Sampling_Rates
http://lists.ettus.com/pipermail/usrp-users_lists.ettus.com/2015-October/016263.html
http://digital.ni.com/public.nsf/allkb/CDDC3DE200B4A9D486257F8700761DF7

ptrkrysik, 2018-06-02:
I'm closing the issue here as it is more question than bug report. In case you would like to add something to this discussion please do it on the mailing-list and link this issue.

closed 2018-06-02 by ptrkrysik