ptrkrysik/gr-gsm #309: "Random" decoding result?

Alster, 2017-08-19:
Hello.
What can influence the "randomness" of the decoding result?

I run the decoding of the same file, but I get different results at the same time.

test.sh
```
echo
echo "-------------------------------------------------------------------------------"
echo " DIFF"
echo "-------------------------------------------------------------------------------"
echo
grgsm_decode -c capture -s 200e3 -f 1810600000 -p > A
grgsm_decode -c capture -s 200e3 -f 1810600000 -p > B
diff A B
```
Sample result:
```
110c110
< 713766 1102510 0: 110010101101000111100111010100110000101110100011010011101101010001001111001001100101011100101010111100010011100101
---
> 713766 1102510 0: 110010101101000111100111010100110000101110100011010011101101010001001111001001100101011100101010111100010011101010
140c140
< 714439 1102827 0: 110010101101000101100111010100110000100010100011010011101101010001001111011000011110001110010111110011000110001100
---
> 714439 1102827 0: 110010101011100101100111010100111000100010100011010001001010111001000111011100010110001110010101110011100110001100
150c150
< 714449 1103157 0: 110010111001000101100111010100110011000110100011010011101101010001001111001000111011010011101011011001011011010100
---
> 714449 1103157 0: 110010111001000101100111010100110011000110100011010011001101010001001111001000111011010011101011011001011011010100
170c170
< 714470 1102192 0: 110010101011100101100010101100110001010101110011010001101010111001001011001000101100101001011011000101101011010100
---
> 714470 1102192 0: 110100101011100001100010101100110001010101110010110001000010111011000011001000101101111000011011000101101011110100
200c200
< 714500 1103156 0: 110010111001000101100111010100110011001010100011010011101101010001001111001001110100010011100101101011010100101111
---
> 714500 1103156 0: 110010111001000101100111010100110011001010100011010011101001010001001111001001110100010011100101101011010100101111
230c230
< 714531 1102521 0: 110010101001000111100111010100110000101000100011010011101101010001001111001001001110101010010001110101101010101011
---
> 714531 1102521 0: 110010101001000111100111010100110000101000100011010011101101010001001111001001001110101010010001110111101010101011
240c240
< 714878 1104232 0: 110010101011100101100111010100110001010101110011010001101010111001000111001001110011001010110111010100110001110011
---
> 714878 1104232 0: 110010101101000111100111010100110010101110100011010011101101010001001111001001110010001010010110001000110001100011
```

[The result of running a lot of calls test.sh](https://drive.google.com/open?id=0Bwqq-p_P2qhHd1MtX0F4dlhnU28)

It can be seen that some differences are often repeated.

What do you think, what can be the reason?

ptrkrysik, 2017-09-04:
Hi, I answered the question here: https://groups.google.com/forum/#!topic/gr-gsm/gve4CcJubgw

closed 2017-09-04 by ptrkrysik

E3V3A, 2017-09-05:
For reference, the answer is: 

> It can happen if you use frequency offset correction loop in gr-gsm.
Messages that control correction are currently asynchronous so timing of
correction isn't currently deterministic. You can disable frequency
correction loop if you have device that doesn't require it. For
grgsm_decode you do this by not giving carrier frequency parameter.
> 
> In gnuradio-companion you can do this by disabling clock offset control
block.