ptrkrysik/gr-gsm #312: Implement headless grgsm_livemon?

petterreinholdtsen, 2017-08-22:
Hi.

I came across this version of grgsm_livemon, which is a version where the X gui has been ripped out and the program work in a terminal: https://gist.github.com/CreateRemoteThread/41e59ef171403e73e3cbbaa71500301f .

As far as I can tell, it is based on the version in commit fe538eba62623cb4462bd4a2085ab754b5121ffd .

Would you be interested in providing a way to run grgsm_livemon in headless mode, for those of us that want to use it in a background script?

petterreinholdtsen, 2017-08-26:
I had a go based on input from Piotr, and this patch to the grc file convert grgsm_livemon into a command line tool.  It change the project type, remove the QT blocks and change the SDR block to use the command line parameters instead of the values from the QT blocks.

Is there some way to get the same effect using a command line argument, say --nogui?

```
diff --git a/apps/grgsm_livemon.grc b/apps/grgsm_livemon.grc
index 40bee70..3a9595f 100644
--- a/apps/grgsm_livemon.grc
+++ b/apps/grgsm_livemon.grc
@@ -38,7 +38,7 @@
     </param>
     <param>
       <key>generate_options</key>
-      <value>qt_gui</value>
+      <value>no_gui</value>
     </param>
     <param>
       <key>hier_block_src_path</key>
@@ -82,195 +82,6 @@
     </param>
   </block>
   <block>
-    <key>variable_qtgui_range</key>
-    <param>
-      <key>comment</key>
-      <value></value>
-    </param>
-    <param>
-      <key>value</key>
-      <value>fc</value>
-    </param>
-    <param>
-      <key>_enabled</key>
-      <value>True</value>
-    </param>
-    <param>
-      <key>_coordinate</key>
-      <value>(304, 11)</value>
-    </param>
-    <param>
-      <key>gui_hint</key>
-      <value></value>
-    </param>
-    <param>
-      <key>_rotation</key>
-      <value>0</value>
-    </param>
-    <param>
-      <key>id</key>
-      <value>fc_slider</value>
-    </param>
-    <param>
-      <key>label</key>
-      <value>Frequency</value>
-    </param>
-    <param>
-      <key>min_len</key>
-      <value>100</value>
-    </param>
-    <param>
-      <key>orient</key>
-      <value>Qt.Horizontal</value>
-    </param>
-    <param>
-      <key>start</key>
-      <value>800e6</value>
-    </param>
-    <param>
-      <key>step</key>
-      <value>2e5</value>
[... 450 lines cut ...]
     <param>
       <key>gain_mode0</key>
@@ -1427,7 +951,7 @@
     </param>
     <param>
       <key>gain0</key>
-      <value>g_slider</value>
+      <value>gain</value>
     </param>
     <param>
       <key>ant10</key>
@@ -2883,12 +2407,6 @@
     <sink_key>0</sink_key>
   </connection>
   <connection>
-    <source_block_id>blocks_rotator_cc_0</source_block_id>
-    <sink_block_id>qtgui_freq_sink_x_0</sink_block_id>
-    <source_key>0</source_key>
-    <sink_key>0</sink_key>
-  </connection>
-  <connection>
     <source_block_id>gsm_bcch_ccch_sdcch4_demapper_0</source_block_id>
     <sink_block_id>gsm_control_channels_decoder_0</sink_block_id>
     <source_key>bursts</source_key>
```

petterreinholdtsen, 2017-08-28:
I discovered that to have a better user experience, the 'prompt' run_option choice is a bad one.  It is better to use the 'run' value, like this:

```
diff --git a/apps/grgsm_livemon.grc b/apps/grgsm_livemon.grc
index 3342fb6..1b9a935 100644
--- a/apps/grgsm_livemon.grc
+++ b/apps/grgsm_livemon.grc
@@ -66,7 +66,7 @@
     </param>
     <param>
       <key>run_options</key>
-      <value>prompt</value>
+      <value>run</value>
     </param>
     <param>
       <key>run</key>
```

With the 'prompt' choice, it is hard to place the livemon process in the background, as it blocks waiting for user input from stdin.

ptrkrysik, 2017-08-30:
Done by you @petterreinholdtsen in pull request #321

closed 2017-08-30 by ptrkrysik