ptrkrysik/gr-gsm #324: Self testing fail (OSError: [Errno 17] File exists: './.unittests')

petterreinholdtsen, 2017-08-29:
When building the Debian package, the build fail because of a failing test:

```
Running tests...
/usr/bin/ctest --force-new-ctest-process -j4
Test project /uio/kant/usit-gid-u1/pre/src/debian/gr-gsm-upstream/obj-x86_64-linux-gnu
      Start  1: qa_decryption
      Start  2: qa_burst_printer
      Start  3: qa_message_printer
      Start  4: qa_burst_timeslot_splitter
 1/10 Test  #1: qa_decryption .....................***Failed    0.70 sec
Traceback (most recent call last):
  File "/uio/kant/usit-gid-u1/pre/src/debian/gr-gsm-upstream/python/qa_decryption.py", line 202, in <module>
    gr_unittest.run(qa_decryption, "qa_decryption.xml")
  File "/usr/lib/python2.7/dist-packages/gnuradio/gr_unittest.py", line 127, in run
    os.makedirs(basepath, 0750)
  File "/usr/lib/python2.7/os.py", line 157, in makedirs
    mkdir(name, mode)
OSError: [Errno 17] File exists: './.unittests'

      Start  5: qa_burst_sdcch_subslot_splitter
 2/10 Test  #4: qa_burst_timeslot_splitter ........***Failed    0.80 sec
Traceback (most recent call last):
  File "/uio/kant/usit-gid-u1/pre/src/debian/gr-gsm-upstream/python/qa_burst_timeslot_splitter.py", line 161, in <module>
    gr_unittest.run(qa_burst_timeslot_splitter, "qa_burst_timeslot_splitter.xml")
  File "/usr/lib/python2.7/dist-packages/gnuradio/gr_unittest.py", line 127, in run
    os.makedirs(basepath, 0750)
  File "/usr/lib/python2.7/os.py", line 157, in makedirs
    mkdir(name, mode)
OSError: [Errno 17] File exists: './.unittests'

 3/10 Test  #3: qa_message_printer ................***Failed    0.90 sec
Traceback (most recent call last):
  File "/uio/kant/usit-gid-u1/pre/src/debian/gr-gsm-upstream/python/qa_message_printer.py", line 165, in <module>
    gr_unittest.run(qa_message_printer, "qa_message_printer.xml")
  File "/usr/lib/python2.7/dist-packages/gnuradio/gr_unittest.py", line 127, in run
    os.makedirs(basepath, 0750)
  File "/usr/lib/python2.7/os.py", line 157, in makedirs
    mkdir(name, mode)
OSError: [Errno 17] File exists: './.unittests'

      Start  6: qa_burst_timeslot_filter
      Start  7: qa_burst_sdcch_subslot_filter
 4/10 Test  #6: qa_burst_timeslot_filter ..........   Passed    1.90 sec
      Start  8: qa_burst_fnr_filter
 5/10 Test  #5: qa_burst_sdcch_subslot_splitter ...   Passed    3.41 sec
      Start  9: qa_dummy_burst_filter
 6/10 Test  #7: qa_burst_sdcch_subslot_filter .....   Passed    3.41 sec
      Start 10: qa_arfcn
 7/10 Test #10: qa_arfcn ..........................   Passed    0.40 sec
 8/10 Test  #9: qa_dummy_burst_filter .............   Passed    1.61 sec
 9/10 Test  #8: qa_burst_fnr_filter ...............   Passed    3.41 sec
10/10 Test  #2: qa_burst_printer ..................   Passed    8.77 sec

70% tests passed, 3 tests failed out of 10

Total Test time (real) =   8.84 sec

The following tests FAILED:
          1 - qa_decryption (Failed)
          3 - qa_message_printer (Failed)
          4 - qa_burst_timeslot_splitter (Failed)
Errors while running CTest
Makefile:152: recipe for target 'test' failed
make[1]: *** [test] Error 8
make[1]: Leaving directory '/uio/kant/usit-gid-u1/pre/src/debian/gr-gsm-upstream/obj-x86_64-linux-gnu'
dh_auto_test: make -j4 test ARGS+=-j4 returned exit code 2
debian/rules:7: recipe for target 'build' failed
make: *** [build] Error 2
dpkg-buildpackage: error: debian/rules build gave error exit status 2
debuild: fatal error at line 1116:
dpkg-buildpackage -rfakeroot -us -uc failed
```

Do you have any idea why it fail and how to avoid the failure?

ptrkrysik, 2017-08-30:
I don't have idea at the moment. Can you provide receipt how to reproduce this problem?

petterreinholdtsen, 2017-08-30:
[Piotr Krysik]
> I don't have idea at the moment. Can you provide receipt how to
> reproduce this problem?

Sure.  Install Debian Stretch, clone gr-gsm, run
./dists/debian/build-package (and install its build dependencies before
trying again. :)

-- 
Happy hacking
Petter Reinholdtsen

petterreinholdtsen, 2017-08-30:
I had a closer look at the failing code, and in gr_unittest.py I find
this piece in line 126:

        if not os.path.exists(basepath):
            os.makedirs(basepath, 0750)

There is no way this could fail, unless some other thread is creating
the requested directory at the same time - ie a race condition.  Perhaps
this is triggered by running the tests in parallel, caused by the use of
'dh ... --parallel' in debian/rules?  Can the test suite be changed to
avoid any race conditions?

-- 
Happy hacking
Petter Reinholdtsen

petterreinholdtsen, 2017-08-31:
I tested the --parallel hypothesis, and did two parallel builds where
the tests failed, and two non-parallel builds where the tests succeeded.
This is a strong indicator that the tests can not run in parallel.  But
I was unsuccessful trying to reproduce it outside the build using this
command (same as the build is running):

  make -C obj-x86_64-linux-gnu/ -j4 test ARGS\+=-j4

I can avoid the problem by disabling parellel execution of the test
suite when building the Debian package like this, but is it the correct
approach?

diff --git a/dists/debian/rules b/dists/debian/rules
index d99b9e5..8bbd44d 100755
--- a/dists/debian/rules
+++ b/dists/debian/rules
@@ -9,6 +9,9 @@ export DEB_HOST_MULTIARCH
 override_dh_auto_configure:
        dh_auto_configure -- -DLIB_SUFFIX="/$(DEB_HOST_MULTIARCH)" -DPythonLibs_FIND_VERSION:STRING="2.7" -DPYTHON_EXECUTABLE:STRING="/usr/bin/python"
 
+override_dh_auto_test:
+       dh_auto_test --no-parallel
+
 override_dh_auto_install:
        dh_auto_install
 

All this testing is done using gnuradio version 3.7.10.1.  I further
tested with gnuradio version 3.7.11 in a chroot on a different machine,
and here I am unable to reproduce the problem.  It is a older machine,
so the problem might require a quicker CPU and IO system to trigger.

-- 
Happy hacking
Petter Reinholdtsen

E3V3A, 2017-08-31:
> It is a older machine, so the problem might require a quicker CPU and IO system to trigger

This is a common problem. Single and dual core machines should avoid building with `-j4`, since obviously you can't run 4 parallel tasks on a single core, unless its core has some kind of thread parallelization enabled. This is especially evident when compiling on VMs or Raspberries 1-2.

ptrkrysik, 2018-06-02:
@petterreinholdtsen, is this problem still present?

closed 2018-07-26 by ptrkrysik