ptrkrysik/gr-gsm #325: Can't aquire any data whatsoever

13lackWidow, 2017-08-31:
This may make me seem like a fool, but nonetheless, I have no idea where else to go.

I'm unable to aquire any data whatsoever with every one of these programs, although the sdr dongle I'm using (the Hack RF One) should be perfectly capable of recieving GSM850.  I have followed many guides to no avail, even going as far as trying to use airprobe with an old version of GRC.

Currently, here is my setup:
OS: Debian 9 x64 in a VM

Programs:
Most recent Gnu Radio Companion (Debian repos)
Most recent [Kalibrate_hackrf](https://github.com/scateu/kalibrate-hackrf)
Most recent gr-gsm with [this patch](https://github.com/gnuradio/gnuradio/pull/1424/files)
Most recent version of wireshark (Debian repos)

Problem: When running grgsm_livemon and tuning to the proper frequency & setting the ppm according to kalibrate (as well as increasing/decreasing it on my own), I am unable to see any GSM packets whatsoever. I fire up wireshark and run "sudo wireshark -k -f udp -Y gsmtap -i lo". Nothing.
I've got abolutely no idea how to fix it or where to go from here and I figured since you guys are the masters on this, I'd ask here.

If it helps, when I run grgsm_capture.py, I recieve this output, then it just hangs there:

```
sdruser@DEBIAN9SDR:~/SDR$ grgsm_capture.py -f 87960000 -p 22 -b test_burstz -v 
linux; GNU C++ version 6.3.0 20170221; Boost_106200; UHD_003.009.005-0-unknown 

gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.10 built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy soapy redpitaya 
Cannot connect to server socket err = No such file or directory 
Cannot connect to server request channel jack server is not running or cannot be started JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for 4294967295, skipping unlock 
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for 4294967295, skipping unlock 

RtApiAlsa::getDeviceInfo: snd_pcm_open error for device (hw:0,0), Device or resource busy.

Number of USB devices: 4 
USB device 1d50:6089: 0000000000000000909864c82c2334cf match 
Using HackRF One with firmware 2015.07.2 
```

petterreinholdtsen, 2017-08-31:
Did you get the frequency you are using from grgsm_scanner?

13lackWidow, 2017-08-31:
No, I the frequency I'm using is a base station frequency recieved from Kalibrate. grgsm_scanner won't even work and all I get from that is a Segmentation fault.

petterreinholdtsen, 2017-09-01:
[13lackWidow]
> No, I the frequency I'm using is a base station frequency recieved
> from Kalibrate. grgsm_scanner won't even work and all I get from that
> is a Segmentation fault.

Perhaps you have other problems that mane gr-gsm fail completely.
grgsm_scanner work here, and when I use the frequencies it show I get
data from grgsm_livemon.  I use the packaging branch on Debian Stretch.

-- 
Happy hacking
Petter Reinholdtsen

Alster, 2017-09-04:
@13lackWidow 

> JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for 4294967295, skipping unlock 

At me such was when installed on the virtual machine.
I still did not understand what leads to this error, but it only came to me once.

E3V3A, 2017-09-04:
It's probably a QT problem when using a VM. See #282. Then try either or both `--args=hackrf` and/or add `QT_X11_NO_MITSHM=1` before anything with sudo. When using a VM the USB is not accessed in the same way as natively. This can lead to a lot of lost packets...

ptrkrysik, 2017-09-04:
@13lackWidow , VM's are known to cause additional problems. I can solve problems caused by gr-gsm. Please install it on non-virtual machine and then try to decode something. In case you can't I will reopen the issue. 

It might be some issue with gr-gsm+hackrf. When you'll have gr-gsm on normal machine, try to use this modified version of grgsm_livemon:
https://gist.githubusercontent.com/ptrkrysik/c1a22075a56c2025f8397f76b205733e/raw/b63914543c670f82a0855f184a53f43fc27485ff/grgsm_livemon_hackrf.grc

closed 2017-09-04 by ptrkrysik