ptrkrysik/gr-gsm #327: Cant rise grgsm_livemon above 199900000

kpeezy707, 2017-09-02:
Trying to use this in the US and just wondering if that is the max frequency allowed? I'm trying to capture IMT which is between 2100 - 2170. Just wondering if i can change that somehow?

Thanks

bnt2025, 2017-09-03:
Hi there, 
The IMT-2100 band is primarily used for UMTS and as such gr-gsm won't work.

The main GSM bands for the US are;
    GSM850. (Uplink 824.2M - 848.8M / Downlink 869.2M - 893.8M)
    GSM1900. (Uplink 1850.2M - 1909.8M / Downlink 1930.2M - 1989.8 M)

See https://en.wikipedia.org/wiki/GSM_frequency_bands for more info

ptrkrysik, 2017-09-04:
@kpeezy707 , you can. Edit https://github.com/ptrkrysik/gr-gsm/blob/master/apps/grgsm_livemon.grc file in gnuradio-companion and change in fc_slider parameter block stop value to whatever you like.

closed 2017-09-04 by ptrkrysik