ptrkrysik/gr-gsm #329: Fix easy error in "grgsm_capture" with "-p / --ppm" no float type value

Nau-84, 2017-09-08:
Hello

The command "grgsm_capture" returns an error, when using a value for the "-p / --ppm" option of number with "comma", only allows to use whole numbers.
Unlike the "grgsm_livemon" command that supports non-integer values.

I corrected this error by defining the value type as "eng_float" in the "grgsm_capture".

I send you congratulations for your great project and work you have done with gr-gsm.

Greetings from Argentina.

Nau.

Nau-84, 2017-09-09:
Hello

In command "grgsm_capture" when type "not integer" value for "-p / --ppm" option, i have error not run command.

In "grgsm_livemon" support OK "not integer" value for "-p / --ppm" option.

I fix this problem, i change in "grgsm_capture" the type of value for "eng_float" for "-p / --ppm" option.

I send you a greeting from Argentina and congratulations on your great work.

Nau

ptrkrysik, 2017-09-11:
Hi Nau, this was done somewhat intentionally as the underlying hardware
(RTL-SDR) supports only integer values of corrections. To apply
non-integer correction - it would have to be done by gsm_input block, so
the change is not as simple as changing parameter type.

Nau-84, 2017-09-11:
Hi, there was actually a reason for the value type to be only integer. But I did not understand why in a command if you can use non-integer values ​​and in another not. Well thank you for your attention. regards

ptrkrysik, 2017-09-12:
In other applications the fractional part is applied somewhere. But you made me think about grgsm_capture application and I think about replacing it with something simpler.

closed 2017-09-12 by ptrkrysik