ptrkrysik/gr-gsm #331: Error: libcrypto required.

rootb3, 2017-09-10:
Hey, when I try to install `sudo pybombs -v install gr-gsm` it returns me the following error:-

`checking for BN_init in -lcrypto... no
configure: error: "Error: libcrypto required."
PyBOMBS.monitor_process() - DEBUG - Thread signaled termination or returned
PyBOMBS.monitor_process() - DEBUG - Return value: 1
PyBOMBS.Packager.source - ERROR - Configuration failed after running at least twice.
PyBOMBS.Packager.source - ERROR - Problem occurred while building package apache-thrift:
Configuration failed
PyBOMBS.install_manager - ERROR - Error installing package apache-thrift. Aborting.`

whats the solution to fix this error?

petterreinholdtsen, 2017-09-11:
Sound like something is unable to find a openssl version it like.  Which platform are you building on?  Note, gr-gsm is now available in Debian Unstable, https://tracker.debian.org/pkg/gr-gsm .

petterreinholdtsen, 2017-09-12:
This error is the same as the one reported in #283.

closed 2017-09-12 by ptrkrysik