ptrkrysik/gr-gsm #332: no idia

toombr, 2017-09-11:
root@USER:/home/user/gr-gsm/build# grgsm_capture -a 650 -c test
Traceback (most recent call last):
  File "/home/YOUR_HOME_DIR/gr-gsm/bin/grgsm_capture", line 243, in <module>
    if grgsm.arfcn.is_valid_arfcn(arfcn, band):
TypeError: is_valid_arfcn() takes exactly 1 argument (2 given)

the same problem installing with pybombs!

toombr, 2017-09-11:
by the way... everything else(grgsm_scan, live) works! everything excepting that lines which have to contain -a (or -f)  if i'm entering value i see message below!

ptrkrysik, 2017-09-11:
Hi, the problem should be resolved by the recent changes in the master branch. In case some problems presist - please reopen the issue.

closed 2017-09-11 by ptrkrysik