ptrkrysik/gr-gsm #334: Cant extract voice from a wideband capture

newuser55555, 2017-09-16:
Hi,
I am new to this gsm thing and have limited knowledge. I was trying to reproduce the freq hopping decode but i am unable to extract voice.

I have done the following,
- captured a wideband cfile
- used channelizer to get individual channels
- made a grc file to get Cx burst file
- Now when i try to extract voice i get nothing
- i have checked the CA, HSN ,TSEQ and TS values
- when i see the burst file after Cx hopper in wireshark i can see that its combining all the arfcn packets
plz help

justhack21, 2017-09-20:
hi buddy,  seems all fine to me. Can you share your cfile and the grc files. I ll see if i can help

justhack21, 2017-09-20:
.

newuser55555, 2017-09-20:
got it resolved on my own brother...anyways thanks...it was a minor issue

newuser55555, 2017-09-20:
do you have any idea if AMR can be decoded

justhack21, 2017-09-20:
Yes it is very much possible. If you have some cfiles with AMR coded audio I ll help you out with them.

b4ss3k, 2018-02-05:
hi, have a question, how you capture in wideband cfile? what hardware do you use?

ptrkrysik, 2018-06-02:
Hi guys, some rates of AMR can be decoded (i.e. 12.2kbit/s), some can't. In case you want to continue the discussion do it here: #226.

closed 2018-06-02 by ptrkrysik