ptrkrysik/gr-gsm #335: Please make 'install' target install .desktop file in /usr/share/applications/

petterreinholdtsen, 2017-09-17:
It would make it easier to package gr-gsm in Linux distributions if the 'install' make target place the .desktop file in /usr/share/applications/ where the desktop environments are looking for it.

On Debian I modified the build system like this to adjust the 'install' target:

```
--- gr-gsm-0.41.1.orig/apps/apps_data/CMakeLists.txt
+++ gr-gsm-0.41.1/apps/apps_data/CMakeLists.txt
@@ -22,7 +22,7 @@ set(grgsm_freedesktop_path ${GR_PKG_DATA
 install(
    FILES
    grgsm-livemon.desktop
-   DESTINATION ${grgsm_freedesktop_path}
+   DESTINATION /usr/share/applications
    COMPONENT "gr-gsm"
 )
 
@@ -34,8 +34,4 @@ if(UNIX AND HAVE_XDG_UTILS)
         ${CMAKE_CURRENT_SOURCE_DIR}/grgsm_setup_freedesktop.in
         ${CMAKE_CURRENT_BINARY_DIR}/grgsm_setup_freedesktop
     @ONLY)
-    install(
-        PROGRAMS ${CMAKE_CURRENT_BINARY_DIR}/grgsm_setup_freedesktop
-        DESTINATION ${GR_PKG_LIBEXEC_DIR} COMPONENT "gr-gsm"
-    )
 endif(UNIX AND HAVE_XDG_UTILS)
```

I guess a more complete patch would remove all traces of grgsm_setup_freedesktop, which no longer have any purpose if the file is placed in the final location after installation.

ptrkrysik, 2018-06-02:
As I said once - if the install prefix is set to place other than /usr I don't want install scripts to write there anyway. In my opinion the solution should be to set /usr as install prefix in your case but I'm not sure how it looks from packager's perspective. I know this issue is really old now, but I'm really interested to open this discussion again.

petterreinholdtsen, 2018-06-03:
I do not quite understand what you propose.  Can you provide more details, to allow me to test the effect of your proposal to set /usr/ as install prefix, as I thought I was already doing this when making the Debian package.

velichkov, 2018-06-03:
Hi @petterreinholdtsen,

The `apps/apps_data/CMakeLists.txt` is based on [gnuradio/grc/scripts/freedesktop/CMakeLists.txt](https://github.com/gnuradio/gnuradio/blob/master/grc/scripts/freedesktop/CMakeLists.txt) 

> I guess a more complete patch would remove all traces of grgsm_setup_freedesktop, which no longer have any purpose if the file is placed in the final location after installation.

The purpose of the `grgsm_setup_freedesktop` script is to install the desktop file into the correct location and to do this it calls `xdg-desktop-menu` which depending on the user install the desktop file in `/usr/share/...` or `~/.local/share/...`. From the `xdg-desktop-menu` man page 
```
       --mode mode
           mode can be user or system. In user mode the file is (un)installed for the current user only. In system mode the file is (un)installed
           for all users on the system. Usually only root is allowed to install in system mode.

           The default is to use system mode when called by root and to use user mode when called by a non-root user.
```
Maybe instead of patching the CMakeLists.txt you could call this script from the postinst script of the package or call it in `override_dh_auto_install`. Probably you will also need to add `xdg-utils` to the `Build-Depends` as well.

ptrkrysik, 2018-06-04:
Hi @petterreinholdtsen,

The background is that I'm doing cleanup of the github issues and trying
to recall what different things were about.

So what basically you mean is that it works the way it is and we don't
need to change anything?

petterreinholdtsen, 2018-06-04:
Nope, that is not what I mean.  I mean I believe the package build system is setting the prefix as you propose, and the desktop file is still installed in the wrong location.

A Debian package is not built as root (for security reasons), and thus the assumtions of xdg-desktop-menu do not match reality.

As a packager, I want 'make install DESTDIR=some/where' to make sure the desktop file end up in some/where/usr/share/applications/, to ensure it is included in the package in its appropriate location.  Installing desktop files using xdg-desktop-menu in the postinst is not appropriate, as it causes automatic package checking and processing systems to fail to see the desktop file (think AppStream and apt-file).

ptrkrysik, 2018-06-04:
Ok, I'm disabling current installation as installing desktop files
correctly in packages is most important ( 7fd5b6f7eeccd498b ).

Do you know how it works for gnuradio? I see only this patch that is
related to desktop files installation:
https://salsa.debian.org/bottoms/pkg-gnuradio/blob/unstable/debian/patches/direct-freedesktop-install

petterreinholdtsen, 2018-06-04:
[Piotr Krysik]
> Do you know how it works for gnuradio? I see only this patch that is
> related to desktop files installation:
> https://salsa.debian.org/bottoms/pkg-gnuradio/blob/unstable/debian/patches/direct-freedesktop-install

I had not looked at gnuradio packaging before you sent this salsa link.
The patch in question seem to do similar things to my patch for gr-gsm,
ie make sure the desktop file is installed in the correct location.

-- 
Happy hacking
Petter REinholdtsen

closed 2026-11-09 by ptrkrysik