ptrkrysik/gr-gsm #338: gr-gsm help

shepardac, 2017-09-23:
**Hello, I am in process of getting my LimeSDR to run with gr-gsm. I have all the programs installed correctly (works with my RTL, however when I try to use the LimeSDR I get the following message: I was wondering if any of yall could give me some help, thank you. ** 

UHD Error:
    setBandwidth(Rx, 0, 0.65 MHz) Failed - RxLPF frequency out of range, available range from 1.4 to 130 MHz
Traceback (most recent call last):
  File "/usr/local/bin/grgsm_livemon", line 370, in <module>
    main()
  File "/usr/local/bin/grgsm_livemon", line 358, in main
    tb = top_block_cls(args=options.args, collector=options.collector, collectorport=options.collectorport, fc=options.fc, gain=options.gain, osr=options.osr, ppm=options.ppm, samp_rate=options.samp_rate, serverport=options.serverport, shiftoff=options.shiftoff)
  File "/usr/local/bin/grgsm_livemon", line 128, in __init__
    self.rtlsdr_source_0.set_bandwidth(250e3+abs(shiftoff), 0)
  File "/usr/local/lib/python2.7/dist-packages/osmosdr/osmosdr_swig.py", line 1792, in set_bandwidth
    return _osmosdr_swig.source_sptr_set_bandwidth(self, bandwidth, chan)
RuntimeError: RxLPF frequency out of range, available range from 1.4 to 130 MHz
libusb: warning [libusb_exit] application left some devices open

**Now I have installed all the UHD, Soapy UHD, GNU Radio etc. when I enter in the find UHD devices , etc I get the following:** 

SoapySDRUtil --probe
######################################################
## Soapy SDR -- the SDR abstraction library
######################################################

Probe device 
linux; GNU C++ version 5.4.0 20160609; Boost_105800; UHD_003.010.002.000-3-g122bfae1

[INFO] Make connection: 'LimeSDR-USB [USB 3.0] 9062000C4251A'
[INFO] Reference clock 30.720 MHz
[INFO] Device name: LimeSDR-USB
[INFO] Reference: 30.72 MHz
[INFO] Init LMS7002M(0)
[INFO] Ver=7, Rev=1, Mask=1
[INFO] LMS7002M calibration values caching Disable

----------------------------------------------------
-- Device identification
----------------------------------------------------
  driver=STREAM
  hardware=LimeSDR-USB
  boardSerialNumber=0x9062000c4251a
  firmwareVersion=3
  hardwareVersion=4
  protocolVersion=1

----------------------------------------------------
-- Peripheral summary
----------------------------------------------------
  Channels: 2 Rx, 2 Tx
  Timestamps: YES
  Sensors: clock_locked, lms7_temp
  GPIOs: MAIN

----------------------------------------------------
-- RX Channel 0
----------------------------------------------------
  Full-duplex: YES
  Supports AGC: NO
  Stream formats: CF32, CS12, CS16
  Native format: CS16 [full-scale=2048]
  Stream args:
     * Buffer Length - The buffer transfer size over the link.
       [key=bufferLength, units=samples, default=0, type=int]
     * Link Format - The format of the samples over the link.
       [key=linkFormat, default=CS16, type=string, options=(CS16, CS12)]
  Antennas: NONE, LNAH, LNAL, LNAW, LB1, LB2
  Corrections: DC removal, IQ balance
  Full gain range: [-12, 61] dB
    LNA gain range: [0, 30] dB
    TIA gain range: [0, 12] dB
    PGA gain range: [-12, 19] dB
[... 210 lines cut ...]
|   |   |   |   Uses LO offset: No
|   |   |     _____________________________________________________
|   |   |    /
|   |   |   |       TX Codec: 0
|   |   |   |   Name: SoapyDAC
|   |   |   |   Gain Elements: None
|   |     _____________________________________________________
|   |    /
|   |   |       TX Dboard: 1
|   |   |     _____________________________________________________
|   |   |    /
|   |   |   |       TX Frontend: 1
|   |   |   |   Name: SoapyRF
|   |   |   |   Antennas: NONE, BAND1, BAND2
|   |   |   |   Sensors: lo_locked
|   |   |   |   Freq range: 30.000 to 3800.000 MHz
|   |   |   |   Gain range PAD: -52.0 to 0.0 step 0.1 dB
|   |   |   |   Bandwidth range: 800000.0 to 60000000.0 step 12000000.0 Hz
|   |   |   |   Connection Type: IQ
|   |   |   |   Uses LO offset: No
|   |   |     _____________________________________________________
|   |   |    /
|   |   |   |       TX Codec: 1
|   |   |   |   Name: SoapyDAC
|   |   |   |   Gain Elements: None

velichkov, 2017-10-05:
> RuntimeError: RxLPF frequency out of range, available range from 1.4 to 130 MHz

Try with bigger sample rate `--samp-rate=1.4M` or do not specify it at all as usually the default is 2M

zodiac-tiger, 2017-10-12:
I'm sure this is not the best fix for this, but instead of changing your sample rate, look in your Osmocom Source block and jack the Ch0 Bandwidth up to a value higher than 1.4 Mhz.  For simplicity, I usually just multiply the default formula by three like this --> 250e3+abs(shiftoff) * 3 .

ptrkrysik, 2018-04-11:
To minimize number of issues and to make the project more manageable for me I close the issue here and continue the topic in issue #393.

closed 2018-04-11 by ptrkrysik