ptrkrysik/gr-gsm #340: FR Voice Decoder not working

ekagero, 2017-09-26:
Hello,

I'm running gr-gsm on Kali 2.0. All capturing and decoding steps seem to work fine but when trying to decode TCHF FR I get an empty (66B) speech file. Funnily the AMR12.2 results in a nearly not understandable but working voice file.

Not sure if this is connected; when capturing and hanging up from the called phone it throws a python2 error:

[INFO] [UHDlinux; GNU C++ version 7.2.0; Boost_106200; UHD_3.11.0.git-191-g1cd96dde] 
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Exact sample rate is: 1000000.026491 Hz
[R82XX] PLL not locked!
*** Error in `/usr/bin/python2': corrupted size vs. prev_size: 0x0000559bb827ad90 ***
======= Backtrace: =========
/lib/x86_64-linux-gnu/libc.so.6(+0x70bfb)[0x7f685c639bfb]
/lib/x86_64-linux-gnu/libc.so.6(+0x76fc6)[0x7f685c63ffc6]
/lib/x86_64-linux-gnu/libc.so.6(+0x780c1)[0x7f685c6410c1]
/usr/local/lib/libgnuradio-runtime-3.7.12git.so.0.0.0(+0x61379)[0x7f685b2a3379]
/lib/x86_64-linux-gnu/libc.so.6(+0x35940)[0x7f685c5fe940]
/lib/x86_64-linux-gnu/libc.so.6(+0x3599a)[0x7f685c5fe99a]
/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf8)[0x7f685c5e92e8]
/usr/bin/python2(_start+0x2a)[0x559bb556d31a]
======= Memory map: ========

Hope somebody can help and many thanks for the great work!

ekagero, 2017-09-28:
Tried a fresh install on Ubuntu but still the same problem :/
Anyone any ideas?

ptrkrysik, 2017-09-28:
I can't reproduce the problem.
gr-gsm/examples/voice_decoding_in_grc/tch_f_reception.grc demo works
without an issue on my Ubuntu 16.04 machine.

ekagero, 2017-09-29:
Thanks a lot for your time! The test file vf_call6_a725_d174_g5_Kc1EF00BAB3BAC7002.cfile is also running here on Kali and Ubuntu, the question is why my live data isn't? Can I dump the my cfile somewhere?

ptrkrysik, 2017-09-29:
Either the capture is incorrect in some way or the codec that is being
used is not TCH/FR.

ekagero, 2017-09-29:
Ok thanks a lot for your help

closed 2017-09-29 by ekagero