ptrkrysik/gr-gsm #341: grgsm_scanner segfaults when run on machine without SDR connected

petterreinholdtsen, 2017-09-30:
Should be easy to reproduce.

velichkov, 2017-10-02:
Hi @petterreinholdtsen,

I'm not able to reproduce this segfault. Here is the my log when there is no SDR device connected and after commenting out the stdout and stderr redirection.

```
./grgsm_scanner 
linux; GNU C++ version 7.1.1 20170503 (Red Hat 7.1.1-1); Boost_106300; UHD_003.010.001.000-0-unknown

Args= 
gr-osmosdr v0.1.x (0.1.5git) gnuradio 3.7.11
built-in source types: file fcd rtl rtl_tcp uhd hackrf rfspace redpitaya 

FATAL: No supported devices found to pick from.

Trying to fill up 1 missing channel(s) with null source(s).
This is being done to prevent the application from crashing
due to gnuradio bug #528.

set_min_output_buffer on block 2 to 4000000
set_min_output_buffer on block 3 to 4000000
```

The `Trying to fill up 1 missing channel(s) with null source(s).` message is coming from `gr-osmosdr` library 

https://github.com/osmocom/gr-osmosdr/blob/c653754dde5e2cf682965e939cc016fbddbd45e4/lib/source_impl.cc#L413-L436

Make sure that `gr-osmosdr` is compiled with `WORKAROUND_GR_HIER_BLOCK2_BUG` flag.

https://github.com/osmocom/gr-osmosdr/blob/c653754dde5e2cf682965e939cc016fbddbd45e4/lib/source_impl.cc#L465

petterreinholdtsen, 2017-10-04:
[Vasil Velichkov]
> Hi @petterreinholdtsen,
>
> I'm not able to reproduce this segfault. Here is the my log when there
> is no SDR device connected and after commenting out the stdout and
> stderr redirection.

Right.  I assumed it would be trivial to reproduce, but apparently
not.  I test on Debian Buster.

> https://github.com/osmocom/gr-osmosdr/blob/c653754dde5e2cf682965e939cc016fbddbd45e4/lib/source_impl.cc#L413-L436
>
> Make sure that `gr-osmosdr` is compiled with
> `WORKAROUND_GR_HIER_BLOCK2_BUG` flag.

I use the official Debian packages, and do not build anything manually
from scratch myself.  Is there a way to see if this is set in the Debian
package?  The build log from Debian is available from
<URL: https://buildd.debian.org/status/fetch.php?pkg=gr-osmosdr&arch=amd64&ver=0.1.4-13&stamp=1505147319&raw=0 >.

Any idea why this workaround isn't the default or what the
GR_HIER_BLOCK2 bug is, and where is it reported to developers that can
fix it?

-- 
Happy hacking
Petter Reinholdtsen

velichkov, 2017-10-04:
> I use the official Debian packages, and do not build anything manually from scratch myself. Is there a way to see if this is set in the Debian package? 

I'm not familiar with Debian packages.

> The build log from Debian is available from <URL: https://buildd.debian.org/status/fetch.php?pkg=gr-osmosdr&arch=amd64&ver=0.1.4-13&stamp=1505147319&raw=0 >. 

Looking at the log I can't say whether it is defined or not as it is not a compile time option but a hardcoded `#define WORKAROUND_GR_HIER_BLOCK2_BUG` in the `source_impl.cc` and `sink_impl.cc`

Try to manually rebuild the debian package and interrupt the build after various patches are applied and after the compilation step has started and then manually check the source_impl.cc 

> Any idea why this workaround isn't the default 

It is default when you manually built it from the git repository. 

> or what the GR_HIER_BLOCK2 bug is,

I have no idea. 

> and where is it reported to developers that can fix it?

I tried to find it but failed. At some point gnuradio have migrated to github and their old bug tracker seems to no longer be available and the bug 528 in github is a different pull request.

Also there is always a chance that you hit a different bug that this workaround does not address at all.

petterreinholdtsen, 2017-10-04:
[Vasil Velichkov]
> Try to manually rebuild the debian package and interrupt the build
> after various patches are applied and after the compilation step has
> started and then manually check the source_impl.cc

I checked, and ./lib/source_impl.cc got '#define
WORKAROUND_GR_HIER_BLOCK2_BUG'.

> I tried to find it but failed. At some point gnuradio have migrated to
> github and their old bug tracker seems to no longer be available and
> the bug 528 in github is a different pull request.

Thank you for trying. :)

> Also there is always a chance that you hit a different bug that this
> workaround does not address at all.

I guess I do. :(

-- 
Happy hacking
Petter Reinholdtsen

velichkov, 2017-10-05:
I was able to reproduce the problem in a Docker container and here is the backtrace
```
(gdb) bt
#0  std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::basic_string (__str=..., this=0x7ffe97b02480)
    at /usr/include/c++/7/bits/basic_string.h:424
#1  soapy_source_c::set_if_gain (this=0x5574378caa40, gain=32, chan=0) at ./lib/soapy/soapy_source_c.cc:257
#2  0x00007f3f68dce2a4 in source_impl::set_if_gain (this=0x5574368de730, gain=32, chan=<optimized out>) at ./lib/source_impl.cc:708
#3  0x00007f3f690be60d in _wrap_source_sptr_set_if_gain (args=<optimized out>, kwargs=<optimized out>)
    at ./obj-x86_64-linux-gnu/swig/osmosdr_swigPYTHON_wrap.cxx:23069
```
https://github.com/osmocom/gr-osmosdr/blob/master/lib/soapy/soapy_source_c.cc#L254-L259
```cpp
double soapy_source_c::set_if_gain( double gain, size_t chan )
{
    //docs specify RF gain is the first element
    const std::string name = this->get_gain_names(chan).front();
    return this->set_gain(gain, name, chan);
}
```
I guess here the problem is that  this->get_gain_names(chan) returns an empty vector.

velichkov, 2017-10-05:
I just sent a patch to the gr-osmosdr mailing list that hopefully fixes this crash

http://lists.osmocom.org/pipermail/osmocom-sdr/2017-October/001606.html

petterreinholdtsen, 2017-10-10:
Thank you very much.  Do you know if the gr-osmosdr community normally react quickly to such patches?

velichkov, 2017-10-10:
> Do you know if the gr-osmosdr community normally react quickly to such patches?

From the mailing list archive it seems they are not very active.

ptrkrysik, 2018-06-02:
I'm closing the issue as it is most probably fault of gr-osmosdr.

closed 2018-06-02 by ptrkrysik