ptrkrysik/gr-gsm #342: some special need

Mitchell1314, 2017-10-01:
Hi sir,
    We have used your "gr-gsm" for a long time and it is wonderful. We have tried every function about this program. 
   And now I have some special need. Is it possible that I just get result of one frequecy not "930-960MHZ". For example, I want the result of frequency "945MHZ" and I just type the command "grgsm 945". Can I do this? Waiting for your reply! Thanks. 
                                                                                                                            Yours sincerely,
                                                                                                                                       Mitchelll

petterreinholdtsen, 2017-10-01:
Not quite sure what you have in mind, and what you mean by the 'grgsm' command.  Could it be that 'grgsm_livemon -f 945M' or 'grgsm_livemon_headless -f 945M' will do what you want?

DRLXX, 2017-10-03:
Hi,you program gave me a lot of help for my study. Is it possible that I change frequency by python code when I use grgsm_livemon, and  the frequency is set by input. For example, The starting frequency is 930MHz, the next time's frequency is by input, Can this be done? Thank you very much!

ptrkrysik, 2018-04-11:
Hi guys, It's hard for me to understand what is the issue here. Generally you can configure initial frequency of grgsm_livemon with --fc parameter. If you don't like the start of frequency range of grgsm_livemon you can easily change it by opening apps/grgsm_livemon.grc file in gnuradio-companion and editing 'start' value for fc_slider block (by default it is 800MHz).

closed 2018-04-11 by ptrkrysik