ptrkrysik/gr-gsm #348: PyBOMBS.Packager.source - WARNING - Configuration failed. Re-trying with higher verbosity.

platinumosetr, 2017-11-14:
Please help me.   sudo pybombs install gr-gsm  
`PyBOMBS.install_manager - INFO - Phase 2: Recursively installing source packages to prefix:
PyBOMBS.install_manager - INFO - Installing package: libosmocore
PyBOMBS.Packager.apt - INFO - Install python-apt to speed up apt processing.
Cloning:     (100%) [=========================================================================================]
PyBOMBS.Packager.apt - INFO - Install python-apt to speed up apt processing.
Configuring: (100%) [=========================================================================================]
PyBOMBS.Packager.source - WARNING - Configuration failed. Re-trying with higher verbosity.
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether make supports nested variables... (cached) yes
checking whether make sets $(MAKE)... (cached) yes
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking for style of include used by make... GNU
checking dependency style of gcc... gcc3
checking build system type... x86_64-pc-linux-gnu
checking host system type... x86_64-pc-linux-gnu
checking how to print strings... printf
checking for a sed that does not truncate output... /bin/sed
checking for grep that handles long lines and -e... /bin/grep
checking for egrep... /bin/grep -E
checking for fgrep... /bin/grep -F
checking for ld used by gcc... /usr/bin/ld
checking if the linker (/usr/bin/ld) is GNU ld... yes
checking for BSD- or MS-compatible name lister (nm)... /usr/bin/nm -B
checking the name lister (/usr/bin/nm -B) interface... BSD nm
checking whether ln -s works... yes
checking the maximum length of command line arguments... 1572864
checking how to convert x86_64-pc-linux-gnu file names to x86_64-pc-linux-gnu format... func_convert_file_noop
checking how to convert x86_64-pc-linux-gnu file names to toolchain format... func_convert_file_noop
checking for /usr/bin/ld option to reload object files... -r
checking for objdump... objdump
checking how to recognize dependent libraries... pass_all
checking for dlltool... no
checking how to associate runtime and link libraries... printf %s\n
checking for ar... ar
checking for archiver @FILE support... @
checking for strip... strip
checking for ranlib... ranlib
checking command to parse /usr/bin/nm -B output from gcc object... ok
checking for sysroot... no
checking for a working dd... /bin/dd
checking how to truncate binary pipes... /bin/dd bs=4096 count=1
checking for mt... mt
checking if mt is a manifest tool... no
checking how to run the C preprocessor... gcc -E
checking for ANSI C header files... yes
checking for sys/types.h... yes
checking for sys/stat.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for memory.h... yes
checking for strings.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for unistd.h... yes
checking for dlfcn.h... yes
checking for objdir... .libs
checking if gcc supports -fno-rtti -fno-exceptions... no
checking for gcc option to produce PIC... -fPIC -DPIC
checking if gcc PIC flag -fPIC -DPIC works... yes
checking if gcc static flag -static works... yes
checking if gcc supports -c -o file.o... yes
checking if gcc supports -c -o file.o... (cached) yes
checking whether the gcc linker (/usr/bin/ld -m elf_x86_64) supports shared libraries... yes
checking whether -lc should be explicitly linked in... no
checking dynamic linker characteristics... GNU/Linux ld.so
checking how to hardcode library paths into programs... immediate
checking whether stripping libraries is possible... yes
checking if libtool supports shared libraries... yes
checking whether to build shared libraries... yes
checking whether to build static libraries... no
checking for pkg-config... /usr/bin/pkg-config
checking for pkg-config... /usr/bin/pkg-config
checking pkg-config is at least version 0.20... yes
checking for ANSI C header files... (cached) yes
checking execinfo.h usability... yes
checking execinfo.h presence... yes
checking for execinfo.h... yes
checking sys/select.h usability... yes
checking sys/select.h presence... yes
checking for sys/select.h... yes
checking sys/socket.h usability... yes
checking sys/socket.h presence... yes
checking for sys/socket.h... yes
checking syslog.h usability... yes
checking syslog.h presence... yes
checking for syslog.h... yes
checking ctype.h usability... yes
checking ctype.h presence... yes
checking for ctype.h... yes
checking netinet/tcp.h usability... yes
checking netinet/tcp.h presence... yes
checking for netinet/tcp.h... yes
checking for size_t... yes
checking for working alloca.h... yes
checking for alloca... yes
checking for library containing dlopen... -ldl
checking for backtrace in -lexecinfo... no
checking for doxygen... /usr/bin/doxygen
checking whether SYS_getrandom is declared... yes
checking if gcc supports -fvisibility=hidden... yes
checking whether struct tm has tm_gmtoff member... yes
checking for TALLOC... yes
checking for PCSC... yes
checking for LIBGNUTLS... no
configure: error: Package requirements (gnutls >= 2.12.0) were not met:

No package 'gnutls' found

Consider adjusting the PKG_CONFIG_PATH environment variable if you
installed software in a non-standard prefix.

Alternatively, you may set the environment variables LIBGNUTLS_CFLAGS
and LIBGNUTLS_LIBS to avoid the need to call pkg-config.
See the pkg-config man page for more details.
PyBOMBS.Packager.source - ERROR - Configuration failed after running at least twice.
PyBOMBS.Packager.source - ERROR - Problem occurred while building package libosmocore:
Configuration failed
PyBOMBS.install_manager - ERROR - Error installing package libosmocore. Aborting.
`

gusgravious, 2017-12-03:
Exact same error under Ubuntu 16.04 LTS and 17.10. Seems that there is no gnutls, I can apt install gnutls-bin though it does not solve the issue.

Manual compilation seems to work though fails on execution with an error not finding osmosdr. On my system I only found osmos_dir though still passing dependency checks.

velichkov, 2017-12-14:
Hi @gusgravious,

>Seems that there is no gnutls, I can apt install gnutls-bin though it does not solve the issue.

Try installing `libgnutls-dev`.

Also take a look how the docker image is built.

https://github.com/ptrkrysik/gr-gsm/blob/4bb3acd1ad05453152dcd70d28d3bc7c5f2ff78f/tests/dockerfiles/Ubuntu_16_04.docker#L4-L17 

This is list of all packages that need to be installed 

https://github.com/ptrkrysik/gr-gsm/blob/4bb3acd1ad05453152dcd70d28d3bc7c5f2ff78f/tests/dockerfiles/Ubuntu_16_04.docker#L24-L27

And these are the steps to manually build gr-gsm (without using pybombs)

You could build the docker image with
```
docker build -f tests/dockerfiles/Ubuntu_16_04.docker  .
```

closed 2018-07-06 by ptrkrysik