ptrkrysik/gr-gsm #354: use libosmocoding AMR decoder

justforfun11, 2017-12-06:
How can libosmocoding AMR decoder from gsm0503 be used for full rate AMR.

I have been trying to use this function:
gsm0503_tch_afs_decode(frameBuffer, bursts_s, 0, codec, 4, &ft, &cmr, &n_errors, &n_bits_total);
		    
But the output file gives a garbled voice.

Please help. How to use this function in tch_f decoder

AlexeyGS, 2017-12-18:
This is because the function returns only payload without a header. You must manually add a header for each codec. For all codecs, header is 0x3c, but for AMR 5.9 header is 0x14.

dabadee, 2017-12-19:
Hi justofrfun11, did you have any luck with using libosmocoding AMR decoder ?

justforfun11, 2017-12-20:
no luck yet buddy

justforfun11, 2017-12-20:
@ AlexeyGS just curious...if this thing works out, will the same thing work for half rate if proper demapping is done

AlexeyGS, 2017-12-20:
@justforfun11 I use the libosmocore functions and it`s work fine for the tch/f channel. But for tch/h channel I can`t get results yet. Maybe I wrote the tch/h demapper incorrectly.

dabadee, 2017-12-20:
@AlexeyGS can you please send your code where you are using libosmocoding for tch/f channel, or may be upload it on your GitHub. I am trying to integrate both of these gsm0503_tch_afs_decode and gsm0503_tch_ahs_decode but not having any success.

justforfun11, 2017-12-24:
@ AlexeyGS 
Thanks, I was able to the integration....header was the issue as u said. 
Demapper for the half rate is ready with me and its working fine. But still the voice is garbled.
Any suggestions.

justforfun11, 2017-12-24:
@dabadee i could finally make that integration work

dabadee, 2017-12-24:
@justforfun11 that's great news. Can you please share your code, may be put it on GitHub ? I can check it on the AMR HR files I have and give you feedback. Also did you check if you are getting 0 in "n_errors" from the gsm0503_tch_ahs_decode function ?

djks74, 2017-12-25:
Hi @justforfun11 ,
can you share demapper/decoder for TCHF/TCHH please?
and all fellows here can decode TCHF voice?

My self never get succes to decode TCHF voice. so please help.....

Thanks,
DUO

ptrkrysik, 2018-06-02:
Hi all, issues with AMR are known #226. In case you want to discuss enhancing AMR support please do it in  that issue or on the mailing-list.

closed 2018-06-02 by ptrkrysik