ptrkrysik/gr-gsm #355: got this error : AttributeError: 'module' object has no attribute 'QWidget'

HaxorGaruda, 2017-12-12:
i got that error after gnuradio upgrade to version 3.7.11 , before is run perfect ...   thanks ...

Os : parrot Os

djks74, 2017-12-26:
try to install gnuradio from master branch.
I get running now!

imranrajjad, 2018-01-03:
I installed from master branch still getting same error on Kali-Linux

djks74, 2018-01-03:
Did you remove all sudo apt remove gnuradio gnuradio-dev already? then build from source agan?

I have just test to install on kali and ubuntu and all is fine mate. make sure you make clean.

jbarea, 2018-01-08:
Hello i recently installed kali linux and tried to run grgsm but encountered the following problem when launching the grgsm_livemon.grc via gnuradio-companion:
<<< Welcome to GNU Radio Companion 3.7.11 >>>

Block paths:
	/usr/share/gnuradio/grc/blocks
	/usr/local/share/gnuradio/grc/blocks

Loading: "/root/Downloads/gr-gsm-master/apps/grgsm_livemon.grc"
>>> Done

Generating: '/root/Downloads/gr-gsm-master/apps/grgsm_livemon.py'

Generating: '/root/Downloads/gr-gsm-master/apps/grgsm_livemon.py'

Executing: /usr/bin/python -u /root/Downloads/gr-gsm-master/apps/grgsm_livemon.py

Warning: failed to XInitThreads()
linux; GNU C++ version 7.2.0; Boost_106200; UHD_003.010.002.000-0-unknown

Traceback (most recent call last):
  File "/root/Downloads/gr-gsm-master/apps/grgsm_livemon.py", line 354, in <module>
    main()
  File "/root/Downloads/gr-gsm-master/apps/grgsm_livemon.py", line 342, in main
    tb = top_block_cls(args=options.args, collector=options.collector, collectorport=options.collectorport, fc=options.fc, gain=options.gain, osr=options.osr, ppm=options.ppm, samp_rate=options.samp_rate, serverport=options.serverport, shiftoff=options.shiftoff)
  File "/root/Downloads/gr-gsm-master/apps/grgsm_livemon.py", line 64, in __init__
    self.restoreGeometry(self.settings.value("geometry").toByteArray())
AttributeError: 'NoneType' object has no attribute 'toByteArray'

>>> Done

On the other hand, if i execute directly the executable grgsm_livemon i got the following:
grgsm_livemon
Warning: failed to XInitThreads()
Traceback (most recent call last):
  File "/usr/local/bin/grgsm_livemon", line 45, in <module>
    from gnuradio import qtgui
  File "/usr/lib/python2.7/dist-packages/gnuradio/qtgui/__init__.py", line 37, in <module>
    from range import Range, RangeWidget
  File "/usr/lib/python2.7/dist-packages/gnuradio/qtgui/range.py", line 67, in <module>
    class RangeWidget(QtWidgets.QWidget):
AttributeError: 'module' object has no attribute 'QWidget'

I have installed pyqt4 library and tried to install pyqt5 but the same error appeared. I followed all the steps correctly so i don't know where is the problem.
Thanks in advance!

velichkov, 2018-01-08:
Hi @jbarea,

> i recently installed kali linux 

Which version?

> and tried to run grgsm but encountered the following problem 

Which version of gnuradio did you install and how did you install it - from the apt repository, using pybombs or manually?

Have you tried to install `gr-gsm` from the apt repository (It is available only in Kali Rolling - `apt-get install gr-gsm`) and if yes is it working?

> I followed all the steps correctly so i don't know where is the problem.

Which instructions you have followed, could you post a link?

Could you try the headless version `grgsm_livemon_headless`?

jbarea, 2018-01-09:
Thanks for your reply. I tried with kali linux light 2017.2 version. Gnuradio is the one from the repository, but grgsm  i got it from git, as i didn't know there was available via repo. The installation of gnuradio is from apt-get, not using pybombs. The grgsm_livemon_headless version gets the hackrf recognized, but no shows data from terminal, so no luck with that one.

velichkov, 2018-01-09:
> I tried with kali linux light 2017.2 version.

I recommend you to update to the latest Kali Rolling. It should be something like 
```
apt update
apt dist-upgrade
apt-get install gr-gsm
reboot
```
See https://www.kali.org/releases/kali-linux-2017-3-release/ and https://www.kali.org/news/kali-linux-2017-2-release/

> but grgsm i got it from git,

If you decide to install gr-gsm from the apt repository make sure you first uninstall the one you compiled using `make uninstall`

> The grgsm_livemon_headless version gets the hackrf recognized, but no shows data from terminal, so no luck with that one

Most probably you need to specify the C0 carrier frequency. You could find it using the `grgsm_scanner` tool and then before starting the ` grgsm_livemon_headless` in another terminal you need to start `wireshark` on the loopback (lo) interface

```
$grgsm_scanner 
linux; GNU C++ version 7.2.0; Boost_106200; UHD_003.010.002.000-0-unknown

ARFCN:  975, Freq:  925.2M, CID:  2852, LAC:  2200, MCC: 284, MNC:   1, Pwr: -70
ARFCN:  978, Freq:  925.8M, CID:  2804, LAC:  2200, MCC: 284, MNC:   1, Pwr: -64
ARFCN:  981, Freq:  926.4M, CID:  2884, LAC:  2200, MCC: 284, MNC:   1, Pwr: -69
ARFCN:  982, Freq:  926.6M, CID:  1174, LAC:  2400, MCC: 284, MNC:   1, Pwr: -80

$grgsm_livemon_headless -f 925.2M
```
and in another terminal 
```
$sudo wireshark -k -f udp -Y gsmtap -i lo
```

See also
https://github.com/ptrkrysik/gr-gsm/wiki/Usage#live-monitoring-of-c0-gsm-channels
https://github.com/ptrkrysik/gr-gsm/wiki/Usage#analyzing-gsm-messages-in-the-wireshark

jbarea, 2018-01-09:
I began a new installation of kali and when installing grgsm, i installed it via repo, but i can't locate the grc files or the executables of grgsm_livemon so i got it from the zip file of the git repo of grgsm. I loaded the grgsm_livemon.grc with gnuradio companion, but same error. Where are located the files of grgsm when using the repo?.

velichkov, 2018-01-09:
Hi @petterreinholdtsen,

It seems that the `gr-gsm-0.41.2-1` package is incompatible with `gnuradio-3.7.11-6` and there is no easy way to fix it. The gnuradio package in Debian Buster has been heavily modified and all distributions that are directly or indirectly based on Debian Buster are affected - Ubuntu 18.04,  Kali Rolling, Parrot Os, ....

On Kali Rolling the only way I was able to run `grgsm_livemon` was the following
```
sudo apt-get remove gr-gsm
git clone https://github.com/ptrkrysik/gr-gsm.git
mkdir gr-gsm/build/
cd gr-gsm/build
cmake ../
make
sudo make install
cd ../apps
grcc -d . grgsm_livemon.grc
```
The last commands generate `grgsm_livemon.py` in the current directory, when executed it fails with 
```
$ ./grgsm_livemon.py 
Warning: failed to XInitThreads()
linux; GNU C++ version 7.2.0; Boost_106200; UHD_003.010.002.000-0-unknown

QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-grgsm'
Traceback (most recent call last):
  File "./grgsm_livemon.py", line 354, in <module>
    main()
  File "./grgsm_livemon.py", line 342, in main
    tb = top_block_cls(args=options.args, collector=options.collector, collectorport=options.collectorport, fc=options.fc, gain=options.gain, osr=options.osr, ppm=options.ppm, samp_rate=options.samp_rate, serverport=options.serverport, shiftoff=options.shiftoff)
  File "./grgsm_livemon.py", line 64, in __init__
    self.restoreGeometry(self.settings.value("geometry").toByteArray())
AttributeError: 'NoneType' object has no attribute 'toByteArray'
```
The workaround here is to comment out or remove line 64 in `./grgsm_livemon.py`
```
64	        self.restoreGeometry(self.settings.value("geometry").toByteArray())
```

jbarea, 2018-01-10:
Commenting the line and trying to execute the file gets me the following error:
Warning: failed to XInitThreads()
linux; GNU C++ version 7.2.0; Boost_106200; UHD_003.010.002.000-0-unknown

gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.11
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for 4294967295, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for 4294967295, skipping unlock
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for 4294967295, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for 4294967295, skipping unlock
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for 4294967295, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for 4294967295, skipping unlock
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for 4294967295, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for 4294967295, skipping unlock
[INFO] Using format CF32.
Segmentation fault

velichkov, 2018-01-10:
Hi @jbarea,

> Commenting the line and trying to execute the file gets me the following error:
> ...
> Segmentation fault

Try starting it with `--args=hackrf` or `--args=hackrf=0` parameter
```
./grgsm_livemon.py --args=hackrf
```
See also https://github.com/ptrkrysik/gr-gsm/issues/336#issuecomment-332957714

> The grgsm_livemon_headless version gets the hackrf recognized,

Does the grgsm_livemon_headless version works on your new installation?

HaxorGaruda, 2018-01-30:
thanks.. Fix with @velichkov tuts, by : 
grcc -d . grgsm_livemon.grc
and 
editing ./grgsm_livemon.py         (remove line 64)
self.restoreGeometry(self.settings.value("geometry").toByteArray())

apptestmail, 2018-02-16:
can you help me with that issue .

**Traceback (most recent call last):
  File "/usr/local/bin/grgsm_livemon", line 45, in <module>
    from gnuradio import qtgui
  File "/usr/lib/python2.7/dist-packages/gnuradio/qtgui/__init__.py", line 37, in <module>
    from range import Range, RangeWidget
  File "/usr/lib/python2.7/dist-packages/gnuradio/qtgui/range.py", line 67, in <module>
    class RangeWidget(QtWidgets.QWidget):
AttributeError: 'module' object has no attribute 'QWidget'**

djks74, 2018-02-16:
> line 45, in from gnuradio import qtgui File
> "/usr/lib/python2.7/dist-packages/gnuradio/qtgui/init.py", line 37, in from

You need to install qt library.

apptestmail, 2018-02-16:
same issue

djks74, 2018-02-16:
after install qt library, do you make uninstall then make sure when
recompile the cmake has no error?

DUO

velichkov, 2018-02-16:
Hi @apptestmail, @djks74,

This error is not because of missing qt library but because of incompatibility between grgsm_livemon and the latest gnuradio versions from Debian/Ubuntu/Kali repositories. The gnuradio in debian has been migrated from qt4 to qt5 (https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=874893) so all QT flowgraphs have to be regenerated.

Open `grgsm_livemon.grc` with gnuradio-companion and go to Run -> Generate (F5), a file with name `grgsm_livemon.py` will be generated next to the `grgsm_livemon.grc`. Then you need to open it and remove or comment out line 64.  You could rename it to `grgsm_livemon` and replace the one in `/usr/local/bin/`.  

See also https://github.com/ptrkrysik/gr-gsm/issues/355#issuecomment-356401976

vequeemnhe, 2018-02-22:
hello  everyone 
i have upgrade kali rolling "Linux  4.14.0-kali3-amd64 #1 SMP Debian 4.14.17-1kali1 (2018-02-16) x86_64 GNU/Linux "

and use command 
i get this error : 

>  grcc -d . grgsm_livemon.grc
Warning: Block with key "gsm_bcch_ccch_demapper" already exists.
	Ignoring: /usr/local/share/gnuradio/grc/blocks/gsm_bcch_ccch_demapper.xml
Warning: Block with key "gsm_bcch_ccch_sdcch4_demapper" already exists.
	Ignoring: /usr/local/share/gnuradio/grc/blocks/gsm_bcch_ccch_sdcch4_demapper.xml
Warning: Block with key "gsm_burst_file_sink" already exists.
	Ignoring: /usr/local/share/gnuradio/grc/blocks/gsm_burst_file_sink.xml
Warning: Block with key "gsm_burst_file_source" already exists.
	Ignoring: /usr/local/share/gnuradio/grc/blocks/gsm_burst_file_source.xml
Warning: Block with key "gsm_burst_fnr_filter" already exists.
	Ignoring: /usr/local/share/gnuradio/grc/blocks/gsm_burst_fnr_filter.xml
Warning: Block with key "gsm_burst_sdcch_subslot_filter" already exists.
	Ignoring: /usr/local/share/gnuradio/grc/blocks/gsm_burst_sdcch_subslot_filter.xml
Warning: Block with key "gsm_burst_sdcch_subslot_splitter" already exists.
	Ignoring: /usr/local/share/gnuradio/grc/blocks/gsm_burst_sdcch_subslot_splitter.xml
Warning: Block with key "gsm_burst_timeslot_filter" already exists.
	Ignoring: /usr/local/share/gnuradio/grc/blocks/gsm_burst_timeslot_filter.xml
Warning: Block with key "gsm_burst_timeslot_splitter" already exists.
	Ignoring: /usr/local/share/gnuradio/grc/blocks/gsm_burst_timeslot_splitter.xml
Warning: Block with key "gsm_bursts_printer" already exists.
	Ignoring: /usr/local/share/gnuradio/grc/blocks/gsm_bursts_printer.xml
Warning: Block with key "gsm_clock_offset_control" already exists.
	Ignoring: /usr/local/share/gnuradio/grc/blocks/gsm_clock_offset_control.xml
Warning: Block with key "gsm_clock_offset_corrector_tagged" already exists.
	Ignoring: /usr/local/share/gnuradio/grc/blocks/gsm_clock_offset_corrector_tagged.xml
Warning: Block with key "gsm_collect_system_info" already exists.
	Ignoring: /usr/local/share/gnuradio/grc/blocks/gsm_collect_system_info.xml
Warning: Block with key "gsm_control_channels_decoder" already exists.
	Ignoring: /usr/local/share/gnuradio/grc/blocks/gsm_control_channels_decoder.xml
Warning: Block with key "gsm_controlled_fractional_resampler_cc" already exists.
	Ignoring: /usr/local/share/gnuradio/grc/blocks/gsm_controlled_fractional_resampler_cc.xml
Warning: Block with key "gsm_controlled_rotator_cc" already exists.
	Ignoring: /usr/local/share/gnuradio/grc/blocks/gsm_controlled_rotator_cc.xml
Warning: Block with key "gsm_cx_channel_hopper" already exists.
	Ignoring: /usr/local/share/gnuradio/grc/blocks/gsm_cx_channel_hopper.xml
Warning: Block with key "gsm_decryption" already exists.
	Ignoring: /usr/local/share/gnuradio/grc/blocks/gsm_decryption.xml
Warning: Block with key "gsm_dummy_burst_filter" already exists.
	Ignoring: /usr/local/share/gnuradio/grc/blocks/gsm_dummy_burst_filter.xml
Warning: Block with key "gsm_extract_cmc" already exists.
	Ignoring: /usr/local/share/gnuradio/grc/blocks/gsm_extract_cmc.xml
Warning: Block with key "gsm_extract_immediate_assignment" already exists.
	Ignoring: /usr/local/share/gnuradio/grc/blocks/gsm_extract_immediate_assignment.xml
Warning: Block with key "gsm_extract_system_info" already exists.
	Ignoring: /usr/local/share/gnuradio/grc/blocks/gsm_extract_system_info.xml
Warning: Block with key "gsm_fcch_burst_tagger" already exists.
	Ignoring: /usr/local/share/gnuradio/grc/blocks/gsm_fcch_burst_tagger.xml
Warning: Block with key "gsm_fcch_detector" already exists.
	Ignoring: /usr/local/share/gnuradio/grc/blocks/gsm_fcch_detector.xml
Warning: Block with key "gsm_input" already exists.
	Ignoring: /usr/local/share/gnuradio/grc/blocks/gsm_input.xml
Warning: Block with key "gsm_message_file_sink" already exists.
	Ignoring: /usr/local/share/gnuradio/grc/blocks/gsm_message_file_sink.xml
Warning: Block with key "gsm_message_file_source" already exists.
	Ignoring: /usr/local/share/gnuradio/grc/blocks/gsm_message_file_source.xml
Warning: Block with key "gsm_message_printer" already exists.
	Ignoring: /usr/local/share/gnuradio/grc/blocks/gsm_message_printer.xml
Warning: Block with key "gsm_msg_to_tag" already exists.
	Ignoring: /usr/local/share/gnuradio/grc/blocks/gsm_msg_to_tag.xml
Warning: Block with key "gsm_receiver" already exists.
	Ignoring: /usr/local/share/gnuradio/grc/blocks/gsm_receiver.xml
Warning: Block with key "gsm_receiver_with_uplink" already exists.
	Ignoring: /usr/local/share/gnuradio/grc/blocks/gsm_receiver_with_uplink.xml
Warning: Block with key "gsm_sch_detector" already exists.
[... 2 lines cut ...]
	Ignoring: /usr/local/share/gnuradio/grc/blocks/gsm_sdcch8_demapper.xml
Warning: Block with key "gsm_tch_f_chans_demapper" already exists.
	Ignoring: /usr/local/share/gnuradio/grc/blocks/gsm_tch_f_chans_demapper.xml
Warning: Block with key "gsm_tch_f_decoder" already exists.
	Ignoring: /usr/local/share/gnuradio/grc/blocks/gsm_tch_f_decoder.xml
Warning: Block with key "gsm_tmsi_dumper" already exists.
	Ignoring: /usr/local/share/gnuradio/grc/blocks/gsm_tmsi_dumper.xml
Warning: Block with key "gsm_universal_ctrl_chans_demapper" already exists.
	Ignoring: /usr/local/share/gnuradio/grc/blocks/gsm_universal_ctrl_chans_demapper.xml
Warning: Block with key "gsm_uplink_downlink_splitter" already exists.
	Ignoring: /usr/local/share/gnuradio/grc/blocks/gsm_uplink_downlink_splitter.xml
[... cut ...]
but when i use command grgsm_livemon 
i get this error : 

> python grgsm_livemon
Traceback (most recent call last):
  File "grgsm_livemon", line 45, in <module>
    from gnuradio import qtgui
  File "/usr/lib/python2.7/dist-packages/gnuradio/qtgui/__init__.py", line 37, in <module>
    from range import Range, RangeWidget
  File "/usr/lib/python2.7/dist-packages/gnuradio/qtgui/range.py", line 68, in <module>
    class RangeWidget(QtWidgets.QWidget):
AttributeError: 'module' object has no attribute 'QWidget'

vequeemnhe, 2018-02-23:
i have code 1 script python 2.7 on Qt5 
> #!/usr/bin/python2
import PyQt5
from PyQt5 import QtWidgets
class modern(QtWidgets.Widget):
    pass
    print "QVN"
if __name__ == '__main__':
    t = modern
    print (t)
and after i get message error : 
> Traceback (most recent call last):
  File "test.py", line 5, in <module>
    class modern(QtWidgets.Widget):
AttributeError: 'module' object has no attribute 'QWidget'
here problem .... PyQt5 don't support QWidget 
or change function replace QtWidget.
hope can fix this error, but i going find orther.... can fix....
hope can get helpfull to everyone....
thanks everyone

velichkov, 2018-02-23:
Hi @vequeemnhe,

> and use command
> i get this error :
>> grcc -d . grgsm_livemon.grc
>> Warning: Block with key "gsm_bcch_ccch_demapper" already exists.
    Ignoring: /usr/local/share/gnuradio/grc/blocks/gsm_bcch_ccch_demapper.xml

These are all warnings, ignore thme for now

>  but when i use command grgsm_livemon
>  i have rename it to grgsm_livemon and replace the one in /usr/local/bin/.
and comment line 64

The file that grcc generates is called `grgsm_livemon.py` and not `grgsm_livemon` so after running the grcc you will have two files in your current folders `grgsm_livemon` and `grgsm_livemon.py`. Use the one with the `.py` extension. You could also remove the `grgsm_livemon` before running the `grcc` command and after it finishes rename and copy to `/usr/local/bin/`

Radio-Lab, 2018-03-10:
Hi, I have tried to install grgsm on **Linux kali 4.14.0-kali3-amd64 #1 SMP Debian 4.14.17-1kali1 (2018-02-16) x86_64 GNU/Linux**   when I try to launch grgsm_livemon I am getting the error : 
root@kali:~# grgsm_livemon
Traceback (most recent call last):
  File "/usr/local/bin/grgsm_livemon", line 22, in <module>
    from gnuradio import qtgui
  File "/usr/lib/python2.7/dist-packages/gnuradio/qtgui/__init__.py", line 37, in <module>
    from range import Range, RangeWidget
  File "/usr/lib/python2.7/dist-packages/gnuradio/qtgui/range.py", line 67, in <module>
    class RangeWidget(QtWidgets.QWidget):
AttributeError: 'module' object has no attribute 'QWidget'

How to troubleshoot it ?

velichkov, 2018-03-10:
Hi @Radio-Lab,

>AttributeError: 'module' object has no attribute 'QWidget'
>
>How to troubleshoot it ?

There is no need to troubleshoot anything.
The reasons for this error and how to fix it is described in this issues in https://github.com/ptrkrysik/gr-gsm/issues/355#issuecomment-356401976, https://github.com/ptrkrysik/gr-gsm/issues/355#issuecomment-366277650 and https://github.com/ptrkrysik/gr-gsm/issues/355#issuecomment-368032541

ahmedaze, 2018-03-24:
Help me please.

Traceback (most recent call last):
File "/usr/local/bin/grgsm_livemon", line 22, in 
from gnuradio import qtgui
File "/usr/lib/python2.7/dist-packages/gnuradio/qtgui/init.py", line 37, in 
from range import Range, RangeWidget
File "/usr/lib/python2.7/dist-packages/gnuradio/qtgui/range.py", line 67, in 
class RangeWidget(QtWidgets.QWidget):
AttributeError: 'module' object has no attribute 'QWidget'

velichkov, 2018-03-24:
Hi @ahmedaze,

It seems you are not using the latest `master` 7476868, could you try with it?

Have you tried one of the solutions described in https://github.com/ptrkrysik/gr-gsm/issues/355#issuecomment-356401976, https://github.com/ptrkrysik/gr-gsm/issues/355#issuecomment-366277650 and https://github.com/ptrkrysik/gr-gsm/issues/355#issuecomment-368032541 and what was the result?

danyhm, 2018-04-12:
@velichkov 
My grgsm was running fine but after updating gnuradio it doesn't run anymore with the QWidget error
I uninstalled it completly and tried to build it again with the latest master however i get a build error

make -j4
[  0%] Built target grgsm_swig_swig_doc
[  2%] Built target pygen_python_b2431
[  5%] Built target pygen_python_misc_utils_ab994
[ 68%] Built target grgsm
[ 71%] Built target pygen_python_receiver_97029
[ 73%] Built target pygen_python_demapping_0f47b
[ 76%] Built target pygen_apps_helpers_75853
[ 77%] Built target man_grgsm_livemon.rst
[ 80%] Built target _grgsm_swig_swig_tag
[ 83%] Built target grgsm_swig_swig_2d0df
[ 87%] Built target pygen_swig_9d1b8
[ 92%] Built target _grgsm_swig
[ 95%] Generating grgsm_livemon
[ 95%] Generating grgsm_livemon_headless
Warning: Block with key "gsm_burst_file_sink" already exists.
	Ignoring: /home/daniel/gr-gsm/grc/misc_utils/gsm_burst_file_sink.xml
Warning: Block with key "gsm_burst_file_source" already exists.
	Ignoring: /home/daniel/gr-gsm/grc/misc_utils/gsm_burst_file_source.xml
Warning: Block with key "gsm_bursts_printer" already exists.
	Ignoring: /home/daniel/gr-gsm/grc/misc_utils/gsm_bursts_printer.xml
Warning: Block with key "gsm_burst_file_sink" already exists.
	Ignoring: /home/daniel/gr-gsm/grc/misc_utils/gsm_burst_file_sink.xml
Warning: Block with key "gsm_burst_file_source" already exists.
	Ignoring: /home/daniel/gr-gsm/grc/misc_utils/gsm_burst_file_source.xml
Warning: Block with key "gsm_bursts_printer" already exists.
	Ignoring: /home/daniel/gr-gsm/grc/misc_utils/gsm_bursts_printer.xml
Warning: Block with key "gsm_burst_timeslot_filter" already exists.
	Ignoring: /home/daniel/gr-gsm/grc/flow_control/gsm_burst_timeslot_filter.xml
Warning: Block with key "gsm_burst_timeslot_filter" already exists.
	Ignoring: /home/daniel/gr-gsm/grc/flow_control/gsm_burst_timeslot_filter.xml
Segmentation fault
apps/CMakeFiles/pygen_apps.dir/build.make:84: recipe for target 'apps/grgsm_livemon_headless' failed
make[2]: *** [apps/grgsm_livemon_headless] Error 1
make[2]: *** Waiting for unfinished jobs....
Segmentation fault
apps/CMakeFiles/pygen_apps.dir/build.make:79: recipe for target 'apps/grgsm_livemon' failed
make[2]: *** [apps/grgsm_livemon] Error 1
CMakeFiles/Makefile2:1099: recipe for target 'apps/CMakeFiles/pygen_apps.dir/all' failed
make[1]: *** [apps/CMakeFiles/pygen_apps.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2

danyhm, 2018-04-12:
@velichkov 
I managed to build it with no errors by cloning into the get_kali_version branch.

but the same Qwidget error still exists and the "grcc -d . grgsm_livemon.grc" command gives this error:

Warning: Block with key "gsm_burst_file_sink" already exists.
	Ignoring: /usr/local/share/gnuradio/grc/blocks/gsm_burst_file_sink.xml
Warning: Block with key "gsm_burst_file_source" already exists.
	Ignoring: /usr/local/share/gnuradio/grc/blocks/gsm_burst_file_source.xml
Warning: Block with key "gsm_burst_timeslot_filter" already exists.
	Ignoring: /usr/local/share/gnuradio/grc/blocks/gsm_burst_timeslot_filter.xml
Warning: Block with key "gsm_bursts_printer" already exists.
	Ignoring: /usr/local/share/gnuradio/grc/blocks/gsm_bursts_printer.xml
Segmentation fault

danyhm, 2018-04-13:
@velichkov 
I tried with a fresh install of kali.
first of all after installing gr-gsm from kali repository the swig error was being displayed and I couldn't even launch gnuradio. after fiddling a lot i found out that the libvolk.so.1.3 was not being found by gnuradio and gqrx while in fact i had it installed. so i tried installing from volk source and that fixed the problem and i got gnuradio and gqrx working.
then I tried your solution again but this time it gave me a different error while running "grcc -d . grgsm_livemon.grc"

Validation failed:

Block - gsm_receiver_0 - GSM Receiver(gsm_receiver):
	Param - Cell allocation(cell_allocation):
		Value "[arfcn.downlink2arfcn(fc)]" cannot be evaluated:
		name 'arfcn' is not defined

Block - import_1 - Import(import):
	Param - Import(import):
		Import "from grgsm import arfcn" failed.
Error during file compilation.

this time I opened the grc with gnu companion and the import grgsm block was red.

any ideas ?

velichkov, 2018-04-13:
Hi @danyhm,

Sorry for the late response, I was on vacation with very limited Internet access. It's great that you have been able to get it working.

> Segmentation fault
apps/CMakeFiles/pygen_apps.dir/build.make:84: recipe for target 'apps/grgsm_livemon_headless' failed

Unfortunately I'm not able to reproduce this segmentation fault.

> I managed to build it with no errors by cloning into the get_kali_version branch.

This branch does not contains the changes that compile `grgsm_livemon.grc` and that's why you were able to compile the branch but you had the QWidget error. I do not recommend using this branch  

> after fiddling a lot i found out that the libvolk.so.1.3 was not being found by gnuradio and gqrx while in fact i had it installed. so i tried installing from volk source and that fixed the problem and i got gnuradio and gqrx working.

I've already reported the volk problem to Debian https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=895255
For the record an alternative solution is to create one symlink  (#391 and https://github.com/ptrkrysik/gr-gsm/pull/378#issuecomment-379583050)
```
ln -sf /usr/lib/x86_64-linux-gnu/libvolk.so.1.3.1 /usr/lib/x86_64-linux-gnu/libvolk.so.1.3
```

ptrkrysik, 2018-06-02:
I'm closing the issue as it should be resolved currently by grcc compilation. In case it is not I'll reopen it.

closed 2018-06-02 by ptrkrysik

Tchakatak, 2019-02-22:
I'm experiencing the same problems with the latest version from source / gnuradio 3.7.13.4.

velichkov, 2019-02-25:
Hi @Tchakatak,

> I'm experiencing the same problems with the latest version from source / gnuradio 3.7.13.4.

Which OS are you using and which version?
How did you install gnuradio - from the repository, manually or using pybombs?
How did you get the gr-gsm sources - git or tar.gz archive?
Could you tell me the exact gr-gsm version - git commit id or URL?

Tchakatak, 2019-02-27:
I'm running archlinux latest. 
Gnuradio from repo / source / pybombs tested
gr-gsm from Aur / Git tested 

grgsm latest 
gr-osmosdr v0.1.4-127-g4d83c606 (0.1.5git) gnuradio 3.7.13.4

velichkov, 2019-02-28:
Hi @Tchakatak,

> I'm experiencing the same problems with the latest version from source / gnuradio 3.7.13.4.

In this issue at least three different problems were discussed.
1. AttributeError: 'module' object has no attribute 'QWidget'
1. Segmentation fault in grcc
1. Value "[arfcn.downlink2arfcn(fc)]" cannot be evaluated:
name 'arfcn' is not defined

Which of these problems you are experiencing? Provide some more information, logs, messages, etc..

> I'm running archlinux latest.

I'm not very familiar with Arch Linux.

> Gnuradio from repo / source / pybombs tested

Do you have them installed all at the same time - one in `/usr/lib/` and one in `/usr/local/`? Having multiple `gnuradio` versions could be a source of many errors. Try to remove the versions installed by pybombs and from source and rebuild gr-gsm - `rm -rvf build && mkdir build && cd build && cmake .. && make`

> grgsm latest

Give me the exact version - branch, commit id, ...