ptrkrysik/gr-gsm #356: problem with 'libosmocore' on macOS while compiling. 'make' throws errors

KirilShishkoski, 2017-12-15:
I'm on a macOS machine here. Only this dependency left to do it but I'm having troubles. After autoreconf -i
./configure

(up to here works ok)

make 

(after 'make') it throws me these errors:

[image]
[image]

velichkov, 2017-12-15:
Hi @KirilShishkoski,

I don't have access to macOS system right now so I can't really help you but have you tried to report the problem to the osmocom project first? 

http://git.osmocom.org/libosmocore/about/
> Discussions related to libosmocore are happening on the openbsc@lists.osmocom.org mailing list, please see https://lists.osmocom.org/mailman/listinfo/openbsc for subscription options and the list archive.

P.S.
And please when reporting issues upload the full build log together with all install/build steps you have executed and not just few screen shots.

BR

KirilShishkoski, 2017-12-17:
Fixed it with brew tap from one of these dirs (don't remember which was exactly try both):

nejohnson2/homebrew-sdr
or
dholm/homebrew-sdr

closed 2017-12-17 by KirilShishkoski

tutur0001, 2021-09-02:
hi men i have similar error can you help me please