ptrkrysik/gr-gsm #358: grgsm-livemon → ImportError: cannot import name qtgui

KirilShishkoski, 2017-12-17:
I finally managed to compile gr-gsm on macOS. All tools work except for the grgsm-livemon which after I run it pops up ImportError: cannot import name qtgui. Here's the full log:

bash4.4$ grgsm_livemon
Traceback (most recent call last):
  File "/usr/local/bin/grgsm_livemon", line 45, in <module>
    from gnuradio import qtgui
ImportError: cannot import name qtgui

Any idea what might be the problem?

gaafouri, 2017-12-18:
hi
I did all the steps when I execute it displays the following errors 

ahmed@ubuntu:~$ grgsm_scanner 
Traceback (most recent call last):
  File "/usr/local/bin/grgsm_scanner", line 33, in <module>
    import grgsm
  File "/usr/local/lib/python2.7/dist-packages/grgsm/__init__.py", line 45, in <module>
    from grgsm_swig import *
  File "/usr/local/lib/python2.7/dist-packages/grgsm/grgsm_swig.py", line 28, in <module>
    _grgsm_swig = swig_import_helper()
  File "/usr/local/lib/python2.7/dist-packages/grgsm/grgsm_swig.py", line 24, in swig_import_helper
    _mod = imp.load_module('_grgsm_swig', fp, pathname, description)
ImportError: libgrgsm.so.0.41.2: cannot open shared object file: No such file or directory
ahmed@ubuntu:~$ 

thank you

KirilShishkoski, 2017-12-18:
I solved it. I did it everything from scratch. But now I'm facing other problem.

KirilShishkoski, 2017-12-18:
I reinstalled gnuradio. Solved it but now I have a problem with multiple versions of Python.

closed 2017-12-18 by KirilShishkoski

gaafouri, 2017-12-20:
I reinstalled gnuradio but it not resolved