ptrkrysik/gr-gsm #363: No install method:autotools-bootstrap while installing gr-gsm on Kali

wzjin2017, 2017-12-26:
Hi! I am new to this field and recently I've been installing gr-gsm on Kali. 

Here is the issue, while I am using the command 
`sudo  pybombs install gr-gsm`

I encountered this error

> pybombs install gr-gsm
PyBOMBS - INFO - PyBOMBS Version 2.3.2
PyBOMBS.install_manager - INFO - Phase 1: Creating install tree and installing binary packages:
PyBOMBS.install_manager - ERROR - Package has no install method: autotools-bootstrap

Thank you guys so much!

velichkov, 2017-12-26:
Hi @wzjin2017,

You need to install first the `automake` and `autoconf`. 

See also the [autotools-bootstrap recipe file](https://github.com/gnuradio/pybombs/blob/master/pybombs/recipes/autotools-bootstrap.lwr)

wzjin2017, 2017-12-28:
@velichkov Thank you so much!  Actually I solved the automake and autoconf part and later on when I reuse the command 
`sudo pybombs install gr-gsm` 
and similar error happens again, it goes like this

> PyBOMBS.install_manager - ERROR - Package has no install method: libevent

And I tried to use the quite similar solution  to solve this issue as I did to the "automake & autoconf"  one and reintall gr-gsm, and the error keeps popping up.
The solution I used to solve "libevent" problem is as follows:
`wget https://github.com/downloads/libevent/libevent/libevent-1.4.11-stable.tar.gz`
`tar xf libevent*`
`cd libevent-1.4.11-stable`
`sh configure --prefix /usr/local`
`make install`
I think that libevent is successfully installed and have no idea what goes wrong here.
Excuse me for asking some newbie questions and being annoying :)

velichkov, 2017-12-28:
Hi @wzjin2017,

> The solution I used to solve "libevent" problem is as follows:
> wget https://github.com/downloads/libevent/libevent/libevent-1.4.11-stable.tar.gz

Try to install the `libevent-dev` with the package manager, e.g `apt-get install libevent-dev`

Which version of Kali you are using?

In the latest Kali Rolling release there is a `gr-gsm` package which you could install through the package manager `apt-get install gr-gsm`. If you are using Kalin 2.0 or older I highly recommend you to upgrade to the latest Rolling release.

See also:
* https://www.kali.org/kali-linux-releases/
* https://www.kali.org/releases/kali-linux-2017-3-release/ 
* https://www.kali.org/news/kali-linux-rolling-edition-2016-1/
* https://www.kali.org/releases/kali-linux-rolling-edition-2016-1/

closed 2018-07-06 by ptrkrysik

chiraglalkiya, 2019-11-23:
root@kali:~# sudo pybombs install gr-gsm
[INFO] Prefix Python version is: 2.7.17
[INFO] PyBOMBS Version 2.3.4a0
[INFO] Install python-apt to speed up apt processing.
[INFO] Phase 1: Creating install tree and installing binary packages:
[INFO] Executing command with elevated privileges: `sudo -H apt-get -y install libqwt-qt5-dev'
Reading package lists... Done
Building dependency tree       
Reading state information... Done
libqwt-qt5-dev is already the newest version (6.1.4-1.1).
The following package was automatically installed and is no longer required:
  libisl19
Use 'sudo apt autoremove' to remove it.
0 upgraded, 0 newly installed, 0 to remove and 1131 not upgraded.
Install tree:
|
\- gr-gsm
   |
   +- gr-osmosdr
   |  |
   |  +- airspy
   |  |
   |  +- hackrf
   |  |
   |  +- uhd
   |  |
   |  +- soapysdr
   |  |
   |  +- gnuradio
   |  |  |
   |  |  +- uhd
   |  |  |
   |  |  \- qwt6
   |  |
   |  +- bladeRF
   |  |
   |  +- gr-iqbal
   |  |  |
   |  |  +- libosmo-dsp
   |  |  |
   |  |  \- gnuradio
   |  |     |
   |  |     +- uhd
   |  |     |
   |  |     \- qwt6
   |  |
   |  +- osmo-sdr
   |  |
   |  \- rtl-sdr
   |
   \- gnuradio
      |
      +- uhd
      |
      \- qwt6
[INFO] Phase 1 complete: All binary dependencies installed.
[INFO] Phase 2: Recursively installing source packages to prefix:
[INFO] Installing package: qwt6
[INFO] Install python-apt to speed up apt processing.
Building:    (100%) [=========================================================]
cd src/ && ( test -e Makefile || /usr/lib/qt5/bin/qmake -o Makefile /usr/local/src/qwt6/src/src.pro ) && make -f Makefile 
make[1]: Entering directory '/usr/local/src/qwt6/src'
compiling qwt_abstract_scale_draw.cpp
In file included from ../../../../include/x86_64-linux-gnu/qt5/QtCore/qlocale.h:43,
                 from ../../../../include/x86_64-linux-gnu/qt5/QtCore/qtextstream.h:46,
                 from ../../../../include/x86_64-linux-gnu/qt5/QtCore/qdebug.h:49,
                 from qwt_interval.h:17,
                 from qwt_scale_div.h:14,
                 from qwt_abstract_scale_draw.h:14,
                 from qwt_abstract_scale_draw.cpp:10:
../../../../include/x86_64-linux-gnu/qt5/QtCore/qvariant.h: In constructor ‘QVariant::QVariant(QVariant&&)’:
../../../../include/x86_64-linux-gnu/qt5/QtCore/qvariant.h:275:25: warning: implicitly-declared ‘constexpr QVariant::Private& QVariant::Private::operator=(const QVariant::Private&)’ is deprecated [-Wdeprecated-copy]
  275 |     { other.d = Private(); }
      |                         ^
../../../../include/x86_64-linux-gnu/qt5/QtCore/qvariant.h:401:16: note: because ‘QVariant::Private’ has user-provided ‘QVariant::Private::Private(const QVariant::Private&)’
  401 |         inline Private(const Private &other) Q_DECL_NOTHROW
      |                ^~~~~~~
In file included from qwt_scale_map.h:14,
                 from qwt_abstract_scale_draw.cpp:14:
qwt_transform.h: At global scope:
qwt_transform.h:110:5: error: ‘QT_STATIC_CONST’ does not name a type
  110 |     QT_STATIC_CONST double LogMin;
      |     ^~~~~~~~~~~~~~~
qwt_transform.h:111:5: error: ‘QT_STATIC_CONST’ does not name a type
  111 |     QT_STATIC_CONST double LogMax;
      |     ^~~~~~~~~~~~~~~
make[1]: *** [Makefile:3807: obj/qwt_abstract_scale_draw.o] Error 1
make[1]: Leaving directory '/usr/local/src/qwt6/src'
make: *** [Makefile:50: sub-src-make_first-ordered] Error 2
[ERROR] Build failed. See output above for error messages.
[ERROR] Problem occurred while building package qwt6:
Build failed.
[ERROR] Error installing package qwt6. Aborting.

velichkov, 2019-11-23:
Hi @chiraglalkiya,

> root@kali:~# sudo pybombs install gr-gsm

Please report pybombs' problems in https://github.com/gnuradio/pybombs/ or https://github.com/gnuradio/gr-recipes as there is nothing really we could do about them here.

See also #491, #480 and #475