ptrkrysik/gr-gsm #364: got this error :Error installing package libosmocore. Aborting

TAPAN1539, 2017-12-28:
got this error :Error installing package libosmocore. Aborting

velichkov, 2017-12-28:
Hi @TAPAN1539 

Try to install `gnutls-dev` package if you are using a Debian based distribution.
 
If this doesn't help provide some more logs (the last 20 - 30 lines for example) and the full version of the  distribution you are using as it's rather difficult to help you having only a single error message.

ptrkrysik, 2018-02-13:
@TAPAN1539 , I'm closing this thread due to lack of more detailed description of the problem. 

@velichkov, many thanks for lots of support and help from you to the gr-gsm users :) . For example in this case I wouldn't be able to give any advice.

closed 2018-02-13 by ptrkrysik